thomluther/anker-solix-api #10: Export 2 solarbanks, MWR MI80

Rilgamon, 2024-02-10:
[rob321.zip](https://github.com/thomluther/anker-solix-api/files/14230096/rob321.zip)

Setup 2 solarbanks, 1 MWR MI 80

thomluther, 2024-02-10:
Thanks for the output. Will have a closer look to it

closed 2024-02-10 by thomluther

thomluther, 2024-02-10:
@Rilgamon  Few questions to understand the output:

- Were the solarbanks on standby when pulling the data? The power fields are empty in the user_devices list & site_details list like with mine which is currently not re-installed yet. But when I look at the scene details, which is the main source output for actual values, they show something (which is weird). This makes me believe user devices list anything that your mobile App connected to somehow (wifi or BT), but consumption data will not even be reported with site details but only with scene_details query? If that is true, I have to modify the main data collection routine of the api, since this currently uses the site_detail query to fetch data...

- Is the inverter fully included / assigned to your system?
The inverter fields indicate following:
It it not a managable device ("bind_site_status": "0")
"status": "0" => Was it off? Since also power values were empty
 "wireless_type": "2" => That means only BT connection, but no Wifi?
Your site details also do not show the inverter, which means it is not assigned to a site yet, therefore probably not manable. I flag this with is_admin = False

Hm, looking further I also found solarbanks showing status of 7 = Standby. That might explain why the site_details do not show data...
Would be helpful, if you can re-pull the data with the new release package 1.2 when you have fully integrated your inverter and the system is on to see which values will by gathered.