thomluther/anker-solix-api #7: No clue how pull requests work

Rilgamon, 2024-02-10:
For my personal use I've moved the credentials to a single file (settings.py)
Since I dont know if this is of use for others and I've no clue how PR are created from shell
I'll provide the changes via this gitea repo https://git.grml.de/bundesloser/anker-solix-api-PR

thomluther, 2024-02-10:
@Rilgamon the link does not work for me.
Potentially I could do the same for the test_api, simply importing the settings as constants and add the file to gitignore.
Then personal changes should not be reflected back in commits as far I understand this mechanism.

Rilgamon, 2024-02-10:
Oh, sorry! https://git.grml.de/robin/anker-solix-api-PR I'm a total noob when it comes to more than push and pull ;)

Exactly! Thats the idea. The new link should show how I did this.

thomluther, 2024-02-10:
Sorry, but link still does not work for me, maybe its private?

thomluther, 2024-02-12:
I found the problem. When git already tracks a file, the gitignore has no effect. It will only be used for newly created files.
I had to remove the files from the git index, so they stay in the repo as static files and local changes are ignored now for the test_api.py and the api/credentials.py.
Please try and let me know if that now works for you after feting the latest release

closed 2024-02-12 by thomluther