thomluther/anker-solix-api #15: v1.2 export 2 solarbanks, 1 MWR

Rilgamon, 2024-02-11:
[rob322.zip](https://github.com/thomluther/anker-solix-api/files/14231771/rob322.zip)

thomluther, 2024-02-12:
Hi @Rilgamon 
Which version did you use to create the export?
It seems that some values in the api_sites and api_devices did not get randomized but the keys got randomized.
I checked that with R1.2 and this was working correctly for me.
If you did not manually change anything there, this could be your real IDs, then I recommend to remove the zip file again and recreate with 1.3. Check the 2 files, their dict keys should match with the site_id resp device_sn listed in the values....(both randomized)

Rilgamon, 2024-02-12:
v1.2.0

This one is created with 1.3.0, I'll delete the other one to be sure.
[rob324.zip](https://github.com/thomluther/anker-solix-api/files/14239166/rob324.zip)

thomluther, 2024-02-12:
Hi
I'm not sure why, but looks the same, only the key was randomized but not the values in the 2 api overview files.
That is weird, since I cannot recreate this on my platform.
I'm using python 3.11  currently, do you use another major version that may explain the difference?
The randomize function is typically traversing all nested objects to randomize certain values.
While on your system it seems to work for the scene details json, it does not work for the api_sites json, which is typically 1 level less nested structure, but the same content wise....
I have no explanation for this different randomization behavior...

Rilgamon, 2024-02-12:
robin@debian:/opt/build2/anker-solix-api$ python3 --version
Python 3.11.7

I've checked and I dont see any values, that I'd consider "secret" that are not different.

thomluther, 2024-02-12:
OK, so you confirm that the site_id and serials in your api_*.json files are randomized indeed?
Weird why they show different value than the key in the dict, because the key is either the randomized device_sn or site_id (depending on the file). Nevertheless, it should match the corresponding value accordingly, and not show a different randomization, since that would invalidate the structure of the output, if the whole set of json files is used for HA integration testing...
Typically, one sensitive value is randomized only ONCE, and then looked up in subsequent randomizations to maintain the relationship across the various outputs accordingly.

thomluther, 2024-02-18:
Hi @Rilgamon , I'm trying to understand the dependencies for the networks given in your setup.
Your inverter shows wireless_type of "2", while the solarbanks show "1" in the scene info. 
However, you site wifi network list also shows 2 networks.
Does that mean the wireless_type  number is an index to the network list given in the wifi_list output?
If you can confirm this, and your inverter is connected to a different WLAN than the solarbanks, we have an understanding of the wireless_type number and I can use this to list more network details under the device details (Network name and signal strength).
See [issue#28](https://github.com/thomluther/anker-solix-api/issues/28)

Rilgamon, 2024-02-18:
I've a Fritz!-Powerline adapter providing wireless lan (my solarpanels are in the backyard) for all 3 devices.

thomluther, 2024-02-18:
Hm, its weird that 2 networks are in the wifi list, and both have the same name.
But the wireless_type for the inverter is different than for the 2 solarbanks.
Any idea why those are reported differently in scene_info output?
Also in bind devices I wonder about the different network output for the devices:
Inverter:
        "wifi_online": false,
        "wifi_name": "wifi-network-1",
        "relate_type": [
          "ble"
        ],

And Solarbank:
        "wifi_online": true,
        "wifi_name": "",
        "relate_type": [
          "ble",
          "wifi"
        ],

So when they are online to the wifi, the wifi_name is empty, which is weird already.
But the inverter is offline, shows the wifi name but no wifi in relate_type list......
I wonder how this fits together? Maybe there is also inconsistency how the api reports wifi states for various device types...

Rilgamon, 2024-02-18:
I checked and all 3 devices are connected over 2.4GHz and not 5GHz.
[image]
So the connection for them is really completely identical to me. I repeated the export right now and made sure the inverter is active. Output is still the same. The MI-80 appears to me as a device that is not supposed to communicate locally like other inverters that provide a webserver/api to fetch info from. Could it be that it is intentionally "crippled" to deny it has network-capabilities? Bluetooth-only?

thomluther, 2024-02-18:
I don't think this is the reason for the different network values, the solarbank has no web interface by itself either....
PS: The 1.4 version will have additional info in the export.
I also got the solar_info query to work. It just shows the defined inverter details for a solarbank device. Anyway, some information that I can also show in the HA integration eventually.
I also added the request body text in the console output when the request fails with an error.
Typically the Api tells you which required field is missing for a request, and also when the type of the field value is wrong.
This way, I could not figure out the correct parameters for the solar_info query.

closed 2024-02-27 by thomluther