thomluther/anker-solix-api #19: Need helper function to change the home load

thomluther, 2024-02-17:
A helper function is needed (also for Home Assistant integration) to simplify change of the set home load.
Points to clarify and validate before this can be provided:

- Can the set_home_load endpoint be used with the same schedule parameter dict as the set_device_parm endpoint?
- Does the schedule parameter dict require a full schedule as received with either get_home_load or get_device_parm endpoints?
- Is it sufficient to modify only the active schedule slot with the new parameters for home load and discharge setting, or may this delete the other slots in the schedule?
- If only a single slot is supported, what happens if the times of the slot are changed? Will the modified timings delete/modify the adjacent slots like in the App?

The help function should provide option switches for:
- new home load value
- setting of discharge switch
- whether the change should be applied to current slot only or all slots
If the whole schedule or parts of it needs to be changed, this should be done via the set_device_parm method providing the full schedule json.

thomluther, 2024-02-22:
The method also need to ensure to update the internal device dictionarie with the modified schedule for each solarbank device. This is important that changes are immediately reflected in the api data structures and not delay until next device details update cycle.

thomluther, 2024-02-26:
The schedule slots are also different for single and dual bank setups which must be considered for the helper.
**Schedule slot with Single Solarbank in system**
```
        {
          "id": 0,
          "start_time": "20:00",
          "end_time": "24:00",
          "turn_on": true,
          "appliance_loads": [
            {
              "id": 0,
              "name": "Benutzerdefiniert",
              "power": 100,
              "number": 1
            }
          ],
          "charge_priority": 80,
          "power_setting_mode": 1,
          "device_power_loads": [
            {
              "device_sn": "25MFU7OHZJTFGFSH",
              "power": 50
            }
          ]
        }

```
**Schedule slot with Dual Solarbank in system**
```
        {
          "id": 0,
          "start_time": "17:30",
          "end_time": "24:00",
          "turn_on": false,
          "appliance_loads": [
            {
              "id": 0,
              "name": "Benutzerdefiniert",
              "power": 170,
              "number": 1
            }
          ],
          "charge_priority": 80,
          "power_setting_mode": 1,
          "device_power_loads": [
            {
              "device_sn": "VQQDB61DTTZEPSVX",
              "power": 85
            },
            {
              "device_sn": "DG34YZMOHGOTW6W2",
              "power": 85
            }
          ]
        }
```

thomluther, 2024-03-06:
First tests showed that the set_device_load endpoint accepts the new schedule without error, but it doesn't seem to get applied.
Schedule modifications are only applied via the set_device_parm method.
The parameter must always be the full schedule. Missing slots will automatically extended to complete a full day with the provided slots.
The Power settings must only be applied to the appliance load dict. This will automatically be devided for the individual device power loads. They don't have to be provided for schedule modifications.

Important and modifyable parameters for a slot:
          "start_time": "17:30",
          "end_time": "24:00",
          "turn_on": false,
          "appliance_loads": [
            {
              "name": "Benutzerdefiniert",
              "power": 170,
            }
          ],
          "charge_priority": 80,

p4i1ipp, 2024-03-07:
Maybe you can have a look in the npm module ankersolixe1600 - there is setting the homeload (or turning home feeding off) implemented.

thomluther, 2024-03-10:
The mechanism for appliance load setting is understood and implemented in a helper function.
App version 2.1.0 implemented new capabilities for dual solarbank setup to modify the load sharing. Default so far was 50% of appliance load. The new mechanism is not understood yet and I can't test it. It can be added at a later point when this is clear.
I assume this is why the set_device_load endpoint was implemented. Maybe this works only for dual bank setups, and does nothing for single bank setup...