thomluther/anker-solix-api #48: Bug in Solar power/Charge power

Rilgamon, 2024-02-22:
The value displayed for "charge power" is not the correct value. This is related to our conversation over at the forum.
My two banks are set to charge and in the app the charge is displayed roughly the same value as the solar power.
No energy to the house.

solarbank_monitor.py output looks like this:

`Solar Power    :  79 W                Charge Power:   0 W
Output Power   :   0 W                (Output Preset:   0 W)

Solar Power    :  80 W                Charge Power:   0 W
Output Power   :   0 W                (Output Preset:   0 W)

AC Power       : 0.00 W`

thomluther, 2024-02-22:
Hi @Rilgamon 
Can you please create a new configuration export at a time when you see such discrepancies?
I had a first look and I'm not sure where the mismatch comes in.
I need to see what the api reports back on the various queries. Maybe there is already a mismatch between the fields reported....

Rilgamon, 2024-02-22:
Created with 1.4
[rob2202.zip](https://github.com/thomluther/anker-solix-api/files/14373731/rob2202.zip)

thomluther, 2024-02-22:
Hi @stuertz 
I think there comes garbage from the Api: This is what I see in the scene info that is used for the values:
```
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C0",
          "device_name": "Solarbank E1600",
          "battery_power": "56",
          "bind_site_status": "",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "3",
          "status": "1",
          "wireless_type": "1",
          "main_version": "",
          "photovoltaic_power": "267",
          "output_power": "0",
          "create_time": 1695394124,
          "set_load_power": ""
        }
      ],
      "total_charging_power": "267",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.56",
      "updated_time": "2024-02-22 14:42:12",
      "total_photovoltaic_power": "267",
      "total_output_power": "0.00",
      "display_set_power": false
    },
    "retain_load": "0W",
```

so pretty much what is reported by the Monitor. I use the charging power of the device:
          "charging_power": "0",
However, this doesn't seem to be tracked per device?
Only the solarbank summary value shows something
      "total_charging_power": "267",

No Idea how I should break this down to a device when more than 1 bank is installed....need to check @Rilgamons data

thomluther, 2024-02-22:
@Rilgamon 
Here are the relevant fields from your dual bank setup:
```
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C0",
          "device_name": "SB2",
          "battery_power": "46",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "3",
          "status": "1",
          "photovoltaic_power": "72",
          "output_power": "0",
          "set_load_power": "0"
        },
        {
          "device_pn": "A17C0",
          "device_name": "SB1",
          "battery_power": "52",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "3",
          "status": "1",
          "photovoltaic_power": "72",
          "output_power": "0",
          "set_load_power": "0"
        }
      ],
      "total_charging_power": "144",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.49",
      "updated_time": "2024-02-22 13:17:30",
      "total_photovoltaic_power": "144",
      "total_output_power": "0.00",
      "display_set_power": true
    },
```

I don't know how to fix this api problem.
Eventually this can be broken down by some percentage based calculations. It might be easy for only charge mode, but difficult in bypass mode and both banks act slightly differently....

thomluther, 2024-02-22:
Yes, that might work:
Device Photovoltaic power - Device Output power should be the charging power.
I could calculate that independently for both devices and distribute the Total charging power accordingly to the percentage of the calculated device ouput theoretical total.

thomluther, 2024-02-26:
Hi @Rilgamon @stuertz 
Can you please retry the monitor with the 1.5 release and provide another system export please?
I want to use your exports as other examples, since they contain quite unique information I haven't seen before
- 2 banks in 1 system + MI80
- 0W output switch fitting

I added this already to the device details output, also additional fields with calculated battery energy and capacity.
The charging value should now also be fixed with a calculation and the output presets per device and schedule should also be reported properly for single banks as well as dual banks setup

Rilgamon, 2024-02-26:
2 banks in 1 system + MI80:
[rob2602.zip](https://github.com/thomluther/anker-solix-api/files/14400453/rob2602.zip)

stuertz, 2024-02-26:
```
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1, Devices: 1
Device         : Solarbank E1600      (Alias: Solarbank E1600)
PN             : A17C0                (Type: Solarbank)
Wifi state     : Online               (Signal: 40 %)
SW Version     : v1.5.6               (Auto-Upgrade: Enabled)
Status         : on                   (Status code: 1)
Charge Status  : charge_priority      (Status code: 3)
State Of Charge:  27 %                (Min SOC: 10 %)
Solar Power    : 269 W                Charge Power: 269 W
Output Power   :   0 W                (Output Preset:   0 W)
Schedule       : 11:29 UTC +0100      (Current Preset: 0 W)
ID Start End   Discharge Output ChargePrio
 0 00:00 07:00    YES     100 W       80 %
 0 07:00 20:00    NO      150 W       80 %
 0 20:00 24:00    YES     100 W       80 %
```
[jan2.zip](https://github.com/thomluther/anker-solix-api/files/14403251/jan2.zip)

thomluther, 2024-02-26:
Hi @stuertz 
Your output looks now good to me, or do you still see some weird value combinations?
I think I will add also the new generated fields for battery_capacity and battery_energy that I calculate from the Solarbank Name (Capacity) and the Soc (energy). Interestingly I don't see those new fields in the dual solarbank export, and don't know why...
But I will use them for the HA integration anyway, just need to make sure they are generated for both constellations.

stuertz, 2024-02-26:
> Hi @stuertz Your output looks now good to me, or do you still see some weird value combinations? I think I will add also the new generated fields for battery_capacity and battery_energy that I [...]

LGTM

thomluther, 2024-02-27:
Hi @Rilgamon @stuertz 
I included more fixes for another weird API result from server 
Also I found some missed serial randomizations in the export.
I restructured the whole export module to avoid this in future. It also uses less queries now 
•	Findings in system exports #55 
•	Bug in Solar power/Charge power #48
**Please remove your old zip files from the repo and export again with the 1.5.1 release.**

@Rilgamon 
Actually one problem is caused by Api inconsistency using alias names for device names in some query results. I'm glad that you did rename your solarbanks, otherwise this Api inconsistency would have probably missed for the HA integration. The Api returns the alias name as device_name in some queries which was therefore mapped incorrectly in the device details. This is now fixed for the scene_info result handling when mapping device details.
It would be very usefull to see if this problem also exists for inverter aliases. 
**Therefore, if possible, can you change your inverter name in the App as well to something different than the default (fixed) device name before you export the data again?**
This will be important for validation, since I consider now all device_name returned in the scene info as 'alias_name' for any device type. I need to make sure to map them correctly for device details fields which are used by the HA integration.
I will change my solarbank name too, and hope to see such API inconsistencies then earlier.

Rilgamon, 2024-02-27:
Sure
[rob2702.zip](https://github.com/thomluther/anker-solix-api/files/14418822/rob2702.zip)

stuertz, 2024-02-27:
For reference: [jan3.zip](https://github.com/thomluther/anker-solix-api/files/14421065/jan3.zip)

thomluther, 2024-03-01:
I'm re-opening this. Found more confusing info in the charging power values for Solarbank device and total SB devices and implemented following:
Work around for weird charging power fields in SB totals and device list: They have same names, but completely different usage
SB total charging power shows only power into the battery. At this time, charging power in device list seems to reflect the output power. This is seen for status 3
SB total charging power show 0 when discharging, but then device charging power shows correct value. This is seen for status 2
Conclusion: SB total charging power is correct total power INTO the batteries. When discharging it is 0
Device list charging power is ONLY correct power OUT of the batteries. When charging it is 0 or shows the output power.
Need to simplify this per device details and SB totals, will use positive value on both for charging power and negative for discharging power
calculate estimate based on total for proportional split across available solarbanks and their calculated charge power.
I will monitor this work around for a few days to see if that works out.
I believe the negative values for discharging power also make the understanding bit easier and better consumable in Home Assistant