thomluther/anker-solix-api #70: Verify if encryption can be added to body like App does

thomluther, 2024-03-11:
The App encrypts also request bodies via ECDH like login requests.
This is indicated by additional header fields. 
Verify if encrypted bodies can be used by the API library as well.

thomluther, 2024-04-01:
See details in  Question about encryption & (hopefully) some useful data #18

thomluther, 2025-04-23:
See additional information in [#107](https://github.com/thomluther/anker-solix-api/issues/107#issuecomment-2819565431)

thomluther, 2025-04-27:
Hi @ovflowd, last couple of days I tried to test your posts for the payload encryption, but I was not successful unfortunately.
I keep tracking this topic in this issue. So let me try to summarize.

## Request Compression

While off topic, in your [curl example](https://github.com/thomluther/anker-solix-api/issues/107#issuecomment-2819565431) as well as in the [former App screenshot here](https://github.com/thomluther/anker-solix-api/issues/18#issue-2139505404) I noticed the client is signaling the compression option to the server by using the header field 'accept-encoding' = 'gzip' or --compression with curl.
Typically the server should compress the data if supporting the compression type and signal to the client with the header 'Content-Encoding' = 'xxx' that data is compressed and which encoding was used according to [this doc.](https://webscraping.ai/faq/curl/what-is-the-purpose-of-the-compressed-option-in-curl).
I tried it but I never noticed that the Anker server responded with a 'Content-Encoding' header field. so I assume while various clients (even the App) request compression, the Anker Cloud server does not support it?

Did you made any observations in this topic, independent of the payload encryption?

I think this would be a useful feature to reduce Api traffic if it would be supported by the cloud server.

## Payload encryption

You mentioned that you got he payload encryption working and posted as well examples summarized by AI.
First I tried to utilize them with the Api session private / shared key of the authentication request. But then I noticed that this key pair is just required for the authentication request itself and the rest of the requests just need the following identification info:
```python
           mergedHeaders.update({"gtoken": self._gtoken})
            mergedHeaders.update({"x-auth-token": self._token})
 
```

The 'x-auth-token' is contained in the login response and the 'gtoken' is just an MD5 hash from the 'user_id' as following:
```python
            # gtoken is MD5 hash of user_id from login response
            self._gtoken = md5(data.get("user_id"))

def md5(text: str) -> str:
    """Return MD5 hash in hex for given string."""
    h = hashes.Hash(hashes.MD5())
    h.update(text.encode("utf-8"))
    return h.finalize().hex()
```

The whole login response is cached locally, together with the public key that was used for the request. I assume that is the public client key, since the public server key is hard coded. Following are the relevant fields of the login response:
```python
{
  "user_id": "304a### masked ###ed### masked ###56dd44", # will be used for MD5 creation of 'gtoken'
  "auth_token": "0f4f### masked ###f7### masked ###d7### masked #", # will be used as is
  "token_id": 66####08,  # number format, not sure where this is used
  "geo_key": "AIza### masked ###Kx0a### masked ###LJQ", # Not sure where this is used (base64 format)
  "server_secret_info": {
    "public_key": "04" + 32 Byte + 32 Byte)
  },
}
```

However, that public key from the login response is no longer of any use, since next session would create a new key pair that will just become required if a new login request must be done, e.g. when the token is expired or was deleted on the server due to another client login request.

Here is the login request format: 
```python
            auth_resp = await self.request(
                "post",
                API_LOGIN,  # "passport/login" endpoint
                json={
                    "ab": self._countryId,  # e.g. 'DE'
                    "client_secret_info": {
                        # client public_key is uncompressed format of points in hex (0x04 + 32 Byte + 32 Byte)
                        "public_key": self._rawPublicKey()
                    },
                    "enc": 0,
                    "email": self._email,
                    # password is AES-256-CBC encrypted by the ECDH shared key derived from server public key and local private key
                    "password": self._encryptApiData(self._password),
                    # time_zone is offset in ms, e.g. 'GMT+01:00' => 3600000
                    "time_zone": round(datetime.utcoffset(now).total_seconds() * 1000),
                    # transaction is Unix Timestamp in ms as string
                    "transaction": self._timestamp(),
                },
[... 30 lines cut ...]
Therefore I believe the payload encryption must process another key exchange just for the payload encryption of that session.
And here is where I start to struggle with your examples.

1. Your java script example for the key exchange are incomplete in terms of required header field generation, the server responds the field names that are missing

2. Your python example seems to be complete in terms of generated header fields, but the code is invalid for the 'raw_public_key'
It needs following change to work:
```python
from cryptography.hazmat.primitives import serialization

        # Get raw public key bytes
        # raw_public_key = self.public_key.public_bytes(
        #     encoding=ec.Encoding.X962,
        #     format=ec.PublicFormat.Uncompressed
        # )
        raw_public_key = self.public_key.public_bytes(
            encoding=serialization.Encoding.X962,
            format=serialization.PublicFormat.UncompressedPoint
        )
```
[... cut ...]
I have no clue whether that was just AI python code that never actually worked, or whether there is a conversion issue from any of your working  java script ? code, that you mention to get working?
Do you have complete java script code that works for key exchange as well as for payload request encryption and payload response decryption?
Or even more usefull, do you have validated python code for the key exchange and payload encryption/decryption that I can verify with my auth_token and then incorporate it into the api session module?

ovflowd, 2025-05-03:
> While off topic, in your https://github.com/thomluther/anker-solix-api/issues/107#issuecomment-2819565431 as well as in the https://github.com/thomluther/anker-solix-api/issues/18#issue-2139505404 I noticed the client is signaling the compression option to the server by using the header field 'accept-encoding' = 'gzip' or --compression with curl.

I've also noticed that.

> I tried it but I never noticed that the Anker server responded with a 'Content-Encoding' header field. so I assume while various clients (even the App) request compression, the Anker Cloud server does not support it?

Genuinely speaking, the data was never compressed, so no idea why the gzip flag was given, I bet it is just the client requesting it by default.

> Did you made any observations in this topic, independent of the payload encryption?

I noticed all requests from the client regardless the endpoint have such request header, so I believe your observations are right.

> You mentioned that you got he payload encryption working and posted as well examples summarized by AI.

Apologies, but I was never able to actually decrypt the message, I got stuck at the exact same pieces at you, and was also wondering the correlation. What I was able to identify is that the Client Key most likely is _not just a key_ but also encompassed by extra data, since apparently the key is prefixed by a timestamp (encoded) my code was trying to break the client key into segments to see if I could find anything useful, but without knowing what characters/bytes and at which byte the key ends or starts it is just a dictionary attack without luck. Even more as it seems the Server will reject with an HTTP Code 466 (not a real HTTP code) if the timestamps are off already.

> First I tried to utilize them with the Api session private / shared key of the authentication request. But then I noticed that this key pair is just required for the authentication request itself and the rest of the requests just need the following identification info:

Right the g-token and auth-token are just the standard Anker API user authentication, the same your API is already doing.

> The whole login response is cached locally, together with the public key that was used for the request. I assume that is the public client key, since the public server key is hard coded. Following are the relevant fields of the login response:

If I recall correctly (pardon my memory) the Server Public Key did not change between requests; This is common to Diffie-Hellmann, the key itself (from server) is not an authorization one, but more for allowing the client to decrypt the data; But there's clearly some extra custom code within the client that uses pieces of the server key to actually decrypt the data, I feel the server key is not just **one** key but a combination of keys (likely the IC,IV maybe Salt and what not...)

> However, that public key from the login response is no longer of any use, since next session would create a new key pair that will just become required if a new login request must be done, e.g. when the token is expired or was deleted on the server due to another client login request.

What I noticed is that this "OAuth" endpoints are not requested when the user is authenticating on the App; they are done during App bootstrap, regardless if you are authenticated or not as an user within the App. Note that it doesn't happen on every Bootstrap, the data seems to get cached on the file system. Only when I clear the App cache/data it requests it again, or after a certain amount of time has passed.

> Therefore I believe the payload encryption must process another key exchange just for the payload encryption of that session. And here is where I start to struggle with your examples.
>
> 1. Your java [...]

Yeah those snippets are derived from some basic JavaScript code I've done for testing, which I unfortunately don't have them anymore; I only asked ChatGPT to generate these counterparts in a hurry.

> I have no clue whether that was just AI python code that never actually worked, or whether there is a conversion issue from any of your working java script ? code, that you mention to get working?

Unfortunately I never got past full decryption; Because I was never able to successfully extract the IV/IC that are fundamental for the decryption here, and that either are hard-coded within the App or decoded from the Server Key; There's definitely something custom here that is out of our grasp.

---

Unfortunately my landlord requested me to uninstall my solar system, so I won't be able to provide any further support here _regarding Solarbank_; With the Solarbank disconnected, apparently the Smart Plugs are still working and Im still able to configure them and turn them on/off; So at least I will be able in the future to provide support on that matter. I just need to dissect more these API requests, I might have time during next week; If there's anything specific you want me to try to inspect or test, let me know.

ovflowd, 2025-05-03:
Also, @thomluther apparently these smart plugs although say that require a Solarbank 2 to work, they seem to be completely working withouit the Solarbank (I even removed it from the home); so they work 100% standalone (at least the part of measuring use, turning off/on, setting schedules, monitoring them, etc)

thomluther, 2025-05-03:
> Also, [@thomluther](https://github.com/thomluther) apparently these smart plugs although say that require a Solarbank 2 to work, they seem to be completely working withouit the Solarbank (I even [...]

Yeah, the question is whether they will use any power Api endpoints during those modifications in the App. It would help already if you can sniff any potential endpoints that are being used while monitoring or modifying their settings. If there are no power api endpoints in use, its worthless to investigate further in that direction of getting Api support for the plugs.
I think these device settings won't be different if assigned to a system.
It is like with some other Solarbank device settings or values that are simply not available in the power Api, only via the cloud MQTT server.

thomluther, 2025-05-03:
> Unfortunately I never got past full decryption; Because I was never able to successfully extract the IV/IC that are fundamental for the decryption here, and that either are hard-coded within the App or decoded from the Server Key; There's definitely something custom here that is out of our grasp.

My hope was that you got a least the key exchange request to work without an error. That would be a big step forward, since I did not know that there are extra keys exchanged at all.
I think all header fields that are MD5 hashes can just be used for comparisons between Client and Server, since MD5 hashes cannot be reversed. That means, client and server need to build them in the same way and then compare those fields to avoid that anything was tampered. So theoretically, they can contain a timestamp, but this timestamp must also be transferred unencrypted to allow the other site to use the same timestamp for MD5 hash creation. The random part in your snippets (16 Byte suffix) therefore does not make any sense to me, since the other site would never be able to rebuild the MD5 in same way.
Maybe the first part is a timestamp, that is also transmitted in clear 'x-request-ts: [timestamp]'.
The question is what is the 'x-auth-ts: [timestamp]'? Same as request timestamp, or maybe a derived timestamp that must match to the auth_token creation? The authentication request does not return a timestamp with the token, only an expiration timestamp, that is + 7 days. But this expiration timestamp could be used for backward calculation of the authentication timestamp.

So generally speaking, this is most likely wrong:
```
    def _generate_key_ident(self, timestamp):
        """Generate key ident using timestamp and random data"""
        data = timestamp.encode() + os.urandom(16)
        return hashlib.md5(data).hexdigest()

    def _generate_request_once(self, timestamp):
        """Generate request once using timestamp and random data"""
        data = timestamp.encode() + os.urandom(16)
        return hashlib.md5(data).hexdigest()
```
Just guessing here, but maybe:
- x-key-ident': built with authentication timestamp + first 16 bytes of public client key from authentication (available in the (cached) login response. The used timestamp is provided also clear
- 'x-request-once': A generated request timestamp + first 16 bytes of public client key from authentication (available in the (cached) login response. The generated request timestamp is provided also clear
Both could then be used/rebuild by the server and compared for tampering.
The even more tricky part is the signature hash generation. This could be any order of various other fields, but the clear text body is probably part of it. 
The body en/decryption is then symmetrical with derived AES key. Tampering could be verified against the transferred and rebuilt signature SHA256 Hash.

A lot of guessing here, but getting only the key exchange to work would help I guess.
What is not possible is to decrypt any sniffed payload, since you don't know the used ECDH private key that was generated by the App.

ovflowd, 2025-05-03:
> Yeah, the question is whether they will use any power Api endpoints during those modifications in the App. It would help already if you can sniff any potential endpoints that are being used while monitoring or modifying their settings. 

That's what I reported before, when modifying the settings, like once I click save, I don't see any HTTP request (at least through HTTP Toolkit) -- I will either later today or next week try to debug the whole thing again via HTTP Toolkit and see. I'm going to keep an eye on the endpoints you mentioned before. Maybe HTTP Toolkit is just not capturing them, that's why I initially thought they were doing direct local comms via a different protocol or port (like gRPC or IDK)

> I think these device settings won't be different if assigned to a system.

As far as I can say, things seem to be working even without the Solarbank, but I still have the system configured.

> Maybe the first part is a timestamp, that is also transmitted in clear 'x-request-ts: [timestamp]'.

Yes, the first bytes of the client key are indeed a timestamp, after comparing the data of numerous requests, (keep in mind I used Cursor with Claude to keep brute forcing it and trying different things just to save some time; And at least it matches the timestamp)

> The question is what is the 'x-auth-ts: [timestamp]'? Same as request timestamp, or maybe a derived timestamp that must match to the auth_token creation? The authentication request does not return a [...]

I'll investigate that once I'm into it 👀 

--

Yeah there's lots of guessing here, I'll try to keep an eye and observe these things once Im at it again.

thomluther, 2026-09-11:
Hi @kb1ibt

Since you do a lot of App debugging with AI assistance, I wonder whether you are able to see the Api queries that are issued by the anker app and you are able with the toolkit to decrypt the payload?

That may not be easily possible, but what about having AI decompiling the Api encryption key exchange in order to get the composition of encryption request header fields as mentioned in [comment above](https://github.com/thomluther/anker-solix-api/issues/70#issuecomment-4160789732)?

I have seen you found a lot of handshaking for the BT communication with the newer devices that may use similar encryption methods as the Api payload encryption.

kb1ibt, 2026-09-11:
I'm not doing APK decompiles; I'm using a mixture of over-the-wire packet captures and iOS app_log.log parsing. By using a special developer profile from Apple, iOS lets you use a tool called PacketLogger (part of Xcode's additional tools) to capture full BLE traffic and then pipe it through tshark (part of Wireshark). The **real** gold mine came from using iMazing to back up the device and extract the debug log Anker stores in its app's Documents folder, since it stores the per-connection handshake private and public keys, along with both the encoded and decoded packets. The hard part there is twofold: first, it logs some things in English and some things in Simplified Chinese (UTF-8 with CJK characters); second, because of how verbose the logs are, sometimes AI tries to use grep or other tools to parse the file contents and then skips the context in surrounding lines. I then used a mix of SolixBLE (with my own PRs) and the mqttmap file in this project to convert the frames into Influx line format for storage and querying in Grafana.
Where I am starting to do significant AI-assisted debugging and decompiling is Anker's firmware .bin files, and I've gotten pretty far in mapping how their ESP32 talks to the MCU and BMS, to more fully understand the frames and fields that are not decodable in the logs, or to determine which unit system or even array/struct/data format is in place.
So yes, I can take a look to see what I can provide, but it'll be blind since I don't have one of those devices.

kb1ibt, 2026-09-12:
@thomluther — a follow-up to my last comment, where I said "it'll be blind". It turned out not to be blind at all.

Claude:
> The payload encryption is account/session-level, not device-specific, so no special device was needed — I reconstructed the whole scheme from kb1ibt's `app_log.log` captures and confirmed it live, end to end, on their account.
>
> The handshake came from kb1ibt's packet captures plus the iOS `app_log.log` (the newer app builds log the whole ECDH exchange in cleartext at the native `AnkUserModuleNative`/`ESIot` layer: app private+public key, keyIdent, presetKey, sharedKey, and full request cURL dumps). No APK decompile needed. Here's the composition, verified against a captured request:
>
> **The missing piece is a `presetKey`** — a region-specific AES-128 constant the app stores in `userConfig` as `presetKey` (constant per region/build, never returned in a response body). The US value is `e8ad18f61bbd3fbd52d5ed12d14d3b9c`, and it decrypts every captured key-exchange point from app 2.8 (2024) through the current 3.23. Everything below keys on it.
>
> **Envelope** (used for both public keys and request bodies):
>
> `base64( IV(16) || AES-128-CBC(key, IV, plaintext, PKCS7) )` — random IV prepended. The wrapped public key is the point's **130-char ASCII hex** string (`04`+X+Y), not the 65 raw bytes. That's the `[:96]` you were slicing: 16 IV + 80 padded.
>
> **Key exchange** `POST openapi/oauth/key/exchange`, `content-type: application/json`, body `{"client_public_key": "<presetKey-wrapped ascii-hex point>"}`. `data.server_public_key` is wrapped the same way. **`sharedKey` = the raw ECDH output (32-byte X coord), used directly — no HKDF.**
>
> **The `x-*` fields — and the key correction: `x-key-ident` and `x-request-once` are not MD5 of anything.**
>
> - `x-key-ident` — a random 32-hex id generated **once per key pair**, sent unchanged on every request under that shared secret (the server uses it to look up the shared secret).
> - `x-request-once` — a fresh random 32-hex **nonce, per request**.
> - `x-request-ts` / `x-auth-ts` — unix seconds.
> - `x-signature` = `HMAC-SHA256( key = presetKey_ascii_hex, msg = "{x-request-ts}+{x-request-once}+{wrapped-body-base64}" )`. The HMAC key is the presetKey **as its hex string**, the fields are joined with a literal `+`, and the signed "body" is the wrapped base64 on the wire (for the key exchange, the `client_public_key` value).
>
> **Normal requests**: `content-type: text/plain`, body = wrapped base64 of the compact JSON; response returns an encrypted `data` field plus a `signature`.
>
> **Worked example** (ephemeral, expired):

```
presetKey       e8ad18f61bbd3fbd52d5ed12d14d3b9c
app_priv        6e7c5bfe57605d35f3c2d4ba310a88cab74d25da4b93e25a09d59fa1826726ff
sharedKey       98176d509b285e9b5a08db6e568c6733209d3b42a6b89f04bcf0d243f970b8ae   (raw ECDH)
x-request-ts    1780350287
x-request-once  3bdbc6f6d00546319b50b6e303645e0e
x-signature     3172cd2ed111afab705c5d0e3fd6c665b413904bae5f3106b3ace3d52fe76514
```

> Unwrap the request's `client_public_key` with `presetKey` → the app public point; `HMAC-SHA256(presetKey_hex, f"{ts}+{once}+{client_public_key_b64}")` → `x-signature`. Both check out.
>
> **Body cipher** (confirmed against captured traffic): request/response bodies are AES-128-CBC, key = a 16-byte session `securityKey`, IV = the leading 16 bytes of the ciphertext, PKCS7. Decrypting real captured bodies with the logged `securityKey` reproduces their plaintext exactly. `securityKey` is a per-auth session key: the app runs the ECDH once per authentication and caches the derived key for that login epoch (stable across months within one login, changes on re-login). The derivation is `shared_secret[:16]` — the app's own BLE negotiation logs show it directly (a logged shared secret, its first 16 bytes taken as the AES key, and the frames then decrypting). One nuance if you cross-check captures: the app runs this exchange per subsystem, so you can see two different keys against `key/exchange` in one session (different SDK layers, each its own ECDH). A single client does one exchange and takes `shared_secret[:16]`.
>
> **The signing key switches after the handshake.** This is the one piece that isn't obvious: the key exchange signs with the **presetKey** (there's no session key yet), but **every request after the handshake signs with the session key `shared_secret[:16]`** — same HMAC-SHA256 over `ts+once+wire-body`, just a different key. Signing a normal request with the presetKey is rejected 403 by the gateway; signing with `shared_secret[:16]` is accepted. This is the same process the BLE client uses for opcode `0x0005` (`0805`/`4805`).
>
> **Confirmed live, end to end:** on `get_relate_and_bind_devices`, the request body encrypted with `shared_secret[:16]` is accepted (`Bizcode: 0`), and the encrypted response (`Encrypt-Algorithm: algorithm_ecdh`) decrypts with `shared_secret[:16]` to the real device list. Handshake, envelope, both signatures, request-body encryption, and response decryption all round-trip against the live server.

Working implementation of all of the above (verified end to end): #345.

thomluther, 2026-09-12:
@kb1ibt 
Many thanks for your testing and the PR. I can confirm, that the same preset key is working for the EU server as well.
There was just missing the deserialization of the decrypted data string, the rest seems to work as is which is great.

Now payload encryption can be enabled optionally. I still have to do more testing with it, eventually it will be enabled per default and just disabled optionally....

Getting this to work was really a pain in my neck, because the whole Api could break if Anker starts to enforce payload encryption on their servers...

thomluther, 2026-09-27:
Implemented with 3.8.2

closed 2026-09-27 by thomluther