thomluther/anker-solix-api #101: Enhance energy statistics for solarbank 2 and smartreader

thomluther, 2024-06-01:
The energy statistics endpoint on the server has been enhanced with totals solarbank 2 and smartreader devices.
Additionally 2 more regular statistic types have been added for 'home_usage' and 'grid' energy.
This additional types should be optionally supported in the daily energy statistics helper methods as well as the energy_export utility.

thomluther, 2024-06-01:
Note: Any solarbank 2 and smartmeter enhancements needs to be tested by the community!

kitokirisaki, 2024-06-19:
@thomluther if you want me to check / test something, just ping me. I don't have the smart meter right now installed but should happen hopefully also soon.

Disane87, 2024-06-20:
If you need some tests, you can ping me too. Got my SolarBank 2 pro today and the smart meter is coming tomorrow. Will be open for assistance :)

thomluther, 2024-06-26:
Added new metrics to export, but will need example exports to understand how they can be used exactely for the energy details and HA integration

closed 2024-06-26 by thomluther