thomluther/anker-solix-api #105: Add Solarbank 2 inverter channel power to device details

thomluther, 2024-06-25:
Solarbank 2 provide additional fields for each inverter channel power.
They will be added to the solarbank device details.

stephan-l, 2024-06-29:
Is it possible that ac_output is the AC socket on the Pro version? I'll connect some devices later today to watch that value.

thomluther, 2024-06-29:
That is my assumption too, would be good if you can prove it with reliable power consumer to compare the value.
I plan to add the new fields only to the solarbank device cache when the model supports the field. Otherwise the HA integration will create entities that always show 0 W which is not usefull for the device.
I did the same already for the solar power chanel 1-4, they are only shown when applicable to the solarbank model type.
The plus version probably uses only channels 1-2 and the later normal version has no inverter included to my knowledge.
The ac_power should then be added only to the pro devices since that is the only one with a socket.

thomluther, 2024-06-29:
By the way, following are the new added solarbank_info fields in the scene_info for any site containing solarbanks.
      "ac_power": "0",
      "to_home_load": "0",
      "is_display_data": true,
      "solar_power_1": "0",
      "solar_power_2": "0",
      "solar_power_3": "0",
      "solar_power_4": "0"

The ac_power may be the power socket to be confirmed.
But what is the to_home_load field?
Is that used in any of your setups and what does it show? Maybe it is duplicate information, since there is already a field as used by Solarbank 1 systems:
      "total_output_power": "97.00",
How does to_home_load relate to total_output_power for the solarbank 2? Sounds to be the same, but could also be the combination of bypass and discharge power, which was not possible for Solarbank 1?

stephan-l, 2024-06-30:
I tried to replicate this but I experienced a flaw. The output is shut down at flat battery. So I had to disconnect my device before I collected data. I need to wait for recharging the battery and it a rainy day...

stephan-l, 2024-07-10:
I just tried it with a small hairdryer on cold. The result wasn't as expected:
```
Running site refresh...
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1,  Devices: 2
--------------------------------------------------------------------------------
Device         : NoName                  Alias         : Solarbank 2 E1600 Pro
Serialnumber   : APC        Admin         : NO
System         : T2  (Site ID: 99e)
Wifi SSID      :
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud-Updated  : 1970-01-01 00:00:00     Valid Data    : NO (Requeries: 3)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:    0 %                  Min SOC       :   10 %
Battery Energy : 1152 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    0 Pcs                AC socket     :    0 W
Solar Power    :    0 W                  Output Power  :    0 W
Solar Ch_1     :    0 W                  Solar Ch_2    :    0 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   :    0 W                  Device Preset :  200 W
Home Demand    :    0 W                  SB Home Load  :    0 W
```

I'll give it a new try after the firmware update of all components.

stephan-l, 2024-07-10:
After Firmwareupdate everything looks better:
```
Running site refresh...
Running device details refresh...
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1,  Devices: 2
--------------------------------------------------------------------------------
Device         : NoName                  Alias         : Solarbank 2 E1600 Pro
Serialnumber   :         Admin         : NO
System         : T2  (Site ID: 99e)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud-Updated  : 1970-01-01 00:00:00     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   35 %                  Min SOC       :   10 %
Battery Energy : 1120 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :  236 W
Solar Power    :    4 W                  Output Power  :  380 W
Solar Ch_1     :    1 W                  Solar Ch_2    :    3 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   : -376 W                  Device Preset :  200 W
Home Demand    :  380 W                  SB Home Load  :  144 W  (-)
--------------------------------------------------------------------------------
Device         : NoName                  Alias         : Smart Meter
Serialnumber   : AUJS020E17406976        Admin         : NO
System         : Trippstadt Flörsheimer Ring 2  (Site ID: 99e3a53b-87cd-4fb9-962d-ca020eb8a579)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :    0 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------
Api Requests: 4 last hour, 4 last minute
Site refresh:  12 sec,  Device details refresh: 282 sec  (CTRL-C to abort)
```

I'll provide an admin account full dump in the other thread

thomluther, 2024-07-10:
Thx, looks good, so AC socket seems to be correct field, and this is also counted to home demand, therefore my indicator for SB homeload shows missing home load, compared to demand, but that is no problem.