thomluther/anker-solix-api #107: Explore usage of new device_attrs endpoints

thomluther, 2024-06-25:
There are new endpoint with Solarbank 2 support that needs to be explored for capability and usability:

```
print(json.dumps((await myapi.request("post", "power_service/v1/app/device/get_device_attrs",json={"device_sn": deviceSn, "attributes": []})), indent=2))
print(json.dumps((await myapi.request("post", "power_service/v1/app/device/set_device_attrs",json={"device_sn": deviceSn, "attributes": []})), indent=2))
```
The get_device endpoint is included now in the export_system tool, but it is not clear which keywords can/should be specified for the mandatory attributes list. 
- Will an empty list export all attributes supported by a device?
- Can the set_device_attrs endpoint be used to modify a supported attribute?

thomluther, 2024-07-12:
There are device attributes returned when the list is empty in the request.
Someone will have to try various elements (strings) for potential attributes or device types or whatever?
Eventually the SB2 user mode setting or the Light setting is returned in the response.
Might also have the temperature? Get creative to find parameters that may return something...

Lenoirio, 2024-07-14:
I did a test with my SB2 (with the request as in your start-post) and this is the response:

{"code":0,"msg":"success!","data":{"device_sn":"APCGQ80E20300662","attributes":{}},"trace_id":"540b02b85b05ec6cb8aebe33b279e276"}

For now I've no time to dig further (e.g. playing with the request parameters) but atleast we know that for the SB2 there's nothing sent out with an "empty" request.

thomluther, 2024-07-14:
OK thanks, yes that is what I have seen also in other system exports.
I think the query needs in the list specific attributes to query.
My hope was that this endpoint may reveal other device specific settings, like light, temperature unit or others that are possible with the app but not available with any of the existing endpoints

thomluther, 2024-09-12:
@Lenoirio have you got some time to explore further with various attributes you may think of?
Temperature, unit, energy, state, type, anything that might match to data that you see in the App?
It will also have to be tested against smart meter and smart plugs. I hope this can be used to query and set the smart plug switch state or maybe other attributes that are supported.
Unfortunately there is no error message indicating which attribute may be required/supported...

Krosty0712, 2024-09-19:
Can anyone help me figure out what to do for the two print commands in front of it?

To read out my SB2, I used the following: https://github.com/thomluther/anker-solix-api
And for every error I received, I searched the internet.
I had to install some things.

Is there a guide or someone who can help me with a meeting?
Or is it possible not to give @thomluther the account information publicly so that he can set that up?

Krosty0712, 2024-09-25:
New Version of Integration has the Export Service. Really Great move.  

[petxx_2024-09-25_1407.zip](https://github.com/user-attachments/files/17131178/petxx_2024-09-25_1407.zip)

thomluther, 2025-04-10:
The only attribute list item that returned an output so far for devices I could test (SB2 AC) was:
- rssi

This may depend on the queried device and my hope was to the attributes for the LED, temperature etc, but not luck with the strings I tested so far. Maybe there is more with the smart plugs or other device types.

thomluther, 2025-04-11:
Nothing new in the export. That may change when the cloud is updated for the SB3 support in May...

Krosty0712, 2025-04-11:
I made me an appointment for end may. 
You made the process so easy, thats no big Deal to deliver Exports. 

Habe a nice Weekend

AchimTheobald, 2025-04-11:
rssi = possibly only the quality of the received WLAN signal?

[image]

thomluther, 2025-04-11:
> rssi = possibly only the quality of the received WLAN signal?

Yes, the meaning is clear, but this value is also available in other queries. I'm looking for new attributes and values that are not available with the other queries. I listed this here only because that was the only attribute name that showed content in the response.

AchimTheobald, 2025-04-11:
Here is a new Export.
Solarbank 2 Pro / 4 non-Anker Panels / 6 SmartPlug (5 active in WLAN)

[Einstellungen_2025-04-11_1219.zip](https://github.com/user-attachments/files/19703823/Einstellungen_2025-04-11_1219.zip)

ovflowd, 2025-04-19:
@thomluther here's the dump of the `get_device_attrs` for two different Smart Plugs (First one consuming 28W and has no Timing Rules):

**cURL Requests**

```bash
curl --request POST \
  --url https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_attrs \
  --compressed \
  --header 'accept-encoding: gzip' \
  --header 'app-name: anker_power' \
  --header 'app-version: 3.6.0' \
  --header 'content-type: text/plain' \
  --header 'country: DE' \
  --header 'gtoken: 2111660b8bd394742bc20deec59b9a02' \
  --header 'host: ankerpower-api-eu.anker.com' \
  --header 'language: en' \
  --header 'model-type: PHONE' \
  --header 'os-type: android' \
  --header 'os-version: 15' \
  --header 'phone-model: A015' \
  --header 'timezone: Europe/Berlin' \
  --header 'x-app-key: ' \
  --header 'x-auth-token: fd2368d6afb67bbca3e67bb334866bbe39851b819e5c5ff5' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: 487c4d59721f02c22112c46c1ea038bf' \
  --header 'x-replay-info: replay' \
  --header 'x-request-once: cba657a9e6b9da45d89658b6a281ff97' \
  --header 'x-request-ts: 1745103679' \
  --header 'x-signature: d93f441a9951cc4f4069f13d21248764619e42336315dcc93022f93fddcc7cbb' \
  --data 'MTc0NTEwMzY3OTIyMjAwMIUJ7p7z1Zk4ra+8lqNwWY8l//5+uEyzztV+MglhAvvv3/xuqxYyn9heRVIRHje482N8BuZEVklJEnernTGFyEY='

curl --request POST \
  --url https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_attrs \
  --compressed \
  --header 'accept-encoding: gzip' \
  --header 'app-name: anker_power' \
  --header 'app-version: 3.6.0' \
  --header 'content-type: text/plain' \
  --header 'country: DE' \
  --header 'gtoken: 2111660b8bd394742bc20deec59b9a02' \
  --header 'host: ankerpower-api-eu.anker.com' \
  --header 'language: en' \
  --header 'model-type: PHONE' \
  --header 'os-type: android' \
  --header 'os-version: 15' \
  --header 'phone-model: A015' \
  --header 'timezone: Europe/Berlin' \
  --header 'x-app-key: ' \
  --header 'x-auth-token: fd2368d6afb67bbca3e67bb334866bbe39851b819e5c5ff5' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: 487c4d59721f02c22112c46c1ea038bf' \
  --header 'x-replay-info: replay' \
  --header 'x-request-once: ad597773156bdb726f38a59ee003e38f' \
  --header 'x-request-ts: 1745103679' \
  --header 'x-signature: 7ddec208438f4382885c8ca874b3cebaa607097eb6470c32603a3ba2a0094cb7' \
  --data 'MTc0NTEwMzY3OTIxNjAwMHuJXdsrGwPbM7g+I43sYCdi7GboCpuyAbJl2hIybPnx62jThXO/WCrvoiS61vQILIwxD56Z9yrnaijRWZ34eSTUCdM5GNF6mhqwfUCjS5WEKJcmu2CVP7FqmUISLpxmww=='
```

**Response Bodies:**

```json
{
  "signature": "98444e7105fbdc29a4e969d8b24d6e757ea9c495c6c8158cefbecdaf2cbf72b0",
  "msg": "success!",
  "code": 0,
  "trace_id": "9d0bd9c0a328bcfde9e60f5a33284977",
  "data": "ybF4Sg9vCKt3V75DnvTddszD0m6qJH4ZhjcN4DQC2RMlvqCthrJ1qK4DLvJ7l8jwYYbMooaE2QXmtfGebt8d1wfzCnl3XqLhpM8dxXX7ghw="
}
```

```json
{
  "data": "vneZeUkS5MV4mSsCWYbEX7qRo3+P5/vI+o725DF71k5vFEg4ZPKnMR7sR70q0EjFIVGGNVyXxK81H/OHhqpswjpNYnfIVyr31LdD6mWSq5Ulh4LI4eaM9V/D4loabYLY",
  "code": 0,
  "msg": "success!",
  "trace_id": "47c5741913f87d87db96ca82ae466094",
  "signature": "e9fb33c14a48c425cfc843457c3ac8c2d37ab8141e0028e491452e8794095860"
}
```

ovflowd, 2025-04-19:
@thomluther here's the dump of the `get_device_attrs` for two different Smart Plugs (First one consuming 28W and has no Timing Rules):

**cURL Requests**

```bash
curl --request POST \
  --url https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_attrs \
  --compressed \
  --header 'accept-encoding: gzip' \
  --header 'app-name: anker_power' \
  --header 'app-version: 3.6.0' \
  --header 'content-type: text/plain' \
  --header 'gtoken: GTOKEN' \
  --header 'host: ankerpower-api-eu.anker.com' \
  --header 'language: en' \
  --header 'x-app-key: ' \
  --header 'x-auth-token: TOKEN' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: 487c4d59721f02c22112c46c1ea038bf' \
  --header 'x-replay-info: replay' \
  --header 'x-request-once: cba657a9e6b9da45d89658b6a281ff97' \
  --header 'x-request-ts: 1745103679' \
  --header 'x-signature: d93f441a9951cc4f4069f13d21248764619e42336315dcc93022f93fddcc7cbb' \
  --data 'MTc0NTEwMzY3OTIyMjAwMIUJ7p7z1Zk4ra+8lqNwWY8l//5+uEyzztV+MglhAvvv3/xuqxYyn9heRVIRHje482N8BuZEVklJEnernTGFyEY='

curl --request POST \
  --url https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_attrs \
  --compressed \
  --header 'accept-encoding: gzip' \
  --header 'app-name: anker_power' \
  --header 'app-version: 3.6.0' \
  --header 'content-type: text/plain' \
  --header 'gtoken: GTOEKN' \
  --header 'host: ankerpower-api-eu.anker.com' \
  --header 'language: en' \
  --header 'x-app-key: ' \
  --header 'x-auth-token: TOKEN' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: 487c4d59721f02c22112c46c1ea038bf' \
  --header 'x-replay-info: replay' \
  --header 'x-request-once: ad597773156bdb726f38a59ee003e38f' \
  --header 'x-request-ts: 1745103679' \
  --header 'x-signature: 7ddec208438f4382885c8ca874b3cebaa607097eb6470c32603a3ba2a0094cb7' \
  --data 'MTc0NTEwMzY3OTIxNjAwMHuJXdsrGwPbM7g+I43sYCdi7GboCpuyAbJl2hIybPnx62jThXO/WCrvoiS61vQILIwxD56Z9yrnaijRWZ34eSTUCdM5GNF6mhqwfUCjS5WEKJcmu2CVP7FqmUISLpxmww=='
```

**Response Bodies:**

```json
{
  "signature": "98444e7105fbdc29a4e969d8b24d6e757ea9c495c6c8158cefbecdaf2cbf72b0",
  "msg": "success!",
  "code": 0,
  "trace_id": "9d0bd9c0a328bcfde9e60f5a33284977",
  "data": "ybF4Sg9vCKt3V75DnvTddszD0m6qJH4ZhjcN4DQC2RMlvqCthrJ1qK4DLvJ7l8jwYYbMooaE2QXmtfGebt8d1wfzCnl3XqLhpM8dxXX7ghw="
}
```

```json
{
  "data": "vneZeUkS5MV4mSsCWYbEX7qRo3+P5/vI+o725DF71k5vFEg4ZPKnMR7sR70q0EjFIVGGNVyXxK81H/OHhqpswjpNYnfIVyr31LdD6mWSq5Ulh4LI4eaM9V/D4loabYLY",
  "code": 0,
  "msg": "success!",
  "trace_id": "47c5741913f87d87db96ca82ae466094",
  "signature": "e9fb33c14a48c425cfc843457c3ac8c2d37ab8141e0028e491452e8794095860"
}
```

thomluther, 2025-04-20:
@ovflowd 
The data cannot be decrypted since the payload encryption rules are unknown.
But it shows that these endpoints are used for the plugs. You need to find out with queries which attributes return values. I only found the 'rssi' attribute, but for sure there are more.

ovflowd, 2025-04-20:
I posted this here https://github.com/thomluther/ha-anker-solix/issues/150#issuecomment-2816899085 -- I do have the feeling the timing and turn/on/off actions on the smart plugs are done completely locally.

> @ovflowd
The data cannot be decrypted since the payload encryption rules are unknown.

I think I found out how the encryption works!

ovflowd, 2025-04-20:
Yup, I was able to find out the algorithm and the entire Key Exchange Process, which ironically is unencrypted.

ovflowd, 2025-04-20:
#### **Description**

The Anker API uses an encrypted communication mechanism for request/response payloads, leveraging **Elliptic Curve Diffie-Hellman (ECDH)** key exchange followed by **AES-based symmetric encryption**. This issue documents the findings, protocol flow, and provides a JavaScript utility to automate the handshake and decryption process.

---

#### **Protocol Summary**

1. **Key Exchange**
   - Endpoint: `POST /openapi/oauth/key/exchange`
   - Client sends a **base64-encoded ECDH public key** via `client_public_key`.
   - Server responds with:
     - `server_public_key` (base64-encoded)
     - `signature` (for integrity)

2. **Shared Secret Derivation**
   ```js
   const sharedSecret = crypto.diffieHellman({
     privateKey: clientPrivateKey,
     publicKey: serverPublicKey
   });
   ```

3. **Symmetric Key Derivation (HKDF)**
   ```js
   const aesKey = crypto.hkdfSync('sha256', null, sharedSecret, Buffer.from('ecdh handshake'), 32);
   ```

4. **Payload Encryption**
   - Encrypted request/response bodies are base64-encoded AES ciphertexts.
   - Likely encryption mode: **AES-CBC** (with IV prepended to the ciphertext).

---

#### **Example Key Exchange Request & Response**

##### ➤ `POST /openapi/oauth/key/exchange`

```bash
curl --request POST \
  --url https://ankerpower-api-eu.anker.com/openapi/oauth/key/exchange \
  --compressed \
  --header 'content-type: application/json' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: e5df912e622f1f72c733675c529eabe7' \
  --header 'x-auth-token: ab667bcb***************' \
  --data '{"client_public_key":"MTc0NTE4Mjc4NzY2MDAwMKGQdMl8bLin5dm1yd4xkAB+0pyhFM8HeSb2hQjUXtPqU1gK/u35mmq/cgezkCDkQ9cmwgAFdElvyvUOAzug6MpkImTKzmE9n+EbCQvXwzYfX9c9JSz0QehQDVrorADYfRZcnFae3fNoXlkLTx+Oh+g1Ak08LG3JFpfvaDdUpiEOef//j6CC0Mf0L2EhlgfzYg=="}'
```

##### Response:

```json
{
  "code": 0,
  "data": {
    "server_public_key": "vWQrlFafQ83NEIpBZ53fuQPHqKheC0oW8qFX52zb4hi96Mha7lgyUXMgzVkJhKjkw4b5O4SFMwi3BLQ+yqYRjAjd74iC+i9FTRwU9eQ3SQ4ajsBkcvdW/zGm+mtBD7oFrERyjyMeomS8Z2kgFcnG0YAlscUZqd/BFn1Z4tV/qCOpj3nheH1OhI4lli0AMt1IWH8b7Zxx/lTN2CRkgtU5Fw==",
    "signature": "1def83500ada8f5a7dec15c20301aec6ff1b5d16b831933fe53ac8a2e417707d"
  },
  "msg": "success!",
  "signature": "1def83500ada8f5a7dec15c20301aec6ff1b5d16b831933fe53ac8a2e417707d"
}
```

---

#### **Example Encrypted Device Attribute Request**

##### ➤ `POST /power_service/v1/app/device/get_device_attrs`

```bash
curl --request POST \
  --url https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_attrs \
  --compressed \
  --header 'content-type: text/plain' \
  --header 'app-name: anker_power' \
  --header 'app-version: 3.6.0' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: 487c4d59721f02c22112c46c1ea038bf' \
  --header 'x-auth-token: ab667bcb***************' \
  --header 'x-request-once: cba657a9e6b9da45d89658b6a281ff97' \
  --header 'x-request-ts: 1745103679' \
  --header 'x-signature: d93f441a9951cc4f4069f13d21248764619e42336315dcc93022f93fddcc7cbb' \
  --data 'MTc0NTEwMzY3OTIyMjAwMIUJ7p7z1Zk4ra+8lqNwWY8l//5+uEyzztV+MglhAvvv3/xuqxYyn9heRVIRHje482N8BuZEVklJEnernTGFyEY='
```

##### Response:

```json
{
  "signature": "98444e7105fbdc29a4e969d8b24d6e757ea9c495c6c8158cefbecdaf2cbf72b0",
  "msg": "success!",
  "code": 0,
  "trace_id": "9d0bd9c0a328bcfde9e60f5a33284977",
  "data": "ybF4Sg9vCKt3V75DnvTddszD0m6qJH4ZhjcN4DQC2RMlvqCthrJ1qK4DLvJ7l8jwYYbMooaE2QXmtfGebt8d1wfzCnl3XqLhpM8dxXX7ghw="
}
```

---

#### **🧰 Node.js Script for Decryption**

```js
const axios = require('axios');
const crypto = require('crypto');

// Step 1: Generate ECDH keypair
const ecdh = crypto.createECDH('prime256v1');
ecdh.generateKeys();
const clientPublicKey = ecdh.getPublicKey();
const clientPublicKeyBase64 = clientPublicKey.toString('base64');

// Step 2: Perform key exchange
async function performKeyExchange() {
  const res = await axios.post('https://ankerpower-api-eu.anker.com/openapi/oauth/key/exchange', {
    client_public_key: clientPublicKeyBase64,
  }, {
    headers: {
      'content-type': 'application/json',
      'x-encryption-info': 'algo_ecdh',
      'x-key-ident': 'your-key-ident',
      'x-auth-token': 'ab667bcb***************'
    }
  });

  const serverPubKeyBase64 = res.data.data.server_public_key;
  const serverPublicKey = Buffer.from(serverPubKeyBase64, 'base64');

  const sharedSecret = ecdh.computeSecret(serverPublicKey);
  const aesKey = crypto.hkdfSync('sha256', null, sharedSecret, Buffer.from('ecdh handshake'), 32);
  return aesKey;
}

// Step 3: Decrypt encrypted payload
function decryptPayload(base64Input, aesKey) {
  const data = Buffer.from(base64Input, 'base64');
  const iv = data.subarray(0, 16);
  const ciphertext = data.subarray(16);

  const decipher = crypto.createDecipheriv('aes-256-cbc', aesKey, iv);
  const decrypted = Buffer.concat([decipher.update(ciphertext), decipher.final()]);
  return decrypted.toString();
}

(async () => {
  const aesKey = await performKeyExchange();

  const encrypted = 'ybF4Sg9vCKt3V75DnvTddszD0m6qJH4ZhjcN4DQC2RMlvqCthrJ1qK4DLvJ7l8jwYYbMooaE2QXmtfGebt8d1wfzCnl3XqLhpM8dxXX7ghw=';
  const plaintext = decryptPayload(encrypted, aesKey);

  console.log('[Decrypted]:', plaintext);
})();
```

---

#### **Acceptance Criteria**

- [x] Documented ECDH + AES encryption strategy
- [x] JavaScript (Node.js) utility to:
  - Generate keypair
  - Perform key exchange
  - Derive AES key
  - Decrypt payloads
- [x] Example `curl` requests and expected responses included

ovflowd, 2025-04-20:
@thomluther would any of the above help you? (I used ChatGPT to summarise the discoveries)

thomluther, 2025-04-21:
@ovflowd Yes, this is useful stuff, but I would need that converted to python. Especially how the header fields must be used and encoded for this to work...

I'd like to add the payload encryption option also in the library since the encryption may be enforced easily and would break the integration.

So with your decryption routine, does that work as well for App payload sniffing since each client/server auth typically has its own ECDH keys?

If you can use it to decrypt App payload, it may be easier for users to discover required parameters and options as used by the App and actually see which parts can be driven via the Api and which not.

One point of interest are the smart plugs, which might or might not be manageable via the Api.

thomluther, 2025-04-21:
What is actually the 'x-key-ident', I don't see how this is generated?

ovflowd, 2025-04-21:
## Anker API Encryption Analysis - Additional Findings

### Key Structure Analysis

1. **Client Public Key Structure**
   - Total length: 160 bytes
   - First 16 bytes: Timestamp (milliseconds since epoch)
   - Byte 17: Marker byte (observed values: 0x1f, 0x11)
   - Remaining 143 bytes: Key data

2. **Server Public Key Structure**
   - Total length: 160 bytes
   - First 96 bytes appear to contain the key material
   - Remaining 64 bytes might be additional metadata/signature

### Required Headers

For API requests, only these headers are strictly required:
```http
content-type: application/json
x-app-key: [empty is fine]
x-auth-token: [your auth token]
x-auth-ts: [timestamp]
x-encryption-info: algo_ecdh
x-key-ident: [generated key ident]
x-replay-info: replay
x-request-once: [generated request once]
x-request-ts: [timestamp]
x-signature: [generated signature]
```

### Observed Request Patterns

1. **Key Exchange Request**
```bash
curl --request POST \
  --url https://ankerpower-api-eu.anker.com/openapi/oauth/key/exchange \
  --compressed \
  --header 'content-type: application/json' \
  --header 'x-app-key: ' \
  --header 'x-auth-token: REMOVED' \
  --header 'x-auth-ts: 1745262550' \
  --header 'x-encryption-info: algo_ecdh' \
  --header 'x-key-ident: ae2bd8a35c80fc80f44e532ff4e68624' \
  --header 'x-replay-info: replay' \
  --header 'x-request-once: 12c5e998035bd02a83b2fc6437b20ef1' \
  --header 'x-request-ts: 1745262550' \
  --header 'x-signature: e16a13c612d95ba43cc9aef5da52a7b2d83f2e502260a42476a83f40ec4d2ef4' \
  --data '{"client_public_key":"MTc0NTI2MjU1MDcxNTAwMB+A1vNJpiodJZAUuzhOXfu1nKtF+H6B0ADJl2ksUuDx0KS9NAQXor11MotpfkC9SzSK+ktVVpGJs6RNQXwHlxvk4HE5eI2m6gj81dRrH+vVZDTxLhwWflG8H4o4xil/6qiB6Ik3jbkip6gLQrfCYu2jsBxNVdi5DgwjLA7vqBM0Ql07MeKXbjplUxq0p558ww=="}'
```

2. **Key Exchange Response**
```json
{
  "code": 0,
  "data": {
    "server_public_key": "d2G/5WLmZ67vHOs2Hk879TyRCvKsDo+DwNZ83m0EY0YGwFSisf5fYAzPFCtHg0pnxIPAt+b5UveqwgfqyXI64bNStRe+v1b0zsw56QWMRV/7vWVn1E9Usa1n1XeO9Qi1me+WJ41zG8H2da5AzSZy9tLmdzzT+Jer/Va3KExYku7wTPIgZcPogYiieD1cLIBHxTJd5DqxIQoJlVX7IPCVdA==",
    "signature": "8ac1ddd8484e81abe5b97e207894eb8f9d7aecf5e704befa6bc01da6918e80f3"
  },
  "msg": "success!",
  "signature": "8ac1ddd8484e81abe5b97e207894eb8f9d7aecf5e704befa6bc01da6918e80f3"
}
```

### Python Implementation

Here's a Python script that implements the key exchange and decryption:

```python
import base64
[... 168 lines cut ...]
### Notes on Implementation

1. **Key Format**: The client public key is constructed with:
   - 16-byte timestamp (milliseconds)
   - 1-byte marker (0x1f)
   - Raw public key data

2. **Server Response**: The server's public key appears to have additional metadata/signature beyond the actual key material. The current implementation assumes the key is in the first 96 bytes, but this may need adjustment.

3. **Headers**: The implementation includes only the required headers for key exchange and API requests. The values for these headers are generated using timestamp and random data, following the observed patterns from the app.

### Testing Notes

1. The script can be used to:
   - Generate properly formatted client keys
   - Perform key exchange with the server
   - Attempt to decrypt encrypted payloads

2. To test with actual API responses:
   - Replace `YOUR_AUTH_TOKEN` with a valid token
   - Use the `decrypt_payload` method with base64-encoded responses

3. The key extraction and decryption logic may need fine-tuning based on actual server responses and payload formats.

This implementation is a starting point and may need adjustments based on further analysis of the protocol and actual API behavior.

ovflowd, 2025-04-21:
> What is actually the 'x-key-ident', I don't see how this is generated?

The `x-key-ident`, `x-request-once`, and `x-signature` generation algorithms are still under investigation. From the observed requests, we can see they follow these patterns:

1. `x-key-ident`: 32-character hex string that changes per request
   - Example: `ae2bd8a35c80fc80f44e532ff4e68624`
   - Likely involves timestamp and some random data
   - Possibly uses MD5 hashing

2. `x-signature`: 64-character hex string
   - Example: `e16a13c612d95ba43cc9aef5da52a7b2d83f2e502260a42476a83f40ec4d2ef4`
   - Likely SHA-256 based on length
   - Probably includes timestamp, request-once, and request body

I've shared a gist with additional debugging code that helps analyze the key structures and attempts different approaches to extract and validate the ECDH keys: https://gist.github.com/ovflowd/7fa70639366fae6c667769c62fb5df49

thomluther, 2025-05-03:
Hi @ovflowd 
Can you post or link to a working Code for the key exchange?
You example here are incomplete and the python example does not work.

ovflowd, 2025-05-03:
Apologies, @thomluther I was extremely busy with work these days, let me reply on the other thread!