thomluther/anker-solix-api #109: Explore how the Solarbank 2 usage mode can be displayed and changed

thomluther, 2024-06-25:
Solarbank 2 supports different usage modes:
- Manual mode via the schedule
- AI mode via the smartreader

How is the mode setting displayed in the API?
Is that via the new device_attrs endpoints?
How can the mode be changed via the API?

Lenoirio, 2024-07-13:
I'm not directly using your API but I'm very thankful for the work you did here because it helped me getting first things rolling. I use a self-written local solution written in Rust.

Anyway, regarding the Solarbank2.
The first info I can contribute here is that get_device_param needs to specify param_type as 6  rather than the default 4 that I saw in your code.  Otherwise you'll get the ranges attribute with a null-value.

The response with the 6 looks like this:
{"code":0,"msg":"success!","data":{"param_data":"{\"mode_type\":3,\"custom_rate_plan\":[{\"index\":0,\"week\":[0,1,2,3,4,5,6],\"ranges\":[{\"start_time\":\"00:00\",\"end_time\":\"24:00\",\"power\":130}]}],\"blend_plan\":null,\"default_home_load\":200,\"max_load\":800,\"min_load\":0,\"step\":10}"},"trace_id":"77f2f0f08a1eb57b600e42af3c1f29ea"}

I actually have only one entry for every day, whole day with power of 130W in this example.  
default_home_load is what they tell the user of the app: "any time-range not defined will be using 200W", 800W are max-output (the app allows selecting 600W or 800W IIRC which makes sense as I live in Germany)

thomluther, 2024-07-13:
@Lenoirio That is very useful information. There is another endpoint get_device_load that does not need a param_type.
Have you checked this? I got once system export with a schedule defined, and there I see also the time slot but in the structure of SB1, so no weekdays etc.

That means to fully support SB2 schedules, the api would have to distinguish which request parameter to use.
My hope is that at least the get_device_load would return the structure as supported by the provided device SN ?

Lenoirio, 2024-07-13:
@thomluther Yes, I did and the response is most likely not what you were hoping for (as ranges is null)

 {"code":0,"msg":"success!","data":{"site_id":"bafa5b83-3a86-4050-b35e-d1906daa18b5","home_load_data":"{\"ranges\":null,\"min_load\":150,\"max_load\":800,\"step\":0,\"is_charge_priority\":0,\"default_charge_priority\":0,\"is_zero_output_tips\":1,\"display_advanced_mode\":0,\"advanced_mode_min_load\":0}","current_home_load":"200W","parallel_home_load":"","parallel_display":false},"trace_id":"dcc78c9c87c33e58ce3cc70b98d6f4cf"}

BTW: I also did a successful set_device_param with a modified structure (I changed the power-value) and it was accepted and my SolarBank2 changed the output-power accordingly.

thomluther, 2024-07-13:
> @thomluther Yes, I did and the response is most likely not what you were hoping for (as ranges is null) [...]

OK, indeed. Good to know that at least the device_parm endpoints can be used.
That would mean I have to restructure the data collection routines and use proper call depending on solarbank generation.

I would really be helpful, if you can share your test structures from getting and setting various schedule options.
I will be helpfull to understand the minimum structure for the set in case no schedule is defined yet.
I will have to build a helper for SB2 schedule anyway, this cannot be mixed with the SB1 helper to modify schedules which is damned  complex already...
SB2 should be easier, since no dual solarbank setups and less interval parameters...but will still needs lots of work and lots of examples to develop without being able to test on my own

thomluther, 2024-07-13:
@Lenoirio 
Any idea what "mode_type":3, means?
Is that an indicator for usage mode (custom or ai?) Or did you find another response to see the home load mode and change that eventually?
If that is part of the schedule structure, it could be handled like other parameter changes of the schedule routines.
And what means "blend_plan":null ?

thomluther, 2024-07-13:
@Lenoirio have you found what the new get and set device_attrs is good for? See #107

Lenoirio, 2024-07-13:
I got my SB2 on last thursday, so I still need some time to get everything up. E.g. the smartmeter needs still to be installed (after our vacation though).

I'd also assume that the mode_type could be related to AI or manual mode.  I quickly did a schedule with more values, looks quite straightforward. The UI of the App blocks the user from creating double entries for same days or overlaping hours.
The following is a setup for  Sunday/Saturday  00-24 110W    Monday-Firday  00-08 90W  08-22 120W 22-24 90W
(they start the week with 0=Sunday)

 {"mode_type":3,"custom_rate_plan":[{"index":0,"week":[0,6],"ranges":[{"start_time":"00:00","end_time":"24:00","power":110}]},{"index":1,"week":[1,2,3,4,5],"ranges":[{"start_time":"00:00","end_time":"08:00","power":90},{"start_time":"08:00","end_time":"22:00","power":120},{"start_time":"22:00","end_time":"24:00","power":90}]}],"blend_plan":null,"default_home_load":200,"max_load":800,"min_load":0,"step":10}

==========
Set_device_param:
#[derive(Debug, Serialize)]
struct SetParamRequest {
    site_id: String,
    param_type: String,
    cmd: i32,
    param_data: String
}
Here param_type is also "6" . Cmd is 17
param_data is the JSON structure of the schedule as String (like the one you find under data->param_data  of the get-request)

I will create the structs needed to holde the JSON-content for the schedules in the next days. When I'm done I'll provide it here (it's Rust but quite readable).

I'll give the get device_attrs a try tomorrow.

thomluther, 2024-07-14:
Manual User mode is mode_type = 3
AI user mode is mode_type = 1 in the schedule structure.
[See examples here](https://github.com/thomluther/anker-solix-api/issues/108#issuecomment-2227297453)

thomluther, 2024-07-29:
Hi @Lenoirio @nicolinuxfr @stephan-l 

It seems there is a new App version 2.8.0 which supports another power usage mode for the smart plugs that will be released soon.
[image]

Can someone switch to that mode in the App and run a System export using the main account please?
I need to know the new mode number to consider this as well in the Api  library and HA Integration

nicolinuxfr, 2024-07-29:
You have to own their plug in order to activate this mode, I'm afraid I won't be able to help…
[image]

stephan-l, 2024-07-29:
I tried it too but no luck without the plug. I got no access to one.
There is also another glitch alert trying this. If you return it stays on custom mode instead of intelligent mode.

thomluther, 2024-07-29:
OK, at least a confirmation that this mode should not be configurable while no Anker plugs installed in the system
Since automode is 1 and custom mode is 3, I assume for now this smartplug mode will be 2
I will prepare support for this and if it turns out to be a different number I just need to change the number definition...

thomluther, 2024-07-29:
> I tried it too but no luck without the plug. I got no access to one. There is also another glitch alert trying this. If you return it stays on custom mode instead of intelligent mode.

That is indeed a bug. In the export of @nicolinuxfr there is still mode 1 active. So if that is changed to 3 it must happen when you cancel or leave that screen without selecting a plug.
But could also be on purpose, because you started the switch to the smartplug mode, but then cancel the switch, so maybe the App defaults back to custom mode as this is always available. Better would be to cancel without an active mode change of course...

thomluther, 2024-08-09:
Full schedule modification is now supported with [release 2.1.0](https://github.com/thomluther/anker-solix-api/releases/tag/v2.1.0)
Anker smart plug support will be tracked with #124

closed 2024-08-09 by thomluther