thomluther/anker-solix-api #114: Understand and work around for the 0 W reponses for Solarbank 2 and smartmeter

thomluther, 2024-06-29:
Solarbank 2 changed the data update frequency to the cloud. For Solarbank 1 it was min 60 sec update interval.
The same interval is used by the App to refresh the data on the home screen.

Solarbank 2 home screen update data much more often. Testing needs to be done by the community to answer following questions:

1. What is the min. interval the data can change on home screen?
2. When using the solarbank monitor with a short query interval for the sites, what is the maximum lag of the Data timestamp when solarbank data was sent to the cloud? Need to understand how often this is updated in the cloud.
3. When there are False 0W values returned, how long does that condition exist?

Following is my assumption on the cloud changes for Solarbank 2.
The devices can now send data much more often to the cloud. Lets assume every 3-5 seconds.
The cloud needs a while to receive and provide updated data upon queries from the App. I assume however this is just a short period for 1-2 seconds only.
When the App however queries the cloud data in such an update interval, the cloud response indicates with the new display_data field whether the data is valid or not to the app. In this case, the App does not refresh the home page and skips it for the next request.

So it depends on how often the condition exists in the cloud, how long it is, and how often the data is queried by clients.
In worst casees the client always hits such an update period of the cloud and receives false data for multiple queries.

If that is the case, a simple work around can be implemented in the update sites method to re-query the cloud up to 5 times with 1 sec. delay until a valid response is received for the update_sites run.
Since the HA Integration uses slower intervals (min 30 sec) to limit the number of Api requests around the clock, each update_sites run should ensure that valid data can be provided.
In worst case when all retry attempts fail, the HA integration will need to make use of the false display data and render the dependent entities to unknown until valid data is provided again on subsequent intervals.

This may break existing automations and scripts if not coded properly to expect unavailable values when calculating with number entities.

Skipping the false data without entity update is not a recommended approach, since that will cause stale and wrong data tracking in HA for unknown periods, without having an evidence in history whether data is valid or not.

thomluther, 2024-06-29:
I updated the solarbank monitor tool and added support for the smartreader info structure, so it will now be created as separate device in the Api devices cache.
The solarbank monitor will show it and also a new metrics for valid device data.
If Solarbank display data is false, data for the solarbank and smartmeter devices are considered invalid, likewise the whole scene_info data in the system cache.
Update_sites routine will now requery up to 5 times when invalid response data is received.
This is shown in the solarbank monitor:
`Cloud-Updated  : 2024-06-22 16:37:52     Valid Data    : YES (Requeries: 0)`

This should be usable to track in realtime whether 5 retries are typically sufficient to ensure valid data per query interval.
The monitor refresh period can also be lowered to 5 sec to compare the query time with the cloud update time and identify how much the cloud data can age. This will give a hint about the cloud update interval of Solarbank 2.

lumeenous, 2024-06-30:
I noticed that in Home Assistant the 0 W responses occur for approx. 5 min, then one real response is transmitted and so on. 
However, when I open the Anker app with my main account on my phone, the 0W responses in Home Assistant are gone. There are just real values until approx. 1 min after the phone app is closed. Then the 0 W responses starting again. 
I assume the App sends a kind of request to the solarbank 2, to send data much more often (every second or so) as the momentary values in the app are changing almost in realtime after they are initially loaded.
Im afraid I can't specify my observations further, as I have not enough programming knowledge to run the python API myself, but hope it will give some hints for the right direction.

nicolinuxfr, 2024-07-01:
> However, when I open the Anker app with my main account on my phone, the 0W responses in Home Assistant are gone. 

I have noted the same thing, anecdotally too. If I keep the Anker app open on my phone, I never have 0 values. After the app goes to the background, I start to have 0 values again. 

[image]

thomluther, 2024-07-01:
> I assume the App sends a kind of request to the solarbank 2, to send data much more often (every second or so) as the momentary values in the app are changing almost in realtime after they are initially loaded.

If that is really the case, then we have an issue here with the Api since it cannot trigger the Solarbank 2 to send continous data.
If possible, please test the latest solarbank monitor tool, it utilizes the requery capability if there is invalid data.
If that does not help to reduce the invalid responses while the App is closed, I have no idea how they could be avoided.
Regular Api request via the integration will then always show invalid/unusable data at indefinite intervals like it does now.

nicolinuxfr, 2024-07-01:
> If possible, please test the latest solarbank monitor tool, it utilizes the requery capability if there is invalid data.

How should we proceed exactly ? Wait until the data disappears inside Home Assistant and then use it ?

thomluther, 2024-07-01:
I fixed that now, used the wrong quotes in fstring.
When you see it in HA showing 0, you can start the monitor tool and watch whether this received data.
If you can use a different account for the monitor tool, you can leave it up running. I would assume, that both accounts get same responses for the Scene Info query, so either both are valid or invalid, except the invalid state is just very short and both query at different times.

thomluther, 2024-07-01:
Another assumption how Solarbank 2 provides data to cloud:
- When App is starting the homepage, the solarbank is triggered to send data very often (every 3 sec?). This would be similar behavior as watching the device details real time data, where the Solarbank is triggered to send every second to the MQTT cloud server when history is watched in App. (Therefore the device history always starts from empty diagramm) 
- When App is not watching the home page, they solarbank provides cloud updates only every 5 min? This could have been implemented to:
  - reduce Api traffic for better scaling with more devices
  - Still sufficient to do all the energy statistics in the cloud (Energy is not done by devices). 
- The sent 5 min refresh is just valid data for 1-2 min, then responses indicate that data is state with the new is_display_data field?

However, just receiving a new Watt value every 5 minutes seems to be too less to me to do a proper energy calculation. The energy values could be way off from the reality if there is much fluctuation.

nicolinuxfr, 2024-07-01:
So, I may have burnt my Anker account really quickly… 😬

```
(100053) Anker Api Error: Please try again in 24 hours.
Response Text: {"code":100053,"msg":"Please try again in 24 hours.","trace_id":"29b5a340169fa5a76353abcf0ac765c7"}
<class 'api.errors.RetryExceeded'>: (100053) Anker Api Error: Please try again in 24 hours.
```

Before that, I had no value at all using the monitoring tool : 

[image]

I did not open the app before trying this and I had only 0 in HA.

thomluther, 2024-07-01:
every time you switch an account to another client/device it has to re-authenticate. You can do that up to about 10 times a day, then you get an IP ban. You can restart internet connection on your router to get a new IP, then it will work again.

thomluther, 2024-07-01:
BTW, when you use the same account in HA and in the solarbank monitor, they kick out each other all the time. After 10 times there is the IP ban. When you want to use the HA account in the monitor, you need to disable the Api usage first in the Solix system entity. Then it will consume just 2 re-authentications when you switch forth and back between the monitor tool and HA

thomluther, 2024-07-01:
Also visible from the screenshot:
- The requery work around does not resolve the problem
- 5 sec refresh interval may trigger too many Api requests per minute. So 10 sec interval is probably better

nicolinuxfr, 2024-07-01:
> BTW, when you use the same account in HA and in the solarbank monitor, they kick out each other all the time. After 10 times there is the IP ban. When you want to use the HA account in the monitor, [...]

I forgot about that, indeed. 

> every time you switch an account to another client/device it has to re-authenticate. You can do that up to about 10 times a day, then you get an IP ban. You can restart internet connection on your router to get a new IP, then it will work again.

I only have one fixed IP, but it's a good general tip for others. As for me, I will try again tomorrow if needed.

thomluther, 2024-07-01:
> Also visible from the screenshot:
> 
>     * The requery work around does not resolve the problem
> 
>     * 5 sec refresh interval may trigger too many Api requests per minute. So 10 sec interval is probably better
> 
> 
> That means, at this point, I have no idea how to resolve the data collection issue with Solarbank 2. That could mean we cannot really support it. Or you would see unavailable entities with just a single value about every 5 minutes.

I made another change to the Api:
- Requery up to 3 times with 5 sec delay

This gives longer retry timespan from 5 => 15 seconds, and reduces also the requests per minute to avoid hitting the Api error.

thomluther, 2024-07-02:
From what I have red so far, there is no way to get around the 0 values issue. It appears Anker has completely changed the mechanism how Solarbank 2 is providing data to the api cloud. That seems to be similar to the real time data transfer to the MQTT cloud, which is triggered and started only when device details are accessed through the app. Likewise this applies now to Api cloud. Per default the Solarbank 2 seems to send Data only  about in 5 min intervals. This data will be kept valid in the cloud and then be marked as stale after 60+ seconds if no new data received. Therefore the Api returns valid data only once or twice every 5 minutes interval when the App is not open and triggering the same System for frequent updates. Any immediate requery will not solve the problem, and also shorter or longer intervals will not solve it. When the App is closed, the shortest data refresh intervals will become 5 minutes with Solarbank 2...a major drawback to my opinion. So any history recording or energy calculations from power values will become inaccurate and maybe useless. 
I assume the App trigger to provide frequent cloud updates is made like for MQTT updates and I have no glue how that is done. Someone needs to reverse engineering the App Traffic, but requests and content will be EDHC encrypted, so not possible to decode without knowing the shared secret from App and Server.

I'm afraid the Solarbank 2 users have to live with that limitation and in worst case this mechanism will be implemented also for Solarbank 1 users, breaking any usefull automation capabilities further.
But pretty certainly this change was on purpose to limit cloud data traffic for scale and/or costs.

Question is how the HA integration should deal with invalid responses?

1. Skip entity update until valid data is received?
2. Report unknown state while data is invalid? 

Each approach has pros and cons

nicolinuxfr, 2024-07-02:
I think the first approach is better to monitor the system, not to automate it. But if it's not possible otherwise, I would rather have new correct data each 5 minutes than regular useless data. It would not be a big problem for the energy dashboard, for instance.

Maybe I'm missing something here ?

stephan-l, 2024-07-02:
To clarify: It only shows data as long as your API shows valid data. So guest accounts seem not triggering an update.

thomluther, 2024-07-03:
> To clarify: It only shows data as long as your API shows valid data. So guest accounts seem not triggering an update.

That is a very good point. If you can validate, that using only the App with a shared account does show the same issues (empty values except every 5 minutes for a minute or so). you have a capability to officially address this as an issue to Anker.
The app must support continous reporting for family members as well, otherwise the member account usage is broken for Solarbank 2 system!

From what I believe, shared accounts have access ONLY to the Cloud Api and the system sharing only occurs there, so a shared account can only query a subset of information from the Api cloud. However, he is not able to use the 'normal' interfaces of the App to the device, since they are assigned a single owner account. That means they probably have no capability to trigger frequent data updates with the device like owner accounts can do.

If Anker will change that to be triggerable by shared accounts via the Api Cloud server, it must be via a normal Rest request that can probably be figured out by someone using tracking apps on the mobile.
Other option would be they reduce the standard interval back to 1 minute again and we have the original mechanism as with Solarbank 1, where data do not have to be marked invalid on the cloud.
**So opening Support tickets with Anker for broken member account usage on Solarbank 2 is the only chance to get this changed to the better by Anker.**

For the HA integration there wont't be a situation of reporting valid data only. Either the reported data is stale and potentially incorrect which you will never know (since the are 5 min old or even more).
Or the entity become unavailable, so they have NO data instead of wrong 0 values, which can be considered properly by graphing tools and other integrations or templates. Power Integral sensors for instance may be able to sum last known value if source entity is unavailable (not sure however). But summing up an 5 min or older value that is no longer refreshed is not the way the integration should work.

stephan-l, 2024-07-03:
I just gave it another try to confirm it. Looks like we are right.
I opened a case at Anker to get a feedback.

stephan-l, 2024-07-09:
> I just gave it another try to confirm it. Looks like we are right. I opened a case at Anker to get a feedback.

So far they requested an app log which I provided on 05.07. They are investigating but no further response so far.

lnxbil, 2024-07-12:
Just to understand it correctly: you're all talking about the use of additional non-admin accounts with respect to api calls. Does the main / admin account also has the problem getting the data without zeros? I could live with this ....

thomluther, 2024-07-12:
> Just to understand it correctly: you're all talking about the use of additional non-admin accounts with respect to api calls. Does the main / admin account also has the problem getting the data without zeros? I could live with this ....

Every account has this problem, if you do not open the Anker App with the main account and leave it open.
The app triggers something in the SB2 to send data more frequently to the cloud while it is open. Therefore you see frequent value updates on the home screen.
When you close the app (or only use the shared account in the app), the SB2 will send data only every 5 minutes to the cloud.
The cloud will mark them stale after appr. 1 minute and no longer return values on subsequent queries by any client, hence the 0W values until SB2 sends new values after 5 minutes.

lnxbil, 2024-07-12:
> Every account has this problem, if you do not open the Anker App with the main account and leave it open. The app triggers something in the SB2 to send data more frequently to the cloud while it is [...]

Thank you very much for clarifying this.

philippreissing, 2024-07-13:
> For the HA integration there wont't be a situation of reporting valid data only. Either the reported data is stale and potentially incorrect which you will never know (since the are 5 min old or [...]

@thomluther : would it make sense to implement it like this as a first stage while we for the Anker feedback?

Having the 0 values treated properly should at least allow us to create template sensors to have statistics in home assistant. 
I know that a 5 minute resolution is not great but I assume that’s also the basis for what I see as statistics in the Anker app. So having the same data in my energy dashboard in HA would already be a big improvement.

thomluther, 2024-07-13:
At the end we never know if Anker will change the behavior. I assume not.
I was thinking to provide a config option for the user to decide whether unavailable is preferred for invalid data, or update should be skipped until next valid data. Probably defaulting to unavailable to make users aware of this stale data issue and let them decide whether they want to skip the gaps.

chiefymuc, 2024-07-16:
I've been following the discussion and think this is a very reasonable approach. If it's not too complicated, maybe some kind of "time since last valid data" entity could make sense to make the user aware of the update frequency?

thomluther, 2024-07-16:
> I've been following the discussion and think this is a very reasonable approach. If it's not too complicated, maybe some kind of "time since last valid data" entity could make sense to make the user aware of the update frequency?

There is already the timestamp for data in the cloud. However, that does not always reflect a valid timestamp in the response even for valid data. Potentially I can make an overwrite of this timestamp if it is invalid but valid data received. Although it would not longer reflect the accurate timestamp when data was provided to the cloud, it could always maintain a 'valid' timestamp and have an approximate idea how old the data may be

thomluther, 2024-07-22:
There is a new endpoint I just found in the Apk 2.7.0 package:
/power_service/v1/site/get_site_detail
It seems to return same structure as Scene Info and get_Homepage, but is does not have values for my site.

Can someone test it against SB2 site type whether this returns values when used with the owner account?
It does not work with the shared account.
Maybe this is just another endpoint to be used to get always good responses for SB2?

`print(json.dumps((await myapi.request("post", "/power_service/v1/site/get_site_detail",json={"site_id": siteId, })), indent=2))`

thomluther, 2024-07-23:
Thanks for confirming, so still not really useful this endpoint unfortunately.

Lenoirio, 2024-07-26:
I'm (was?) also suffering from this issue with my local solution.  For a reason that I don't know I'm currently getting values every time I pull the end-point (in my case 60s).   Before I even had cases when I got no updates (display_data=false) for 20min.
Either this is just luck or Anker did something on their side.

Does anyone else see that this changed?

thomluther, 2024-07-26:
I think Anker changed that on the cloud a few days ago when I was releases 2.0.0 of the integration by co-incidence.
I understood they do not send data more often, but obviously the cloud does not mark the data invalid anymore, so it keeps reporting the old data until the device sends new ones. This is the behavior as used for SB1, but SB1 sends them every 60 seconds when awake, otherwise maybe just once an hour (but then the values are 0 anyway)
SB2 likely still sends every 5 minutes, unless triggered in by the mobile App when it sends every ~3 seconds?
Typically the data time is reflected in the SB cloud data timestamp, which was always valid for SB1.
For SB2, this was invalid when data was invalid, but also sometime invalid even if the data was valid. Not sure if Anker fixed the timestamp too for returning always valid timestamp.
The point is that I implemented a work around in the Api, to use the current time when data is valid but timestamp is not. So you cannot say for sure in the monitor whether the timestamp is from the cloud, or just the actual runtime when reading an invalid timestamp.
You can see the cloud provided timestamp only in the direct response of the scene_info query if you want to validate that as well.

Lenoirio, 2024-07-26:
I just checked my log-files and you're right: they repeat the same values for 5min now.  (I don't log their timestap, just my own struct that I write to the database; so no idea yet if it changes or not).
Good thing is: when using a family-account for the Anker-App I finally see values and not just "---".  

For my own needs this is ok now because at the end I only wanted to control the "output-to-home" according to the information that I get from my SmartMeter (the one from the grid operator; not the one from Anker). 
This works pretty reliable now and is good enough so that I don't have to spend the money on an electrician to install the Anker SmartMeter.

thomluther, 2024-07-26:
> I just checked my log-files and you're right: they repeat the same values for 5min now. (I don't log their timestap, just my own struct that I write to the database; so no idea yet if it changes or [...]

Yes. This was the easiest fix for the Family Account bug that many reported back to Anker. Unfortunately they did not reduce the SB2 data update interval back to 1 minute instead like for SB1...

> For my own needs this is ok now because at the end I only wanted to control the "output-to-home" according to the information that I get from my SmartMeter (the one from the grid operator; not the [...]

How do you control the SB2 output according to your other smart meter?
I do that for SB1, but for SB2 it still has to be implemented into a helper method to adjust the new schedule structure accordingly and modify the output value in the proper weekday/daytime structure...

Lenoirio, 2024-07-26:
I just write a schedule with only one entry that is valid for every day and 00:00-24:00  (that's the same structure that you get when you set it up like that in the AnkerApp).
If the value needs to be changed (at the moment I use a 20W granularity and 90%-percentile mean of that last minute), I just send a new schedule.
My SB2 is connected via a Shelly Plug. I use that unit to see the real input from the SB2 to our home-grid and together with the SmartMeter it's easy to calc the current consumption in our house.

thomluther, 2024-07-26:
OK got it, yes that's easy if you do not have to consider various schedule formats.
Individual value change in existing schedule is also easy. However, it becomes fairly complex when (sub) elements of the structure should be changeable via Api, like modifying time slots, creating new ones or modifying weekday assignments.
I have a rough understanding how to do that for SB2 schedule format, while allowing all elements of the structure to be optional.
I think in first step the existing weekday plans need to be evaluated to pick one with most common weekdays as specified in the change. If no weekday matches, a new weekday plan must be generated.
The selected or created weekday plan can then be used in same way as for SB1 to update/create time slots with given times and/or output preset value.

closed 2024-07-28 by thomluther