thomluther/anker-solix-api #115: Collection of Solarbank 2 system exports

thomluther, 2024-06-29:
Please provide your solarbank 2 system export examples here.
You can extract all data with the system_export tool and zip the output files.
You can add a text file to the zip containing info about some specifics when the export was taken, for example the components of your setup.

stephan-l, 2024-06-30:
I just packaged two exports and a system description
[dev2024063019.zip](https://github.com/user-attachments/files/16045693/dev2024063019.zip)

Something I wonder about. Is it possible that the extension battery is not yet visible? The app shows both but in the api there is only 1600 Wh.

nicolinuxfr, 2024-07-01:
Here's my export, I have the same setup as @stephan-l (meaning, one Solarbank 2 E1600Pro, one battery expansion and one smart meter). 

[nicolinux.zip](https://github.com/user-attachments/files/16050265/nicolinux.zip)

Let me know if you need more.

stephan-l, 2024-07-01:
I made another export especially for reference in the status thread:
[dev2024070113.zip](https://github.com/user-attachments/files/16052265/dev2024070113.zip)

thomluther, 2024-07-01:
> Here's my export, I have the same setup as @stephan-l (meaning, one Solarbank 2 E1600Pro, one battery expansion and one smart meter).
> 
> [nicolinux.zip](https://github.com/user-attachments/files/16050265/nicolinux.zip)
> 
> Let me know if you need more.

Thanks for the data. The export looks like being done with the shared account, since many responses have no data.

In general, when providing exports to get a full picture of what is possible with the devices, you need to run the export tool with the main account.

nicolinuxfr, 2024-07-01:
> Thanks for the data. The export looks like being done with the shared account, since many responses have no data.

Indeed, I thought it was already useful that way, but for science™, I can do another export using my main Anker account.

thomluther, 2024-07-01:
> Something I wonder about. Is it possible that the extension battery is not yet visible? The app shows both but in the api there is only 1600 Wh.

It was visible, but not properly considered for the cap and energy calculation of the battery. This is fixed now, also update the solarbank monitor and export tool based  on recent data.
Pls use the latest versions before doing another export.

stephan-l, 2024-07-01:
I made a full dump with the owner account.
The system was running at 140W of battery. 0 from grid and solar.
[ste20240701222735.zip](https://github.com/user-attachments/files/16058819/ste20240701222735.zip)

thomluther, 2024-07-02:
@stephan-l thanks for the new dump. That looks much more useful. First observations:
- OTA update info is not working anymore. It was working for Solarbank 1, not sure what the issue is to query a flag whether a firmware update is available
- get_upgrade_record for the site is empty. I'm pretty sure everybody does firmware upgrades after setup. But maybe they are not reported by the site report if done prior adding the devices to the site. This is no problem since not used in the Api cache data
- new Energy statistics: I need to review them and compare to understand which of the values is returned only as total for the queries interval. Those queries have to be repeated to get specific total statistics for previous day. Maybe the mechanism to query daily statistics for today and yesterday must be changed and optimized to reduce the min. required Api requests.
- Solarbank 2 device load schedule: There is non defined yet. We need exports of various Solarbank 2 schedule definitions. Even if they are not used with AI mode, they can probably be defined and will be included in exports. We need examples of which parameters/fields can be available in the new schedules and how the various weekday configuration structure will be used in the schedule structure. This is most likely pretty much different to Solarbank 1
- device_attrs does not show any data. More testing and exploration of possible/required json parameters for those endpoints is required. see #107. We need to understand which devices are supported by this endpoint and what can be changed with the set_device_attrs endpoint (and how)
- Scene Info is already prepared for a smart_plug device list structure, which is coming probably later this year. So smart plugs may be additional devices to be added to Api caches
- The Solarbank data time stamp is now always valid even if the data itself seems to be valid. This is strange, since my assumption was that correct cloud update time is presented typically when data is valid. So this timestamp cannot really be used to understand how often data is sent to the cloud by Solarbank 2?

thomluther, 2024-07-02:
Here is an actual extract of owner data from export in the monitor after I implemented another workaround for a missing field value in scene_info for the smartmeter wireless_type. I assume now this to be 1 if there is only 1 network in the list for the site and smartmeter reports an online wifi state. In that case it must be using the same network (don't want to make assumptions for anything else, since there could also be smartmeters without solarbank in a site).
```
Solarbank Monitor (refresh 60 s, details refresh 600 s):
Using input source folder: /workspaces/core/anker-solix-api/exports/SB2onwer_1
Sites: 1,  Devices: 2
--------------------------------------------------------------------------------
Device         : Solarbank 2 E1600 Pro   Alias         : Solarbank 2 E1600 Pro
Serialnumber   : OY7WE1AOZVGS1RLX        Admin         : YES
Site ID        : 38cac050-0128-0eff-c9ba-61dabb1f3df1
Wifi SSID      : wifi-network-1         
Wifi state     : Online                  Signal        :   92 %
SW Version     : v1.0.4.4                Auto-Upgrade  : Disabled
Cloud-Updated  : 1970-01-01 00:00:00     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   44 %                  Min SOC       :   10 %
Battery Energy : 1408 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
Solar Power    :    0 W                  Output Power  :  140 W
Solar Ch_1     :    0 W                  Solar Ch_2    :    0 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   : -140 W                  Device Preset :  200 W
Schedule  (Now): 19:28:58 UTC +0200      System Preset :  200 W
ID Start End   Export Output ChargePrio    SB1    SB2  Mode Name
--------------------------------------------------------------------------------
Device         : Smart Meter             Alias         : Smart Meter
Serialnumber   : FYGDVE1ZXEJ6OJAM        Admin         : YES
Site ID        : 38cac050-0128-0eff-c9ba-61dabb1f3df1
Wifi SSID      : wifi-network-1         
Wifi state     : Online                  Signal        :   92 %
SW Version     : v0.0.0.27               Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :    0 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------
```
Since both Grid values were 0W, it is difficult to relate them to the new `"to_home_load": "140"` field which was showing 140 W.
I assume that could actually be the calculated home consumption value, but would need to be proven with other snapshots showing values for more fields. So an export while little Solar power is coming in, there is bypass to the house + discharge from battery + grid is not showing 0 W would be good to correlate all power fields properly.
Please also observe grid_status code, I haven't seen anything else than 0 yet, now describing this as OK state for the grid. I would assume there is another code on power outage, maybe others too ?

stephan-l, 2024-07-02:
Grid status for a power outage will be difficult to obtain. the smart meter is powered by the voltage sensing ports. so it will go offline on power loss. any idea what I shall test next week when I am onsite next time?

I gave it a new try (full dump:  [ste202407022222303.zip](https://github.com/user-attachments/files/16073955/ste202407022222303.zip) )

```
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1,  Devices: 2
--------------------------------------------------------------------------------
Device         : Solarbank 2 E1600 Pro   Alias         : Solarbank 2 E1600 Pro
Serialnumber   : APCGQ80Exxxx        Admin         : YES
Site ID        : 99e
Wifi SSID      : xxx       
Wifi state     : Online                  Signal        :   90 %
SW Version     : v1.0.4.4                Auto-Upgrade  : Disabled
Cloud-Updated  : 2024-07-02 22:23:03     Valid Data    : YES (Requeries: 3)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   64 %                  Min SOC       :   10 %
Battery Energy : 1024 Wh                 Capacity      : 1600 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
Solar Power    :    0 W                  Output Power  :    0 W
Solar Ch_1     :    0 W                  Solar Ch_2    :    0 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   :    0 W                  Device Preset :  200 W
Schedule  (Now): 22:22:50 UTC +0200      System Preset :  200 W
ID Start End   Export Output ChargePrio    SB1    SB2  Mode Name
--------------------------------------------------------------------------------
Device         : Smart Meter             Alias         : Smart Meter
Serialnumber   : AUJS020xxxx        Admin         : YES
Site ID        : 99e3a
Wifi SSID      : xxx       
Wifi state     : Online                  Signal        :   90 %
SW Version     : v0.0.0.27               Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :  132 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------
```
My grid meter was at approx 140W. So input is right.

Also the Capacity is broken again. Before todays git pull it was 3200Wh for my setup:
```
Cloud-Updated  : 2024-07-02 21:57:12     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   65 %                  Min SOC       :   10 %
Battery Energy : 2080 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
```

Somehow the battery is charged but now output to the home. I need to understand when it is charged by solar power. Today there wasn't any output only charging.

stephan-l, 2024-07-02:
After another try I got the correct capacity again. Kind of confusing.
```
Wifi state     : Online                  Signal        :   88 %
SW Version     : v1.0.4.4                Auto-Upgrade  : Disabled
Cloud-Updated  : 2024-07-02 23:04:22     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   64 %                  Min SOC       :   10 %
Battery Energy : 2048 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
```

tagauch1, 2024-07-02:
Over the last ~23 hours I have collected some data every 15min via cron, see attachment. Unfortunately most of them are with "is_display_data"="false". I have attached my interpretation of the values in a screenshot of the Anker app.
[solix.zip](https://github.com/user-attachments/files/16074441/solix.zip)

thomluther, 2024-07-03:
> I made a full dump with the owner account. The system was running at 140W of battery. 0 from grid and solar. [ste20240701222735.zip](https://github.com/user-attachments/files/16058819/ste20240701222735.zip)

@stephan-l thanks for the many extracts, they help to get a better picture on various situations after sorting out the invalid data extracts.

One situation puzzles me however to correlate the values properly:
```
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C1",
          "device_sn": "VKYPTREWMTYU48Q5",
          "device_name": "Solarbank 2 E1600 Pro",
          "device_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png",
          "battery_power": "10",
          "bind_site_status": "",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "6",
          "status": "1",
          "wireless_type": "1",
          "main_version": "",
          "photovoltaic_power": "0",
          "output_power": "0",
          "create_time": 1718909497,
          "set_load_power": "",
          "sub_package_num": 0,
          "output_cutoff_data": 10,
          "is_display": true
        }
      ],
      "total_charging_power": "0",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.10",
      "updated_time": "2024-07-02 21:14:53",
      "total_photovoltaic_power": "0",
      "total_output_power": "0.00",
      "display_set_power": false,
      "battery_discharge_power": "0",
      "ac_power": "0",
      "to_home_load": "0",
      "is_display_data": true,
      "solar_power_1": "0",
      "solar_power_2": "1",
      "solar_power_3": "1",
      "solar_power_4": "1"
    },
    "retain_load": "200W",
    "scene_mode": 1,
    "home_load_power": "0",
    "updated_time": "01-01-0001 00:00:00",
    "power_site_type": 5,
    "site_id": "5e6f13ae-93f7-83bf-d119-b41c4bdc41ba",
    "powerpanel_list": [],
    "grid_info": {
      "grid_list": [
        {
          "device_pn": "A17X7",
          "device_sn": "HVCNROR98JH6W3SG",
          "device_name": "Smart Meter",
          "device_img": "",
          "create_time": 1719137035,
          "status": "1",
          "bind_site_status": "",
          "wireless_type": ""
        }
      ],
      "photovoltaic_to_grid_power": "0",
      "grid_to_home_power": "423",
      "grid_status": 0,
      "has_history_data": true
    },
Generated Solar power is 3 W from all, but this is not reflected somehow in any of the other fields.
grid_to_home_power is 423 W (ok that is the grid import)
The solarbank charging_status = 6 which probably means full_bypass (as for Solarbank 1)
Soc is 10% in the evening at 21:00. Seems weird, but explains the full bypass with 3 W remaining Solar power.
"charging_power": "0", => Nothing is loaded to empty battery, another indication that it is doing full bypass and BMS is not actively used.
"output_power": "0", => Weird, where are the 3W Solar power gone? Maybe output_power not provided when doing full bypass without active BMS?

How is the display in the App in the morning with little solar power and empty battery when this full bypass mode is used without charge and discharge? Is the solar power reflected somewhere else in the figures going to home?

```

thomluther, 2024-07-03:
@stephan-l  Following is another weird situation making it difficult to correlate the values accurately. It appears the individual fields are not used consistently as with Solarbank 1, which makes it pretty difficult to extract the properly for correct monitoring. I build some work arounds for such fields, where e.g. the SB totals do not match the usage of individual solarbank device fields. But I'm reluctant to break anything with those workarounds for anything that does not match with Solarbank 2 reporting.

```
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C1",
          "device_sn": "YL5C48EDP7UZ108S",
          "device_name": "Solarbank 2 E1600 Pro",
          "device_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png",
          "battery_power": "15",
          "bind_site_status": "",
          "charging_power": "800",
          "power_unit": "W",
          "charging_status": "1",
          "status": "1",
          "wireless_type": "1",
          "main_version": "",
          "photovoltaic_power": "597",
          "output_power": "800",
          "create_time": 1718909497,
          "set_load_power": "",
          "sub_package_num": 0,
          "output_cutoff_data": 10,
          "is_display": true
        }
      ],
      "total_charging_power": "0",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.15",
      "updated_time": "2024-07-02 08:30:13",
      "total_photovoltaic_power": "597",
      "total_output_power": "800.00",
      "display_set_power": false,
      "battery_discharge_power": "203",
      "ac_power": "20",
      "to_home_load": "780",
      "is_display_data": true,
      "solar_power_1": "210",
      "solar_power_2": "206",
      "solar_power_3": "87",
      "solar_power_4": "94"
    },
    "retain_load": "200W",
    "scene_mode": 1,
    "home_load_power": "2419",
    "updated_time": "01-01-0001 00:00:00",
    "power_site_type": 5,
    "site_id": "9a7e3ee0-c6d7-67c0-d1bf-3ea08747bcea",
    "powerpanel_list": [],
    "grid_info": {
      "grid_list": [
        {
          "device_pn": "A17X7",
          "device_sn": "PQP04MMR9VCGO3DG",
          "device_name": "Smart Meter",
          "device_img": "",
          "create_time": 1719137035,
          "status": "1",
          "bind_site_status": "",
          "wireless_type": ""
        }
      ],
      "photovoltaic_to_grid_power": "0",
      "grid_to_home_power": "1619",
      "grid_status": 0,
      "has_history_data": true
    },
```
For example the Api reports for device:
          "battery_power": "15",  => SOC
          "charging_power": "800", => Api does not provide correct value here, this is typically 0 or output power, therefore I built in a work around for the Api site cache structure. This work around also uses negative values for discharge power, which is taken from the SB total_discharge_power (seems to be the only trustable field, but shows discharge of all solarbanks only). Therefore this vlue must also be adjusted to multiple solarbank setups based on their output_power percentage of the reported total_output_power. Its complicated, but the only way to get reliable charge and discharge values per Solarbank 1 device.
          "charging_status": "1", => This is Bypass for Solarbank 1 (neither charge nor discharge), looks to be the similar for SB2, but here bypass can mean also adding discharge power, a mode that is not supported by Solarbank 1. I think that can be distinguished with a pseudo code and dedicated description for bypass_discharge
          "output_power": "800", => Valid Max demand as requested from smartmeter upon high house consumption of 2419 W
SB totals:
      "total_photovoltaic_power": "597", => OK, sum of all solar channels
      "total_output_power": "800.00", => OK, same as device output power
      "battery_discharge_power": "203",  => Only valid discharge value provided. This will be taken by the library to put -203 W into the invalid 'charging_power' field of a solarbank device.
      "ac_power": "20",  => Why are there 20 W on the AC socket? Is there anything connected that draws AC socket power?
      "to_home_load": "780", => That means from 800 W output, 20 W go to the AC socket (true?) and 780 W into house?
Grid total:
      "photovoltaic_to_grid_power": "0", => OK, no grid export at this time
      "grid_to_home_power": "1619", => OK, required import for high house consumption. Assuming this is valid as reported from smartmeter, then we have 2419 -1619 = 800 W. And here is the big questionmark: Since we have only 780 W to the house, where are the 20 W? Reported incorrectly to the AC power, which is not used (to be confirmed)? Or if the 20 W AC power correct, then the house consumption value is calculated incorrectly by the appliance and 20 W too high (since it does not consider the AC output in the calculation and using total_output_power instead of to_home_load)?
      "grid_status": 0,
Site total:
    "home_load_power": "2419", => Questionable value as per previous comments...

thomluther, 2024-07-03:
> After another try I got the correct capacity again. Kind of confusing.
>
> ```
> Wifi state     : Online                  Signal        :   88 %
> SW Version     : v1.0.4.4                Auto-Upgrade  : [...]

This might be related to the capacity and energy calculation. If the initial response is invalid, the battery packs are 0 too, Maybe the calculations are not refreshed properly afterwards. The code also needs to make sure, that re-calculation is not done when the response contains invalid data, so it must cover both special cases:
- initially invalid but calculated fields must be generated (potentially with a value), otherwise HA integration would not generate entities.
- updates must ignore re-calculation with invalid data

stephan-l, 2024-07-03:
I think I will do some measurements onsite next week to verify the theses. I can monitore questionable values and test different scenarios then.

thomluther, 2024-07-04:
Can someone explain this situation?
I don't remember who provided the T2 system export... There is home demand and import from grid, but SB2 is not delivering in the evening even if SOC is 64%
Can home delivery be disabled somehow or why is there no discharge?
```
--------------------------------------------------------------------------------
Device         : Solarbank 2 E1600 Pro   Alias         : Solarbank 2 E1600 Pro
Serialnumber   : IDT679D215D635EA        Admin         : YES
System         : T2  (Site ID: c316e3a9-befc-47c9-a8bf-c211dd37d31b)
Wifi SSID      : wifi-network-1         
Wifi state     : Online                  Signal        :   90 %
SW Version     : v1.0.4.4                Auto-Upgrade  : Disabled
Cloud-Updated  : 2024-07-02 22:23:03     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   64 %                  Min SOC       :   10 %
Battery Energy : 2048 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
Solar Power    :    0 W                  Output Power  :    0 W
Solar Ch_1     :    0 W                  Solar Ch_2    :    0 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   :    0 W                  Device Preset :  200 W
Home Demand    :  132 W                  SB Home Load  :    0 W  (-)
Schedule  (Now): 07:19:51 UTC +0200      System Preset :  200 W
ID Start End   Export Output ChargePrio    SB1    SB2  Mode Name
--------------------------------------------------------------------------------
Device         : Smart Meter             Alias         : Smart Meter
Serialnumber   : 9Q4CSMNJLAPJORP7        Admin         : YES
System         : T2  (Site ID: c316e3a9-befc-47c9-a8bf-c211dd37d31b)
Wifi SSID      : wifi-network-1         
Wifi state     : Online                  Signal        :   90 %
SW Version     : v0.0.0.27               Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :  132 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------

```

thomluther, 2024-07-04:
Here is another interesting example, showing kind of charge_priority in the morning even if in detection mode, since SOC is just at 7%. I don't think this should get a different pseudo state since it is charging code 0, and all kind of charging states typically have 3.
But clearly it shows that demand is higher than solar and even then not all is bypassed to home to 'enforce' charging of low battery.
```
--------------------------------------------------------------------------------
Device         : NoName                  Alias         : Solarbank 2 E1600 Pro
Serialnumber   : AXKI2U6VS8IQ7NZS        Admin         : NO
System         : Maison  (Site ID: adb7f7c0-c8c0-dcdc-ebc6-6e0a7a7b3ec1)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud-Updated  : 1970-01-01 00:00:00     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:    7 %                  Min SOC       :    5 %
Battery Energy :  224 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
Solar Power    :  573 W                  Output Power  :  467 W
Solar Ch_1     :  144 W                  Solar Ch_2    :  141 W
Solar Ch_3     :  147 W                  Solar Ch_4    :  141 W
Charge Power   :  106 W                  Device Preset :    0 W
Home Demand    :  624 W                  SB Home Load  :  467 W  (-)
--------------------------------------------------------------------------------
Device         : NoName                  Alias         : Smart Meter
Serialnumber   : 3SJSUU0Z3W0VXQKZ        Admin         : NO
System         : Maison  (Site ID: adb7f7c0-c8c0-dcdc-ebc6-6e0a7a7b3ec1)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :  157 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------

```

thomluther, 2024-07-04:
OK, so just a firmware flaw, nothing to be considered for the field interpretation.

thomluther, 2024-07-04:
I updated the api with new charging states and the monitor accordingly. #104 
I also added the new house demand and SB_home_load fields to the monitor. This is what is new on the SB2 systems in the home screen of the app.
The house demand ("Zuhause") is a new system field (home_load_power) and the SB home load is a solarbank devices total field (to_home_load). The house demand does not reflect the house consumption all the time. When the solarbank is on standby, it is 0. So it must not be confused with a value that can be used for permanent house consumption.
The former SB output total refers to all power that is going out from the solarbanks. The new SB to home load is the same, but subtracted by the AC socket power. So for SB2, only the new field is reflecting what is going into the house.
The smartmeter values seem to provide also data when SB2 is in standby.

However, someone needs to confirm with the monitor whether the smartmeter values are valid when the SB data is invalid. Right now I consider them invalid too, but they are 0W most of the time also with valid SB data, so it is difficult to recognise that in a snapshot.

thomluther, 2024-07-05:
One more note to make here:
- The new SB to_home_load field is not used by Solarbank 1 (always 0). In fact I think it reflects the AC power from the inverter that goes into the house, so it is SB output power (used by SB1 and SB2) - AC power field which is likely the AC socket.
- That means without Socket (SB2 Plus and later SB2) should always reflect same values for old SB output power and new to_home_load fields.
- The system power field home_load_power is the AC power house demand for the solarbank. This is the consumption of the house, while the Solarbank is ON, when the demand seems to reflect 0 W. I assume that is the value used as target for the Solarbank regulation
- All power values sum up to same totals typically including the solar power channels. That means they cannot be real, since AC and DC power is mixed an no inverter efficiency loss is factored into the values. I assume the DC power values which cannot be really proven (solar, Battery etc) are adoped / decreased? to match the AC power values used for smartmeter/grid and house.
- The AC power values could be easily compared with other metering tools, but I understood they seem to be pretty accurate. Therefore I believe the power values would need to be 5-10% higher in reality to consider inverter loss etc. So tracking the DC values for energy measurements is likely providing optimized measurements in terms of efficiency for the solarbank 2 when comparing them with the energy output to the house...

thomluther, 2024-07-10:
@nicolinuxfr , @stephan-l , @tagauch1 can anyone assist with the other open issues to continue work on remaining SB2 support implementation?:
- #106 
- #107 
- #108 
- #109 
- #116

nicolinuxfr, 2024-07-11:
Sorry, my knowledge is too limited to really help. I can do more exports, but I'm not sure it's useful ? Here's one I just did using my admin account this time : 

[nicolinux.zip](https://github.com/user-attachments/files/16178970/nicolinux.zip)

stephan-l, 2024-07-11:
I'll try to help as much as possible but my system is installed in a remote location so I need to drive there for physical tests.

thomluther, 2024-07-11:
> Sorry, my knowledge is too limited to really help. I can do more exports, but I'm not sure it's useful ? Here's one I just did using my admin account this time :
> 
> [nicolinux.zip](https://github.com/user-attachments/files/16178970/nicolinux.zip)

Would already help to define various schedules and export then to see various interval setting and weekday settings.

thomluther, 2024-07-12:
@nicolinuxfr I have seen you have already a basic schedule defined.
Can you let me know which options can be defined in SB2 schedule intervals?
Please also create some additional intervals and different schedules for a subset of weekdays, we need to see how the schedule structure will change then and whether there come new fields with the SB2 intervals....

thomluther, 2024-07-12:
PS: The new api version has been refactored to have smaller files in the library. I separated also all type definitions of the Api, therefore I also renamed _API_ENDPOINTS to API_ENDPOINTS to avoid code checker warnings for accessing internal variables.
In case you used them in you test modules, you have to rename them.
I also added the energy exports for each PV channel in the Api and the solarbank monitor.
However, I have not seen their structure yet, so could be that the display in the monitor is not working yet, in case the structure is different than for other energy stats.
A new system_export with the additional solar production energy files per PV channel will be usefull to verify their structure.

nicolinuxfr, 2024-07-12:
Yes, only one to stop the system to send electricity during the night, when I have off-peak prices. 

> Can you let me know which options can be defined in SB2 schedule intervals?

Here are some screenshots from the app, I can't export more systems right now. If I understand what you need, I should create more complex schedules and then ask for an export for each variation ?

[image]
[image]
[image]
[image]
[image]

thomluther, 2024-07-12:
OK, so not much parameters in the App, but the same as for SB1 are still in the schedule structure.
What is new however are the weekdays. Pls create intervals for subset of weekdays and export the system again with the main account using the latest api version and export tool.
This will also give an example of the PV1-4 energy stats queries that I added.

nicolinuxfr, 2024-07-13:
Here is an export after setting this schedule inside the app :
[schedule.zip](https://github.com/user-attachments/files/16202362/schedule.zip)

[image]
[image]

Let me know if you need a different setup next.

thomluther, 2024-07-13:
@nicolinuxfr 
Thanks, just figured out that the current request parameter cannot be used to get SB2 schedule formats.
See #109 
I need to adopt the system export tool to use also dev_parm=6 in the collection, that should provide the correct format of the schedule. Seems the set and get device_load endpoints are not suitable for SB2 at all....

thomluther, 2024-07-14:
@nicolinuxfr 
I updated the system export to get now device parm for type 4 & 6, so should now extract the new schedule structure.
I assume the user mode setting could be part of the schedule structure although it appears as separate setting in the App.
So please extract with user mode active as well as with AI mode active to see whether the schedule structure is different between the 2 exports.
Thanks

nicolinuxfr, 2024-07-14:
@thomluther here you go ! (automatic for AI mode, Manuel for a schedule)

[automatic.zip](https://github.com/user-attachments/files/16216353/automatic.zip)
[manuel.zip](https://github.com/user-attachments/files/16216354/manuel.zip)

thomluther, 2024-07-15:
@nicolinuxfr Please see my [update in the schedule structure issue](https://github.com/thomluther/anker-solix-api/issues/108#issuecomment-2227895854).
If you can validate the changes in the monitor and comment/confirm the questions that would be great.

thomluther, 2024-07-15:
@stephan-l @tagauch1 
If you have a smartmeter installed, please see [update in the schedule structure issue.](https://github.com/thomluther/anker-solix-api/issues/108#issuecomment-2227895854)
Would be helpfull if you can confirm that Api schedule extraction now works and is displayed correctly in the monitor.
You need to run the monitor with the main account to see the schedule. Please also run it with energy stats and validate the the extracted numbers for today and yesterday reflect all daily stats that you can find in the App. I added also the stats for each PV channel. Don't know if there are more in the app.
PS: For the energy stats, you can run the monitor with the shared account and compare with the App values using the main account there.

stephan-l, 2024-07-15:
I tried it and it looks legit:
```
Running site refresh...
Running device details refresh...
Running energy details refresh...
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1,  Devices: 2
--------------------------------------------------------------------------------
Device         : Solarbank 2 E1600 Pro   Alias         : Solarbank 2 E1600 Pro
Serialnumber   : AP        Admin         : YES
System         : T2  (Site ID: 99e3)
Wifi SSID      : xxxx
Wifi state     : Online                  Signal        :   90 %
SW Version     : v1.0.5.4                Auto-Upgrade  : Disabled
Cloud-Updated  : 1970-01-01 00:00:00     Valid Data    : YES (Requeries: 0)
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:   73 %                  Min SOC       :   10 %
Battery Energy : 2336 Wh                 Capacity      : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket     :    0 W
Solar Power    :    0 W                  Output Power  :  138 W
Solar Ch_1     :    0 W                  Solar Ch_2    :    0 W
Solar Ch_3     :    0 W                  Solar Ch_4    :    0 W
Charge Power   : -138 W                  Device Preset :  200 W
Home Demand    :  138 W                  SB Home Load  :  138 W
Schedule  (Now): 21:50:38 UTC +0200      System Preset :  200 W
Usage Mode     : Automatic (1)           Sched. Preset :  200 W
ID Start End   Output Weekdays
--------------------------------------------------------------------------------
Device         : Smart Meter             Alias         : Smart Meter
Serialnumber   : AUJ        Admin         : YES
System         : T2  (Site ID: 99e3)
Wifi SSID      : xxx
Wifi state     : Online                  Signal        :   90 %
SW Version     : v0.0.0.29               Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Grid Status    : ok                      Status code   : 0
Grid Import    :    0 W                  Grid Export   :    0 W
--------------------------------------------------------------------------------
Energy details for System T2 (Site ID: 99e3):
Today          : 2024-07-15              Yesterday     : 2024-07-14
Solar Energy   :  4.94 kWh               Solar Energy  :  4.58 kWh
Solar Ch 1/2   :  2.47 / 2.47 kWh        Solar Ch 1/2  :  2.29 / 2.29 kWh
Solar Ch 3/4   :  0.00 / 0.00 kWh        Solar Ch 3/4  :  0.00 / 0.00 kWh
SB Charged     :  3.07 kWh               SB Charged    :  2.40 kWh
SB Discharged  :  1.43 kWh               SB Discharged :  1.24 kWh
House Feed     :  1.41 kWh               House Feed    :  1.23 kWh
House Usage    :  3.39 kWh               House Usage   : 16.17 kWh
Grid Import    :  0.12 kWh               Grid Import   : 12.77 kWh
Grid Export    :  0.02 kWh               Grid Export   :  0.01 kWh
AC Socket      :  0.00 kWh               AC Socket     :  0.00 kWh
Smartplugs     :  0.00 kWh               Smartplugs    :  0.00 kWh
Sol/Bat/Gri %  :  38/ 62/  0 %           Sol/Bat/Gri % :  48/ 52/  0 %
--------------------------------------------------------------------------------
```
Statistics look fine and the state of charge is the average of my internal and external battery.
Latest export: 
[ste20240715215000.zip](https://github.com/user-attachments/files/16239805/ste20240715215000.zip)

thomluther, 2024-07-16:
Hello
Can someone provide me an actual export as owner account when NO schedule is defined (everything cleared in your schedule settings). Please upload this to #108 to discuss follow on activities there.

I need to see what the minimal schedule structure looks like and if that is something to be usable as template for composing a minimal response to change any schedule related settings (usage mode or actual output preset). 

Once I know the minimal structure, I will provide examples of possible structures to apply a change, and someone need to verify them what works and what won't work.
The helper method required for the SB2 schedule need to be able to modify a single setting also when no schedule is defined yet.
For the usage mode this will be simple, since it is not tight to the schedule itself. But for the output preset, a rate plan and a range interval will have to be created from scratch if none exists for actual time. Will have to think how to deal with different days in that case since some could already be defined and just a range for an actual time gap needs to be created to apply the actual output preset.

thomluther, 2024-07-29:
@stephan-l did you use the latest release of the export tool?
Your output does not contain the OTA file without SN...

stephan-l, 2024-07-29:
> @stephan-l did you use the latest release of the export tool? Your output does not contain the OTA file without SN...

Yes it is the current main branch.
Just tried again:
``
Saved JSON to file /Users/sl/PycharmProjects/anker-solix-api/./exports/ste20240729190900/device_fittings_X9K0XWK8P04DRHR9.json
Exporting load...
(10004) Anker Api Error: Failed to request.
Response Text: {"code":10004,"msg":"Failed to request.","trace_id":"4f311ef1dd9dff7f8e9241f6ba284f5f"}
Exporting OTA update info for device...
Saved JSON to file /Users/sl/PycharmProjects/anker-solix-api/./exports/ste20240729190900/ota_update_X9K0XWK8P04DRHR9.json
Exporting device attributes...
``
I just tried to find the change in the code but didn't found it. Which commit do you refer to?

Krosty0712, 2024-09-09:
[krostysun.zip](https://github.com/user-attachments/files/16928720/krostysun.zip)
in the app a OTA Update is available 

in the ota files the request is failed.
anything i can, test ? 
My linux skills are not the best, but i will try my what i can.

thomluther, 2024-09-18:
@Krosty0712 the OTA requests do not work for the SB2 and smartmeter, don't know why. For SB1 they are also not reliably showing an available OTA update.
I have no clue how to query that and what would be valid versions, don't think this is provided via the cloud api.

What you can test with your setup are new information that can be queried via Api. See #107

amor20git, 2024-10-03:
> Please provide your solarbank 2 system export examples here. You can extract all data with the system_export tool and zip the output files. 

Where can I find the "system_export" tool? I do have an Anker Solix2 E1600 System and could provide information requested
Thx

thomluther, 2024-10-07:
@amor20git 
Standard SB2 exports are not longer required if you do not have unsupported devices or settings in your system,
Please refer to following required exports:
- #124 Support for smartplugs and settings to Smart Plug usage mode
- Exports of systems with Shelly 3EM (Pro) smart meters

Beside the exports for smart plugs, the community has to provide more Api Endpoint testing with the smart plugs to see whether they can also be controlled via Api (Switch on / off, change their schedule, basically anything what is possible via App needs to be evaluated via Api endpoints as well). 
It is also questionable and unknown, what the #107 device_attrs endpoints can be used for?
They have been implemented with SB2 support. I haven't found any attribute name yet that provides any data with SB1, but maybe that is working only with SB2, Smart Meter or Smart plugs which I cannot test.

rwe87, 2024-10-17:
Here is the export of a system with:

- 4 Anker SOLIX FS20 Flexible Solar Panels
- 1 Anker SOLIX Solarbank 2 E1600 Plus
- 1 Anker SOLIX Smart Plug

[robxxx.zip](https://github.com/user-attachments/files/17409450/robxxx.zip)

steghoja, 2024-10-21:
This is an export of a system with the following components:

- Solarbank E1600 Pro without an extra battery pack
- Anker Solix Smart Plug under load with a schedule set to turn off at 21:24 every day but Monday
- Anker Solix Smart Plug under load with a delayed switch set to turn off after about 2 hours
[nkr.zip](https://github.com/user-attachments/files/17465559/nkr.zip)

thomluther, 2024-10-21:
> This is an export of a system with the following components:
>
>     * Solarbank E1600 Pro without an extra battery pack
>
>     * Anker Solix Smart Plug under load with a schedule set to turn off at [...]

That contains some interesting new data with the latest export module:
- The ota_batch query really seems to list all firmware levels including sub components and whether new update is available. It will be helpful to provide a new export with that response once there is really an update available for the SB2 and/or smart meter/ or smart plugs. Hopefully the new version will be listed as well, so this could be extracted for the HA integration as well. That also means I can remove the other OTA queries again since they do not work most of the time...
- The device_parm_6 response contains not only a custom rate plan (SB2 schedule), but also a blend_plan which was always null so far. Both are whole day, every day of the week, but rate plan shows 50 W while blend plan shows 20 W. Is that schedule with 20 W something you configured in the App and what is the blend_plan used for?
```
"param_data": "{\"mode_type\":2,\"custom_rate_plan\":[{\"index\":0,\"week\":[0,1,2,3,4,5,6],\"ranges\":[{\"start_time\":\"00:00\",\"end_time\":\"24:00\",\"power\":50}]}],\"blend_plan\":[{\"index\":0,\"week\":[0,1,2,3,4,5,6],\"ranges\":[{\"start_time\":\"00:00\",\"end_time\":\"24:00\",\"power\":20}]}],\"default_home_load\":200,\"max_load\":800,\"min_load\":0,\"step\":10}"
```
- I can see power values for both plugs which is good. The same for energy of the plugs. It seems the smart plug data can then be added properly to the Api structures.
- What I'm still missing are responses showing the smart plug setting, the delayed switch setting and the schedule. You will need to explore some Api queries whether you can find the proper endpoints/parameters to get those from the cloud as well.

steghoja, 2024-10-22:
The blend plan must be what the Anker app calls "Smart Plug Mode". As you had suggested in a previous post, I had set a base load of 20W for that mode that was active on all days of the week from 0:00 to 24:00. That's what we see in the data. What the solarbank then does is to always supply those 20W plus whatever is consumed according to the smart plugs.

thomluther, 2024-10-22:
So you are saying that you have 2 different schedules in the App?
- One schedule for the Manual Usage mode
- A different schedule for the Smart Plug mode
My understanding was there is only one schedule, and the system will use that schedule and add the power of the plugs.
If there are indeed 2 schedules, more work is required to support the smart plug mode since that would require to modify the blend plan structure instead of the custom_rate_plan structure.
Can you confirm that you really can manage 2 different schedules in the App when switching between Manual mode and Smart Plug mode?

steghoja, 2024-10-22:
Indeed, it's exactly as you say. I have one schedule set for the manual mode and one for the smart plug mode and I can switch between the two.

thomluther, 2024-10-22:
Oh dear.....Anker comes always up with new structures....I have no clue how to implement handling of multiple schedules for a device, because so many values/entities are derived from the schedule and now it must be distinguished which is the active schedule....
On the other hand, it is not sufficient for HA to handle only the active schedule, but the services must be able to manage both, the active and inactive one. Maybe this can be done with an additional service selection field for which schedule the service should be applied.....
But all entities will only work against the active schedule and selection of usage mode will also switch the active schedule...

thomluther, 2025-06-21:
> what's about the new solar bank 3?

See #179