thomluther/anker-solix-api #116: Explore how to identify OTA upgrades for Solarbank 2 and Smartmeter

thomluther, 2024-07-02:
The original method to query available OTA updates is not working for Solarbank 2 sites.
```
                        await myapi.request(
                            "post",
                            api._API_ENDPOINTS["get_ota_update"],  # noqa: SLF001
                            json={"device_sn": sn, "insert_sn": ""},
                        ),
```
Either this is not supported for Solarbank 2 and Smartmeter device SNs, or json parameters are missing for them?
Eventually also the mandatory insert_sn field must be specified for those devices?

thomluther, 2024-07-10:
Can someone test the get_ota_update endpoint for required parameters with SB2 and Smartmeter devices?
This was supposed to show whether firmware update is available, but request show errors when running against the new devices.
This would cause permanent warnings if used by the library in the HA integration...

stephan-l, 2024-07-11:
I missed to export before updating yesterday. I saw the new firmware in the app and clicked before doing my last export.
Maybe someone else not yet upgraded?

thomluther, 2024-07-11:
The problem is the query itself. It's needs different parameters to avoid returning a request error, so someone need to play around with that endpoint and the parameters to see what is working. It must also work when no update is available.

stephan-l, 2024-07-11:
At least I got one useful response, but it is useless:
```
await myapi.request(
            "post",
            api._API_ENDPOINTS["get_ota_update"],  # noqa: SLF001
            json={"device_sn": "", "insert_sn": ""},
        ),
```
```
{
  "code": 0,
  "msg": "success!",
  "data": {
    "is_ota_update": false,
    "need_retry": false,
    "retry_interval": 0,
    "device_list": null
  },
  "trace_id": "10dxxxxx"
}
```

thomluther, 2024-07-11:
> At least I got one useful response, but it is useless:
>
> ```
> await myapi.request(
>             "post",
>             api._API_ENDPOINTS["get_ota_update"],  # noqa: SLF001
>             json={"device_sn": [...]

No, that is exactely the response I expect. Its just strange, that you did not specify a SN at all, and when using the solarbank or smartmeter SN as during system export, that seem to cause an error....

Well, it will be good if someone can try this query when there is an update for any of the devices. In that case,  "is_ota_update": should be true.
This is a boolean sensor that any update is available. However if this is not returned per device SN, it is questionably whether the update should be flagged against the SB2 or the smartmeter.
More testing will be necessary how this query can be used with the new devices

nicolinuxfr, 2024-07-13:
I have OTA updates waiting and I'm trying to help, but how should I run this code ? I'm not a developper and it's hard to me to understand how to run this request, I tried to create a python file but got this error : 

`<class 'AttributeError'>: module 'api.api' has no attribute '_API_ENDPOINTS'`

Could you send me the content of the python file to run ?

thomluther, 2024-07-13:
Hi, this error might be because I refactored the api code and removed the _ prefix from the constants.
They are now defined in types module and you can import them:
from api import api, types

then you can refer them with types.API_ENDPOINTS

or 
from api.types import API_ENDPOINTS
then you can directly use API_ENDPOINTS definitions.

But for testing you may also use the endpoint directly
print(json.dumps((await myapi.request("post", "power_service/v1/app/device/get_device_attrs",json={"device_sn": deviceSn, "attributes": ["anyMeaningfulString?"]})), indent=2))

For the OTA update, I figured out that the request also works without any SN specified, so you can pass empty strings to the required parameters. When you use the helper method, this would work:
print(json.dumps(await myapi.get_ota_update(deviceSn="", insertSn=""), indent=2))

If that shows "is_ota_update": true, in your case, then the flag is maybe just for any update in your account (for any site, since also no site parameter is required). If it works without SN, test how it works with specific device SNs for either of the 2 parameters.
I would like to understand if this flag has only a common meaning, or is tied for a given SN in the query...

Currently I record this flag against the solarbank device, but that might be wrong if it is a common OTA flag for the account.

nicolinuxfr, 2024-07-14:
I think what I don't understand is basic Python stuff. Where should I use the print command ? I'm guessing inside a file that import everything, but if you could help me set it up, I always get errors and I don't know how to debug these.

thomluther, 2024-07-14:
@nicolinuxfr , for an example python test module look the [usage example](https://github.com/thomluther/anker-solix-api?tab=readme-ov-file#usage)
It also shows how you can use the print statements. But you don't have to run all 4 api cache poller each time as in the example.
I single call to your initialized myapi class with the myapi.request method will do it. The print and json statements are just for proper formatting and printing of the request response.

But actually I'm a bit puzzeled whether that  "is_ota_update":  flag is usable at all. It went on in my HA integration today for SB1, but when checking it in the app, there is no update information visible in my account for any device...So I wonder how that flag is used with the account? Will need to observe further, whether the App will also show an update within the next hours/days...

nicolinuxfr, 2024-07-14:
Here's the file I created : 

```python
import logging, json
import asyncio
from aiohttp import ClientSession
from api import api
import common

_LOGGER: logging.Logger = logging.getLogger(__name__)
# _LOGGER.setLevel(logging.DEBUG)    # enable for detailed Api output

async def main() -> None:
    """Create the aiohttp session and run the example."""
    async with ClientSession() as websession:
        """put your code here, example"""
        myapi = api.AnkerSolixApi(
            common.user(), common.password(), common.country(), websession, _LOGGER
        )
        
        print(json.dumps((await myapi.request("post", "power_service/v1/app/device/get_device_attrs",json={"device_sn": deviceSn, "attributes": ["anyMeaningfulString?"]})), indent=2)),

# run async main
if __name__ == "__main__":
    try:
        asyncio.run(main())
    except Exception as err:
        print(f"{type(err)}: {err}")
```

Here's the error I got : 

```shell
(.venv) ➜  anker-solix-api git:(main) ✗ source .venv/bin/activate && pipenv run ./essai.py
Loading .env environment variables...
Courtesy Notice: Pipenv found itself running within a virtual environment, so it will automatically use that environment, instead of creating its own for any project. You can set PIPENV_IGNORE_VIRTUALENVS=1 to force pipenv to ignore that environment and create its own instead. You can set PIPENV_VERBOSITY=-1 to suppress this warning.
Traceback (most recent call last):
  File "/home/nicolas/anker-solix-api/.venv/bin/pipenv", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/cli/options.py", line 58, in main
    return super().main(*args, **kwargs, windows_expand_args=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/decorators.py", line 92, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/vendor/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/cli/command.py", line 424, in run
    do_run(
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/routines/shell.py", line 106, in do_run
    run_fn(*run_args, **run_kwargs)
  File "/home/nicolas/anker-solix-api/.venv/lib/python3.11/site-packages/pipenv/routines/shell.py", line 136, in do_run_posix
    os.execve(
OSError: [Errno 8] Exec format error: './essai.py'
```

thomluther, 2024-07-14:
Hm, looks like a runtime error with your setup.
You may also open the virtual environment manually and from within, just execute your python module directly.

thomluther, 2024-07-29:
@stephan-l , there should be mutliple OTA update files in the export. One for each SN and another one without SN.
The latter seems to be the only one that works with SB2 and smart meters.
So actually I'm not sure what the output is reflecting, since it only queries OTA updates per site without SNs
Please verify the output of the file without SN before and after the updates.
Would be good to know what the `is_ota_update `field is reflecting with and without outstanding updates:
```
  "data": {
    "is_ota_update": true,
    "need_retry": true,
    "retry_interval": 2000,
    "device_list": null
  },
```

The weird thing for my site is that it reflects true since last week, but I have no outanding updates for anything...

stephan-l, 2024-07-29:
I made a new full dump:
[ste20240729160515.zip](https://github.com/user-attachments/files/16415649/ste20240729160515.zip)
Not sure what's wrong but I always get Failed to request

thomluther, 2024-07-29:
> I made a new full dump: [ste20240729160515.zip](https://github.com/user-attachments/files/16415649/ste20240729160515.zip) Not sure what's wrong but I always get Failed to request

That is correct for the device related queries, but[ line 372](https://github.com/thomluther/anker-solix-api/blob/1a56579abc1336117fc60af56bfa244bf394c09f/export_system.py#L372) in the code does the OTA update also as last query per site without SNs before exporting energy stats
```
                CONSOLE.info("Exporting OTA update info...")
                try:
                    export(
                        os.path.join(folder, "ota_update.json"),
                        await myapi.request(
                            "post",
                            api.API_ENDPOINTS["get_ota_update"],  # noqa: SLF001
                            json={"device_sn": "", "insert_sn": ""},
                        ),
                    )  # works only for site owners
                except (ClientError, errors.AnkerSolixError):
                    if not admin:
                        CONSOLE.warning("Query requires account of site owner!")

```

stephan-l, 2024-07-30:
Yes over there the same happens:
``
Exporting OTA update info...
(10000) Anker Api Error: Failed to request.
Response Text: {"code":10000,"msg":"Failed to request.","trace_id":"ad22eac5867f17..."}
Exporting site energy data for solarbank...
``

thomluther, 2024-07-30:
OK interesting, because that was working before the recent cloud change on 25. July. I have system exports showing this response...
Maybe the cloud change broke something with OTA update structures for Solarbank 2 system, because also the automatic update settings is no longer returning valid data, just a null response field which is strange.
You should see that in the App as well, because you no longer have selectable devices for automatic FW upgrades, while the SB2 was automatically upgrade before this cloud change....
Anyway, I will remove this query from the Api device data refresh since the content for SB1 is not reliable either...

closed 2024-07-30 by thomluther