thomluther/anker-solix-api #117: Add F3800/BP3800 Equipment when connected to Home Power Panel

doublehelix46, 2024-07-02:
Hello - 

Attached is a system export of two F3800's (with two attached BP3800 expansion batteries) connected to an Anker Home Power Panel  in the USA. I ran several exports with the API tools (with the owner account) and each time I received minimal output. I don't believe the existing API work to date accounts for this equipment - but figured I'd shared the output if it may aid in future development. I am more than happy to provide further logs in the future.

[F3800 System Export.zip](https://github.com/user-attachments/files/16075315/F3800.System.Export.zip)
[F3800 System Export.zip](https://github.com/user-attachments/files/16075315/F3800.System.Export.zip)
[F3800 System Export.zip](https://github.com/user-attachments/files/16075315/F3800.System.Export.zip)

thomluther, 2024-07-03:
Hi @doublehelix46 
your output is not even showing any sites listed for your account.
Since you the first one with US country code, I assume the server assignment of the Api library may be incorrect. US is being assumed for the *.com api server.
You may try the export again but using an EU country code, e.g. DE. This should assign your login to the EU server and show if your account is located there as well.
From what I have seen, there are only 2 Cloud servers ww in use. If you access any kind of Beta Server, that will be also different and not accessibly via the Api library since not known...

doublehelix46, 2024-07-03:
Hi @thomluther -
I believe that issue was on me. I originally had trouble using the "US" code and tried "USA" and started receiving some outputs. I cleared the auth cache and went back to "US" and received much more detail. FYI the DE server did not work for me - not sure if my account is bound or, similarly, would have required an authcache clearing to correct itself.

I ran this a few times but I'm still getting "data_valid": false, in api_sites - as such, all my energy values are 0'd out. I will say this has been my experience with the HACS integration in a 24 hour period as well, so I'm making the assumption these devices are not yet supported. 

More than happy to run additional outputs. If I get a sample that is set to true, I will upload.

[US2.zip](https://github.com/user-attachments/files/16083161/US2.zip)

thomluther, 2024-07-03:
USA is no valid country code. Any invalid or unknown should default to the EU server, hence no data for your authentication.
While the username is not changed, the cached login response will be used and the entered password and ccountry code will be ignored. So to enforce re-authentication you have to clean the Api auth-cache file for your username.

> I ran this a few times but I'm still getting "data_valid": false, in api_sites - as such, all my energy values are 0'd out. I will say this has been my experience with the HACS integration in a 24 hour period as well, so I'm making the assumption these devices are not yet supported.

I have to double check that, its a new field I had to introduce shortly for Solarbank 2 responses.
However, this does nothing with the returned data from the cloud. They are all saved in the Api cache as is, just marked as invalid eventually.
So if you get 0 values, that is what the cloud responds on your request. May be a common new issue as introduced with Solarbank 2, see #114 

If you have a send account to share with your system, test the 2nd account in the App.
If that should the same 0 values (or empty) and only shows valid numbers about every 5 minutes, you need to address this to Anker support, that shared accounts cannot longer see valid data on the system home page.

thomluther, 2024-07-03:
Actually there are no power data fields in your power panel structures when you look at scene info, and the common power fields of the system are not showing real data:
```
    "retain_load": "0W",
    "scene_mode": 0,
    "home_load_power": "0",
    "updated_time": "01-01-0001 00:00:00",
    "power_site_type": 4,
    "site_id": "d9db3ea0-d48c-ecf8-57df-9ad78ec60ebf",
    "powerpanel_list": [
      {
        "device_pn": "A17B1",
        "device_sn": "PM6T2AS3Z62D3GEG",
        "device_name": "SOLIX Home Power Panel",
        "device_img": "https://public-aiot-ohi-prod.s3.dualstack.us-east-2.amazonaws.com/anker-power/public/product/2024/01/23/iot-admin/Qcjo9ALaRqFzFNeF/picl_A17B1_normal.png",
        "create_time": 1719792264,
        "bind_site_status": "",
        "status": "1",
        "wireless_type": "1",
        "initial_process": 0
      }
    ],
```
Also your home_page data looks the same 
So I wonder where the App gets the power values from, but it does not seem to get them from the cloud Api.
Since the cloud does not receive power values, it would explain why all your energy statisitcs are also empty.
Do you see statistics data in the App?
Have you connected all your devices to Wifi? Otherwise they cannot send data to the cloud.

doublehelix46, 2024-07-03:
@thomluther Thanks for the quick responses. Here's some insight on my setup and my own observations.

1) I had a delay in having the Home Power Panel installed - so I had setup both of the F3800's on wifi as two individual devices. However, they have since (and are currently) been connected to a Power Panel and I was able to create a home in the Anker App.

[image]

2) Once connected to the Power Panel, the F3800s no longer connect to wifi (or make any standalone calls via the internet from what I can see in my network monitoring.) Additionally the solar bank monitor suggests they are offline as well.
[image]

The Power Panel is really only connecting to the MQTT and API servers
[image]

3) I can see what appears to be live stats in the app (even remotely away from my home network.) I've attached a screen recording as well to step through the various tabs on the Home Screen. Just a quick note - I switched my app to a shared account so I could monitor while generating the previous outputs. (sorry for the low res I needed to adjust for file size.)

https://github.com/thomluther/anker-solix-api/assets/69429590/e52c0401-5142-4ee7-bce9-479ee26463cd

thomluther, 2024-07-03:
Hm, seems weird from what I have seen so far for Solarbank systems.
There all power values are provided in the Solarbank structure of the scene_info or homepage response.
For Power Panels that doesn't seem to be the case, rathermore the home page is already triggering device updates via MQTT servers and maybe the usage values are merged from that api into the home screen?
In other words the cloud api only knows the system structure and devices, but no consumption values, therefore the energy_statistic queries are also empty.

I can try to add as much as possible from the provided Api fields, but that will be pretty much useless in HA in the long term if we do not get power values via the cloud Api...

thomluther, 2024-07-04:
When I look at your homepage structure, it seems your PPS (F3800) devices are not really added / joined to the system since the pps list is empty. That means the cloud does not consider them belonging to the site
```
  "data": {
    "site_list": [
      {
        "site_id": "d9db3ea0-d48c-ecf8-57df-9ad78ec60ebf",
        "site_name": "My Home",
        "site_img": "",
        "device_type_list": [
          5
        ],
        "ms_type": 0,
        "power_site_type": 0,
        "is_allow_delete": false,
        "support_device_models": null,
        "current_site_device_models": null
      }
    ],
    "solar_list": [],
    "pps_list": [],
    "solarbank_list": [],
    "powerpanel_list": [
      {
        "device_pn": "",
        "device_sn": "PM6T2AS3Z62D3GEG",
        "device_name": "SOLIX Home Power Panel",
        "device_img": "https://public-aiot-ohi-prod.s3.dualstack.us-east-2.amazonaws.com/anker-power/public/product/2024/01/23/iot-admin/Qcjo9ALaRqFzFNeF/picl_A17B1_normal.png",
        "create_time": 0,
        "bind_site_status": "",
        "status": "",
        "wireless_type": "",
        "initial_process": 0
      }
    ]
  },
```

Furthermore the bind_devices show that only the Power panel is online to Wifi, but both F3800 are not.
On top, the wifi list output does not show any Wifi network, which is weird since the power panel is online to wifi. Then the wifi list output typically shows at least one wifi network with signal strength.

I guess there is something weird in your system setup. Maybe you have to delete that completely in the app, make sure that all then standalone devices are connected to Wifi and then recreate a new system and add all device again.

But that typically will scratch all your energy history data that is recorded with the system. However in your case that was empty as well, so maybe it does not hurt to recreate the system from scratch and see if it gets better

doublehelix46, 2024-07-07:
Hi @thomluther - I apologize for the delay (I was away for the US Holiday.)

I completely erased my Home; created a new Anker Account; first setup each F3800 on wifi as individual devices (and bound them to the new account); added the Home Power Panel to the new account/wifi > created the new home with the existing F3800. 

However, based on the export, looks like there's no difference in the system export. 

You'll notice in the below screenshot, that Anker's App disable's the wifi in F3800 devices when linked to a Power Panel. 
[image]

I can see all the devices listed in the Site_Detail.json, but you're right... the remaining scripts don't return the connected devices. It appears they're routing all data and management through the Home Power Panel.

[US Server Refreshed.zip](https://github.com/user-attachments/files/16120860/US.Server.Refreshed.zip)

thomluther, 2024-07-08:
Hi
It seems that's the way it is implemented.
Your site_list shows type 4 and the 'known' model types, which is only the Power panel

```
        "device_type_list": [
          5
        ],
        "ms_type": 1,
        "power_site_type": 4,
        "is_allow_delete": true,
        "support_device_models": [
          "A17B1"
        ],
        "current_site_device_models": [
          "A17B1"
        ]
```

The site_rules show all supported power site types and supported model types. For type 4 you see:
```
      {
        "power_site_type": 4,
        "main_device_models": [
          "A17B1"
        ],
        "device_models": [
          "A17B1"
        ],
        "can_empty_site": true,
        "quantity_min_limit_map": null,
        "quantity_max_limit_map": {
          "A17B1": 1
        }
      },

Type 4 seems to be the only site type supporting the power panel and there is not listed any other device model.
This is differently for other site types.
So I'm afraid there is no support for getting the PPS device infos in a power panel site type.
Furthermore the power panel iteself does not provide any usage values in the scene_info response, so there is nothing that can be usefully extracted for HA.
All data what you see in the app home screen must be merged it with the data pulled in realtime from the MQTT server, which just receives data when the App is open and triggers the devices to send their data to the MQTT server in realtime.
Eventually there are default sporadic data updates every 5 minutes or so like it is done now for Solarbank 2 systems. But key is that frequent data transmission is triggered only by main account watching the system. Nothing that the cloud Api can do in general.
```

doublehelix46, 2024-07-08:
Ahh that is unfortunate news, but I realize that's not your doing. I greatly appreciate you reviewing the issue as in-depth as you could. I hope this isn't the direction Anker takes moving forward.

Some additional notes that may be helpful in the future...

1) Yes, it seems the MQTT server is where the Power Panel is getting the bulk of the data via downloads (very minimal uploads).

[image] 

2) What is interesting though, and I don't want to say exactly counter to your theory above, is that it is regularly transmitting data to the MQTT server without the App. It's making calls for the server on a schedule which looks like every 30 minutes (and that's without the app open.) Each out reach looks the same, about 770kb uploaded... until the morning hours (once solar starts generating.) Now I'm seeing uploads greater than 1.5MB, which alludes that there are more energy stats in the upload as all of the overnight stats were very consistent. (Just a note, at this point, I haven't opened the app for the day - so this change was not prompted by app usage.)

[image]

3) I had been logged in with a sub account on my phone opposed to the owner account to avoid any issues while running the APIs, the sub account in the app gets all details, except for the PPS temp and system settings. This is true, even when I'm away from home not on my network. I was showing some friends and family the system and could show the real time stats in the sub account while not on the same network.

4) I just opened my app at 7:38am to see if it would make a call, and it did despite having just recently uploaded data in its cycle. It was slightly larger than the last upload, but nothing dramatically different. 

I wish you luck in your further developments - if you happen to stumble across anything PPS or Powerwall related, feel free to reach out any time. I'll do the same... off to explore MQTT now, haha (though I saw it wasn't fruitful in your past experience.)

thomluther, 2024-07-08:
Thx for the update. Yeah, MQTT is nothing I start with. The problem is that you will have to do reverse engineering how the data inside the requests are encryted with the EDHC mechanism. The api supports this data encryption too and its used via the app, the the api library does not use it currently for request data, only for the authentiation requests.
Its not clear how the request data header fields must be created to use enable encryption for requests. I assume same way it may be used for the MQTT api and you will have to find a way how to decrypt the data.
See #71 and #70 if you want to start with that journey.
If you get behind the required header fields for encryption and their content creation just let me know.

doublehelix46, 2024-07-09:
@thomluther - I might be delusional at this point, but I think I found the endpoint the PPS/Powerwall is using for energy statistics (and I don't think you're currently leveraging it) - at least for the US Server

https://ankerpower-api.anker.com/charging_energy_service/energy_statistics

[request.txt](https://github.com/user-attachments/files/16136905/request.txt)
[response.txt](https://github.com/user-attachments/files/16136915/response.txt)

Likewise there's a few others, but I haven't seen anything useful with them yet...

https://ankerpower-api.anker.com/charging_energy_service/report_device_data
https://ankerpower-api.anker.com/charging_energy_service/get_system_running_info
https://ankerpower-api.anker.com/charging_energy_service/get_installation_inspection
https://ankerpower-api.anker.com/charging_energy_service/get_rom_versions

doublehelix46, 2024-07-09:
Seems to be slightly different logic implemented in this endpoint compared to the Solarbanks... I believe our theory that everything is running through the Power Wall is correct. 

/charging_energy_service/energy_statistics = building the "Home Page" in the app.
[image]

POST request always contains the following:

"siteID": 
"end": (notes within dateType on when/how used)
"start": (notes within dateType on when used)
"global": false (so far I've only seen this set to false - not sure how it's used.)
"productCode": (so far I've only seen this null - not sure how it's used.)

**Numbers below correspond with the attached picture/UI elements**
dateType:

6. day (end: ""; start "yyyy-mm-dd")
7. week (end: "yyyy-mm-dd"; start: "yyyy-mm-dd") - runs Monday thru Sunday (not sure if that's required yet)
8. month (end: ""; start "yyyy-mm")
9. year (end: ""; start: "yyyy")

sourceType: 

1. solar
2. home
3. hes (this is the "home energy system")
4. grid

5. The "cumulative stats" are sitting in the **/charging_energy_service/get_system_running_info** endpoint

[image]

The POST request only sends "siteId" 

I will try and get some samples tomorrow of Request/Response for each page. I'd like to capture as much Solar/Battery/Grid behavior as possible in the responses.

I hope this helps!

thomluther, 2024-07-09:
> @thomluther - I might be delusional at this point, but I think I found the endpoint the PPS/Powerwall is using for energy statistics (and I don't think you're currently leveraging it) - at least for [...]

That is interesting indeed. Most of the endpoints I found start with power_service, your endpoints I never had on the radar.
Yes, the more examples you can provide, the easier an implementation can be.
In a first step it will be good to know all these endpoints and with which parameters they can be used. Then I can integrate them into the system_export tool in a first step to get a full snapshot of actual system data. From that point on, the meaning of the fields must be understood and described to use the properly for enhancing the Api cache structures and avoid clashes with existing fields.

Basically the power stats from the home screen should be different endpoint than the energy stats. Energy stats are more difficult to implement, since you need to run multiple queries to get all type of stats. Furthermore some data are only returned in the total of the queries time span, so to have those values on a daily base as used by HA, I need to query some types twice for today and yesterday to get all values for actual and past day.

doublehelix46, 2024-07-09:
Well, from what I can tell (and I think you mentioned this in another issue,) the real-time energy stats/display within the app are on the mqtt server and are merged into the daily energy stats every 5 mins to the cloud api. 

I attempted a MitM to monitor the data flows. When I included the MQTT port, not only did I fail to capture the data, but the real-time stats were effectively frozen until I removed the port from monitoring. Instantly the real-time stats loaded.

I think I found one or two more endpoints but need to double check your working list to be sure. I will update with examples as soon as possible.

thomluther, 2024-07-09:
I found following endpoints for the charging_service:
```
PPS and Power Panel related:
    "charging_energy_service/energy_statistics",  # Energy stats for PPS and Home Panel
    "charging_energy_service/get_system_running_info",
    "charging_energy_service/get_device_infos",
    "charging_energy_service/get_rom_versions",
    "charging_energy_service/get_error_infos",
    "charging_energy_service/get_wifi_info",
    "charging_energy_service/get_installation_inspection",
    "charging_energy_service/sync_installation_inspection",
    "charging_energy_service/get_utility_rate_plan",
    "charging_energy_service/report_device_data",
    "charging_energy_service/restart_peak_session",
    "charging_energy_service/preprocess_utility_rate_plan",
    "charging_energy_service/ack_utility_rate_plan",
    "charging_energy_service/get_configs",
    "charging_energy_service/adjust_station_price_unit",
    "charging_energy_service/sync_config",
    "charging_energy_service/get_sns",
    "charging_energy_service/get_world_monetary_unit",
```
They do not work on the EU Api server with my account however:
Api Request Error: 503, message='Service Temporarily Unavailable'
The power_service endpoints work there...weird. Maybe the servers have different versions of the Api, otherwise I would have expected another type of response error.

You will have to explore them manually and figure out required parameter and response structures. Anything that is meaningful and working could I then add in the system_Export tool for the beginning.

doublehelix46, 2024-07-09:
Hi @thomluther -

I'm still testing calls and compiling the results... but out of curiosity... do you see any additional endpoints within: /charging_hes_svc ?

I've found /charging_hes_svc/get_device_product_info - but it appears to just be a list of Anker products (ironically none in the US.

thomluther, 2024-07-09:
Indeed, another whole set of endpoints you can explore.
```
    "charging_hes_svc/ota",
    "charging_hes_svc/check_update",
    "charging_hes_svc/get_installer_info",
    "charging_hes_svc/download_energy_statistics",
    "charging_hes_svc/get_wifi_info",
    "charging_hes_svc/update_wifi_config",
    "charging_hes_svc/get_device_product_info",
    "charging_hes_svc/report_device_data",
    "charging_hes_svc/get_auto_disaster_prepare_status",
    "charging_hes_svc/get_install_info",
    "charging_hes_svc/get_device_pn_info",
    "charging_hes_svc/get_device_card_list",
    "charging_hes_svc/get_device_card_details",
    "charging_hes_svc/get_system_running_time",
    "charging_hes_svc/get_system_profit_detail",
    "charging_hes_svc/get_tou_price_plan_detail",
    "charging_hes_svc/get_electric_utility_and_electric_plan_list",
    "charging_hes_svc/get_station_config_and_status",
    "charging_hes_svc/get_current_disaster_prepare_details",
    "charging_hes_svc/adjust_station_price_unit",
    "charging_hes_svc/update_hes_utility_rate_plan",
    "charging_hes_svc/quit_auto_disaster_prepare",
    "charging_hes_svc/restart_peak_session",
    "charging_hes_svc/check_function",
    "charging_hes_svc/remove_user_fault_info",
    "charging_hes_svc/user_event_alarm",
    "charging_hes_svc/sync_back_up_history",
    "charging_hes_svc/cancel_pop",
```
At least the 'charging_hes_svc/get_device_product_info' is also working for my account, can start looking at the others

thomluther, 2024-07-09:
Examples that work for my account but not with my devices. I also just tried with shared account that has no permission to devices, therefore I may see many permission errors:
```
await myapi.request("post", "charging_hes_svc/report_device_data",json={"siteIds":[siteId]}))
(200104) Anker Api Error: The installer system does not exist

await myapi.request("post", "charging_hes_svc/get_installer_info",json={"siteIds":[siteId]}))
(160003) Anker Api Error: The device has no access permission.

await myapi.request("post", "charging_hes_svc/get_wifi_info",json={"sn":deviceSn}))
  "data": {
    "ssid": "",
    "rssi": 100
  },

await myapi.request("post", "charging_hes_svc/get_install_info",json={"siteId":siteId}))
  "data": {
    "installLocation": "",
    "installEnv": ""
  },
```

Hints for exploring: 
- use the owner account to have full access to your devices
- Include at least one Json parameter, e.g. "siteId":siteId to get a more meaningful error response that might reveal which parameter is missing.

doublehelix46, 2024-07-09:
> I found following endpoints for the charging_service:
>
> ```
> PPS and Power Panel related:
>     "charging_energy_service/energy_statistics",  # Energy stats for PPS and Home Panel [...]

My initial pass at the charging_energy_service endpoints. I think there's a lot of overlap between these and the new ones, but I'll continue exploring and expanding. 

```
PPS and Power Panel related:
    "charging_energy_service/energy_statistics",  # Energy stats for PPS and Home Panel
    "charging_energy_service/get_system_running_info", # Cumulative Home/System Energy Savings since Home creation date
    "charging_energy_service/get_device_infos", # Home Panel Info
    "charging_energy_service/get_rom_versions”, # Check for firmware update and download available packages
    "charging_energy_service/get_error_infos", # Unknown at this time
    "charging_energy_service/get_wifi_info", # Displays WiFi network connected to Home Power Panel
    "charging_energy_service/get_installation_inspection", # Unknown at this time - appears to say which page last viewed on App
    "charging_energy_service/sync_installation_inspection", #Unknown at this time
    "charging_energy_service/get_utility_rate_plan",
    "charging_energy_service/report_device_data",
    "charging_energy_service/restart_peak_session",
    "charging_energy_service/preprocess_utility_rate_plan",
    "charging_energy_service/ack_utility_rate_plan",
    "charging_energy_service/get_configs",
    "charging_energy_service/adjust_station_price_unit",
    "charging_energy_service/sync_config",
    "charging_energy_service/get_sns", # Displays Serial Numbers of attached PPS in Home
    "charging_energy_service/get_world_monetary_unit",

# /charging_energy_service/get_system_running_info Request 

{
    "siteId": "932a71aa-i1b3-4f0c-a232-nj9023c02bab"
}

# /charging_energy_service/get_system_running_info Response

{
    "code": 0,
    "msg": "success!",
    "data": {
        "connect_infos": {
            "PowerPanelSN": true
        },
        "connected": true,
        "total_system_savings": 0,
        "system_savings_price_unit": "$",
        "save_carbon_footprint": 52.18,
        "save_carbon_unit": "kg",
        "save_carbon_c": 0.997,
        "total_system_power_generation": 52.34,
        "system_power_generation_unit": "KWh"
    },
    "trace_id": “redacted”
}

# /charging_energy_service/get_sns Request

{
    "main_sn": "PowerPanelSN",
    "macs": [
        "F38001MAC001",
        "F38002MAC002"
    ]
}

# /charging_energy_service/get_sns Response

{
    "code": 0,
[... 226 lines cut ...]
                "introduction": "1. Fixed a bug. \n2. Optimized some functions."
            }
        ]
    },
    "trace_id": “redacted”
}

# /charging_energy_service/get_error_infos Request

{

}

# /charging_energy_service/get_error_infos Response

{
    "code": 0,
    "msg": "success!",
    "data": {
        "error_infos": []
    },
    "trace_id": “redacted”
}

```

doublehelix46, 2024-07-10:
> charging_hes_svc/get_device_card_details

This service appears to be used for the X1 series. The bulk of these APIs I don't have access to or return empty responses as you experienced.

Few exceptions:

charging_hes_svc/download_energy_statistics # Creates a CSV file export of all energy stats for selected time period

`{
    "siteId": "932a71aa-i1b3-4f0c-a232-nj9023c02bab",
    "end": "",
    "start": "2024-07-09",
    "global": false,
    "productCode": "",
    "dateType": "day",
    "sourceType": "solar"
}`

Response provides a link to a file: 

`
{
    "code": 0,
    "msg": "success!",
    "data": {
        "fileUrl": "https://edge-aiot-ohi-prod.s3.dualstack.us-east-2.amazonaws.com/anker_power/edge/station_statistics_data/truncated",
        "fileName": "Energy Data.csv"
    },
    "trace_id": "redacted"
}
`

charging_hes_svc/get_wifi_info works exactly the same as charging_energy_service/get_wifi_info;

the charging_hes_svc introduces a new key on "station_id" opposed to "siteId" (though I will tell you the formatting has been inconsistent across the endpoints (stationId, station_id, stationID)

I dought I'll find much more in these services, but will loop back to the charging_energy_service and report back anything useful.

thomluther, 2024-07-10:
Yes, It appears the *hes* endpoints are for managed power services (home energy services?), since they can address multiple sites in the siteIds parameter. End consumer probably should use the charging_power endpoints.
It might be a way for external services to look end customer site data and modify them if they are listed as installer for the site...

doublehelix46, 2024-07-12:
I've continued to monitor some activity in the app and documenting the params - I'll have some updates in the near future...

However I've noticed the 
/power_service/v1/site/get_scen_info endpoint is in fact still being used for the PowerPanel setup, but it is an encrypted body request/response.

The only thing I'm able to determine so far is that the requests all start with an epoch timestamp that has been bas64encoded. The rest of the request is encrypted using other methods I have yet to determine.

thomluther, 2024-07-12:
This request and response encryption is done by the app, but can also be used without encryption.
The login response uses EDHC encryption and this method is probably also used for the request data encryption, but there are additional header fields used in that case and I have not clue how they are generated, see #70

doublehelix46, 2024-10-16:
Hi @thomluther -
I dug in as best I could, but understanding you made some changes I'm happy to go through the proposed exercises. I can add one of your users to my system as a shared user. Which email address would you like me to provision?

thomluther, 2024-10-18:
> Hi @thomluther - I dug in as best I could, but understanding you made some changes I'm happy to go through the proposed exercises. I can add one of your users to my system as a shared user. Which email address would you like me to provision?

I registered a new alias with Anker which you can use for the invite: lollygames2@gmail.com

doublehelix46, 2024-10-18:
@thomluther - Great, an invite has been sent. I would wait about 2 hours for my solar to activate for the day.

thomluther, 2024-10-18:
> @thomluther - Great, an invite has been sent. I would wait about 2 hours for my solar to activate for the day.

Great, I accepted and that works, can see the home page of your system in read only mode as usual for shared accounts. But this is sufficient for the time being to start testing some queries against it and see if the power values are somewhere in the cloud and how to extract the energy values that are accessible for shared accounts as well.
I also checked again the actual site rules and as mentioned before, the site type with a power panel only supports the power panel device, but not the F3800 devices, which is weird compared to other site types...
But that might be the reason, why the F3800 consumption and SOC values are not reported to the cloud at all...
We will see which data can be pulled from power panel system types and how that can be integrated into the existing Api library structures to pick it up in the HA integration...

thomluther, 2024-10-21:
Hi @doublehelix46 , some updates on this:

- I noticed in the app, that the home screen needs a couple of seconds to show the power consumption. Sometimes I even have to restart the App, to get them. I assume the reason is, that those are really not available on the cloud Api, but the App must trigger the F3800's to push their data to the mqtt server, so they can be merged into the home page in real time (about every 5-10 seconds). So it seems sometimes this trigger does not work when I just login to this user profile and the App must be restarted to get the mqtt trigger to work...
- My conclusion on this: We cannot get the F3800 power values from the cloud Api, not even when they are configured into a power panel system (since the power panel system does weirdly only support the power panel model type alone, but not the F3800 model types. So they behave the same as any other Portable Power Supply model. Those devices publish their data ONLY to the MQTT servers when they are asked to do so (by the Anker app)... Someone would need to explore the mqtt server usage, a possible interface to it, how devices can be triggered to push their data, how communication is encrypted and how the mqtt servers could be utilized to grab the real time data. I'm not digging into the mqtt server topic by any means, since I have neither experience nor time to do that.
Attention: Triggering devices to publish data to the mqtt server 24x7 may increase their power consumption significantly, which may be a drawback for any battery device and noticeably decrease its efficiency. For that and other reasons, they probably push their real time data ONLY when queried by the App. I also don't think that the Anker power devices support local MQTT, since that is not foreseen in Anker's power device architecture. While saying that, the Anker Smart Meter may be an exception, since this also transmits real time data to the Solarbank 2 while internet is down as long as both are in the same WLAN. But this can also be realized via propriatary direct encrypted interfaces that will be established once the smart meter is configured into a SB system.
- I added a lots of additional Api andpoint queries to the export class. As shared user, you can at least query the energy stats of a power panel system now. Those could be added to the Api library once the overall information architecture of the charging_service endpoints is better understood.
- A power panel system owner can run more endpoint queries. **Please use the new system_export tool from the main branch to run create a new anonymized export from your system** You can create with default prompts (auto discover mode). I need to see, whether the minimum required payload parms are OK for those queries, and what information is returned and if all sensitive data is randomized. Since I believe the charging endpoints do not add too much new information, those could be incorporated to the existing Api class and data poller routines if valuable for power panel type systems (seems to be the only system type so far where they are used)
- I added also queries for HES systems to the export class. Those will be used for X1 systems (which typically must be installed by professional installer, who will also initialize the system in the professional app for customer use by the consumer App? The data structure and queries for HES systems may be pretty different and may require a new Api class to handle such systems...
- I added a new query to get latest OTA versions for any device. This might be usable to see latest available OTA versions and properly flag whether updates are available for a device. I need to get more examples of those response to see whether this is reliable (best case when an update is available for a device), and also how structure with children devices will look like (e.g. battery packs for Power Systems could be listed in the children structure)

thomluther, 2024-11-07:
Hi @doublehelix46 
Did you have a chance to run the system export tool with your owner account?

Here is what I implemented so far which is also visible in the solarbank monitor tool. That is from the live query as member account  with anonymized IDs.
```
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1, Devices: 1
--------------------------------------------------------------------------------
Device         : SOLIX Home Power Panel  Alias         : SOLIX Home Power Panel
Serialnumber   : ABCDEF12345567890        Admin         : NO
System         : My Home  (Site ID: 12345678-abcd-efgh-ijkl-123456789012)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade  : Unknown (OTA Unknown)
Cloud Status   : online                  Status code   : 1
--------------------------------------------------------------------------------
Energy details for System My Home (Site ID: 12345678-abcd-efgh-ijkl-123456789012):
Total Produced :    2.85 MWH             Carbon saved  :    2.84 T
Max savings    :   358.3 $               Price kWh     :   --.-- 
Today          : 2024-11-07              Yesterday     : 2024-11-06
Solar Energy   :   0.00 kWh              Solar Energy  :  16.38 kWh
Charged        :   0.00 kWh              Charged       :  12.02 kWh
Charged Solar  :   0.00 kWh              Charged Solar :  12.02 kWh
Charged Grid   :   0.00 kWh              Charged Grid  :   0.00 kWh
Discharged     :   2.11 kWh              Discharged    :   5.90 kWh
House Usage    :   2.12 kWh              House Usage   :  10.05 kWh
Solar Usage    :   0.00 kWh              Solar Usage   :   3.73 kWh
Battery Usage  :   2.11 kWh              Battery Usage :   5.90 kWh
Grid Usage     :   0.00 kWh              Grid Usage    :   0.42 kWh
Grid Import    :   0.01 kWh              Grid Import   :   0.43 kWh
Grid Export    :   0.00 kWh              Grid Export   :   0.63 kWh
Sol/Bat/Gri %  :   0/100/  0 %           Sol/Bat/Gri % :  37/ 58/  5 %
--------------------------------------------------------------------------------
Api Requests: 17 last hour, 17 last minute
Site refresh:  16 sec,  Device details refresh: 217 sec  (CTRL-C to abort)
```
However, the shared account has limited capabilities and only subset of information that might be possible.
While we haven't found yet the power values, at least the energy consumption is now available:

I enhanced the daily energy stats in the cached api object to be common with the fields and units as used already for solarbank systems. Additional fields have been added.
The advantage of a common structure in the cache is that the HA integration can easily reuse the defined entities and just need to be enhanced for the new ones. A few may also required a breaking change in the current naming, to make it more common and hopefully also usable for other storage systems like X1.

The app is showing more percentages, however I limit the percentage share only for the home consumption view. The percentages 
actually reflect the share of energy sources for the home consumption, which is probably most relevant.
The shares how battery was loaded or how solar was used can be calculated by the energy values if really needed. I hope the naming makes it clear what the value is about (In HA the naming may be slightly different and will also be translated to other languages)

Last but not least, I noticed in the App that there is a diagram for SOC over time and I wonder where this data is coming from.
The energy stat response has fields for it, but in my weekly type requests that I do for today and yesterday, those fields are always empty:
```
    "totalEnergy": "37.23",
    "totalEnergyUnit": "KWh",
    "totalImportedEnergy": "",
    "totalImportedEnergyUnit": "",
    "totalExportedEnergy": "37.23",
    "totalExportedEnergyUnit": "KWh",
    "power": null,
    "powerUnit": "",
    "chargeLevel": null,
```
Interestingly there is also power field and I wonder whether there may be a combination of paramters to get the actual power or charge level of the F3800s...

thomluther, 2024-11-07:
@doublehelix46 The energy stats when used with the "dateType": "day" parameter will provide large lists for the power and chargeLevel fields.
However, these lists cover full day in 5 min intervals, so are really large.
But they seem to provide power and SOC values over time, depending which "sourceType" is used.
Potentially those queries could be used to report Power and SOC value, but power will be roghly in kW only, and refresh max. in 5 min intervals.
However, due to the hugh amount of data contained in those reponses, I'm reluctant to implement a regular power /soc refresh on those queries.
I haven't found yet parameters to limit the response just to e.g. last hour of the day. Do you know how that could be achieved?

thomluther, 2024-11-07:
I did nore testing on it and usability is limited for power values. Some caveats:
- The time stamp is the local timestamp. For local usage of the Api, there should not be a time gap between the Api client and the time as recorded in the cloud. However, in my testing I have to consider timezone offset when looking for the last used timestamp in your data.
- The usage is delayed by about 2-3 minutes. That means the 10:00 timestamp values will be filled only by about 10:02 - 10:03.

As a summary, it would be possible to extract some data from the energy stats, but you have to consider it as the last valid data point that you see in the App diagrams. It is delayed by 2-3 minutes, its rounded to kW with 2 decimals and it is averaged by 5 minute intervals. So I'm not sure whether it makes sense to report them as power values, since that could raise more questions than necessary.

doublehelix46, 2024-11-08:
Hi @thomluther -

I apologize for the delay I've been traveling, but I will absolutely run the extracts and tests with the Owner account. I recall most things being very similar between the Master/Sub accounts... only exception seemed to be more in the setup/device calls. 

I agree, I had not personally found a way to get the real time stats - I'm fairly confident (and to your earlier comment) that the MQTT server handles the energy stats for the Power Panel. I did a lot of testing with the energy stats. I'll go back to my statements and see if I was able to get down to a specific time interval to reduce the response load.

I'll think through the solution further, but for me personally, even having the daily totals of Solar generated, Energy Used (and where - Load vs. Charging Battery) and then the depletion of the battery to plug into the Energy Monitor Dashboard is a huge plus. I happen to have a Sense energy monitor but I lose insight when leveraging the Home Panel to power the load when solar is not being generated. 

I recognize that's my specific use case - but I will do my best to figure out if there's broader value.

Thank you for testing!

thomluther, 2024-11-12:
@doublehelix46 

I added some code to extract the last 5 min avg values from todays energy stats. It was not as simple as I thought, because I want to limit the queries to once every 5 minutes max (more often does not make sense). On the other hand, I want to query the last average stat once it becomes available in the cloud to avoid the delay is too big.
Therefore at the beginning there are queries in each update interval until the change to a new datapoint is recognized and from then its limited to 5 minutes at most.

Here is what I get from your system in the monitor:
```
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 1, Devices: 1
--------------------------------------------------------------------------------
Device         : SOLIX Home Power Panel  Alias         : SOLIX Home Power Panel
Serialnumber   : AZCBDFGH12345678        Admin         : NO
System         : My Home  (Site ID: <site-id>)
Wifi SSID      :                        
Wifi state     : Unknown                 Signal        :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade  : Unknown (OTA Unknown)
Cloud Status   : online                  Status code   : 1
Last Check ⌀   : 2024-11-12 22:36:11     Valid before  : 2024-11-12 16:39:58
Solar Power ⌀  : 0.00 KW                 Battery SOC   :   95 %
Charge Power ⌀ : 0.00 KW                 Discharge ⌀   : 0.43 KW
Home Usage ⌀   : 0.43 KW                 Grid Import ⌀ : 0.00 KW
--------------------------------------------------------------------------------
Energy details for System My Home (Site ID: <site-id>):
Total Produced :    2.94 MWH             Carbon saved  :    2.93 T
Max savings    :   372.5 $               Price kWh     :   --.-- 
Today          : 2024-11-12              Yesterday     : 2024-11-11
Solar Energy   :  17.89 kWh              Solar Energy  :  12.85 kWh
Charged        :  11.54 kWh              Charged       :   8.37 kWh
Charged Solar  :  11.26 kWh              Charged Solar :   7.79 kWh
Charged Grid   :   0.28 kWh              Charged Grid  :   0.58 kWh
Discharged     :   0.64 kWh              Discharged    :   3.96 kWh
House Usage    :  13.69 kWh              House Usage   :  17.45 kWh
Solar Usage    :   6.09 kWh              Solar Usage   :   5.01 kWh
Battery Usage  :   0.64 kWh              Battery Usage :   3.96 kWh
Grid Usage     :   6.96 kWh              Grid Usage    :   8.48 kWh
Grid Import    :   7.24 kWh              Grid Import   :   9.06 kWh
Grid Export    :   0.54 kWh              Grid Export   :   0.05 kWh
Sol/Bat/Gri %  :  44/  6/ 50 %           Sol/Bat/Gri % :  28/ 24/ 48 %
--------------------------------------------------------------------------------
Api Requests: 25 last hour, 25 last minute
```
So you see, including energy stats, there will be lots of initial queries, but that should go down to 3-4 per normal refresh after a while.
I compared the values with the App and that seems what I can get so far.
If you provide an export zip, I can see if there is more available and all sensitive data will be randomized.
Than I can start adding the powerpanel support to the HA integration.

doublehelix46, 2024-11-28:
Hi @thomluther -

I'm so sorry for the long delay. Attached is the export from the System Admin as requested. I still need to play around with the energy stats and the payloads to try to limit the data pull. I will do my best to get another update out to you next week.

Thank you again for your patience. Best,
Frank

[SystemAdminExport112724.zip](https://github.com/user-attachments/files/17941589/SystemAdminExport112724.zip)

thomluther, 2024-11-28:
Hi @doublehelix46 
I found in the export log that charging queries were stopped on first parameter error. I fixed that in the export class.
Now all charging endpoint queries I have coded should be run, even if they may still hit parameter errors that will have to be fixed in the code.
For the energy query data reduction I'm not sure if that is possible. Since this response is basically what the App charts are displaying for today, and the chart always wants the full day data I guess since that is the default interval in the app.
According to the file sizes, each refresh for the Power Panel power values (5 min average) will receive about 225 KB data in total for required 4 energy queries. Initially this average power refresh method will be run by Home Assistant for each update interval until the minimum offset to last valid datapoint was found. From there, the query will run at most every 5 minutes when the next valid data point is expected.
So with default refresh interval of 60 seconds, in first hour there will be ~17 data refreshes to get latest average power (3,8 MB), and 12 refreshes in subsequent hours (2,7 MB). And this is only to get the Power Panel average power values. But the other queries do not return such large responses, so they can be ignored.
In HA, if those power average queries will cause issues or too much data traffic, you can exclude the power panel energy stats category in the configuration options.

doublehelix46, 2024-12-02:
Hi @thomluther - I reran the export after your updates. Providing a copy for reference.

Thank you again for your support!

[SystemAdminExport_2024-12-02_1638.zip](https://github.com/user-attachments/files/17984311/SystemAdminExport_2024-12-02_1638.zip)

thomluther, 2024-12-03:
Hi @doublehelix46 
Unfortunately there is no additional info or there are still missing parameters with the queries I used.
So this is a pretty cumbersome process. I suggest you explore more of the charging_service Api endpoints for your system with the owner account and see if there are useful queries to get more details or to change any system settings via any of the Api endpoints. If you can sniff the App requests somehow, that might help to see how the cloud api requests are structured.

I think following endpoints may be worth further exploration:
- charging_energy_service/get_utility_rate_plan
- charging_energy_service/ack_utility_rate_plan
- charging_energy_service/preprocess_utility_rate_plan
- charging_energy_service/report_device_data
- charging_energy_service/get_configs
- charging_energy_service/restart_peak_session
- charging_energy_service/adjust_station_price_unit

I don't think the charging_hes_svc endpoints are relevant for your power panel system. since many of them fail too in your setup.
What you might explore too are standard power services endpoint queries against your F3800 SNs, since they are typically done by the export class only when those SNs are part of the system. However, for power panel systems, they appear to be stand alone devices since they are not listed in the scen_info response.
So if you find more data for your system or devices on the cloud server and document the parameters and possible responses, we can add that to the library too. One first easy thing could be the energy price which can probably changed with the  charging_energy_service/adjust_station_price_unit endpoint. But I haven't seen a query or response to list the actual station price.

thomluther, 2025-03-10:
@doublehelix46 

I found additional endpoints that you may explore against your power panel or PPS devices. Not sure what they are used, but when a device sn parameter is required you typically need to use the owner account to avoid permission errors.

Endpoints
```text
    mini_power/v1/app/charging/get_charging_mode_list
    mini_power/v1/app/charging/update_charging_mode
    mini_power/v1/app/charging/add_charging_mode
    mini_power/v1/app/charging/delete_charging_mode
    mini_power/v1/app/setting/set_charging_mode_status
    mini_power/v1/app/egg/get_easter_egg_trigger_list
    mini_power/v1/app/egg/add_easter_egg_trigger_record
    mini_power/v1/app/egg/report_easter_egg_trigger_status
    mini_power/v1/app/setting/get_device_setting
    mini_power/v1/app/power/get_day_power_data
```

Example how you can try them with the Api library request method:
```python
_out(await myapi.apisession.request("post", "mini_power/v1/app/charging/get_charging_mode_list",json={"device_sn": deviceSn}))
```

Following endpoints may also work with shared account, but not sure what kind of information they are showing:
```
charging_disaster_prepared/get_site_device_disaster
charging_disaster_prepared/get_site_device_disaster_status
charging_disaster_prepared/set_site_device_disaster
charging_disaster_prepared/quit_disaster_prepare
charging_disaster_prepared/get_support_func
```
Example how you can try them with the Api library request method:
```python
_out(await myapi.apisession.request("post", "charging_disaster_prepared/get_site_device_disaster",json={"identifier_id": siteId, "type": 2}))
```

Examples:

`POST https://ankerpower-api.anker.com/charging_disaster_prepared/get_site_device_disaster_status {'identifier_id': '<site_id>', 'type': 2}`
```json
  "data": {
    "auto_disaster_status": 2,
    "manual_disaster_status": 2,
    "current_disaster_detail": null
  }
```

`POST https://ankerpower-api.anker.com/charging_disaster_prepared/get_site_device_disaster {'identifier_id': '<site_id>', 'type': 2}`
```json
  "data": {
    "auto_disaster_switch": true,
    "manual_disaster_switch": true,
    "disaster_details": [
      {
        "uuid": "<id masked>",
        "disaster_type": 1,
        "event": "",
        "start_time": 1739754540,
        "end_time": 1739840940,
        "charging_time": null
      }
    ]
  }
```

`POST https://ankerpower-api.anker.com/charging_disaster_prepared/get_support_func {'identifier_id': '<site_id>', 'type': 2}`
```json
  "data": {
    "support_auto_disaster": true,
    "support_country_code": [
      {
        "code": "US",
        "google_code": "<code masked>"
      }
    ]
  }
```

It seems that only type 2 works.

thomluther, 2025-08-07:
> Are there any simple example of getting a single parameter ?

No, the easiest way is you run the poller methods as done by the monitor and then you can extract any value from the Api chaches as done by the monitor.