thomluther/anker-solix-api #119: Question: How resilient is Anker Solix stuff? Concerns regarding constant writes..

nlincke, 2024-07-28:
Hey,

i am a new owner of a Solarbank 2 and i am looking forward to see the integration progress.
I was wondering if any of you have any knowledge or just an option regarding the resilience of the anker stuff.

Imagine that we have the ability to steer the traget outpot watts and can adjust it as required. This might (assumption) trigger constant rewrites on a system that has a limited number of write cycles... (e.g. nvram) it could kill my battery/inverter and i am not sure if the 10 year warranty would cover this....

I know that the anker app offers a intelligent mode that uses an extra device for the fuse box to minimize the supply of power to the grid . Has anyone experience with this? I would like to use the smart meter i am going to be forced to use from my power company... and not add an extra device to my network...

Thanks!

thomluther, 2024-07-28:
I cannot say for sure how the changes are stored, but would be surprised if they go to NVRAM since the solarbank is battery powered all the time. It never goes below its reserve, so why not storing this in normal RAM.
On top, the schedule is probably also stored in the cloud, so in case Solarbank lost any of its schedule settings, it could probably be recovered after reconnecting to the cloud.
But again, not sure whether this goes to RAM or NVRAM, and NVRAM could also be rewritable multiple times, since that may also be battery backed up RAM which the Solarbank can easily do...

What you probably refer to is EEPROMs which are limited in write cycles like in inverters, which keep the persistent settings. Those should not be used too frequently of course, but no idea if such are used in the Solarbank too which would be changeable through the App.
The integration does no different changes than the App does, it all goes via the cloud while the App can even do more via Bluetooth and MQTT cloud settings on the devices directly.

But no warranty, therefore usage of the Api is at your own risk.

I can just say, that I control the SB1 via Schedule changes all the time. But when you do something like that, of course you should limit the changes as much as possible. Changing it every couple of seconds does not make sense.

nlincke, 2024-08-02:
Thanks for your answer. 
Two days ago they introduced the new mode with the power plug mode. With this the behavior changed and the default 200w power delivery target was causing the system to only deliver in case there are 200w to deliver. 

Before that App update, it was delivering what ever solar power was available. Even if it was below 200w....
This let me think that the steering is done via the cloud and perhaps stored within the RAM of the device only.
I added an explicit 200w delivery schedule via the app and it was delivering again what was available by the cells.

But the fact that i can not change the profiles while the device is in hibernation during the night raises some new doubts.
I was also thinking about the BT interface but i have not tried it jet. 

In what cases do you change your schedule? Are you trying to reduce the amount of power you are delivering to the public grid by changing the schedule?

thomluther, 2024-08-03:
Basically every change in any of the schedule attributes requires to apply a complete new schedule.
Even if you just want to change the actual output power, you have to change that value in the actual time slot of the existing schedule (or create a new time slot if none exist yet) and send always the WHOLE schedule to the cloud.
With SB2 it is the same. Even the usage mode is part of the schedule object, even if it is a general attribute independent of the time slots and weekdays. But it can be changed only by sending the whole schedule structure to the cloud.
So it does not matter what you change, since a modified schedule must always be provided to apply any change

nlincke, 2024-08-04:
Understood. Do you use a HA automation to adjust your schedule etc or do you make use of e.g. the smart meter from Anker directly?

I am still trying to decide if I should create something in HA or install the Anker Smart Meter...

At the moment, I am not in a rush since my meter is running backwards... :)
Thanks for the great work by the way!

thomluther, 2024-08-04:
@nlincke 
I use these automations that I documented in the HA integration repo
[Surplus charge automation for Solarbank E1600 (1st generation)](https://github.com/thomluther/hacs-anker-solix/discussions/81)
For manual adjustments they might be usefull as well for the SB2 without the need to adjust the Export switch.
But the other automations might be useful with required adjustements for SB2 behavior.
You also don't have to split it to daytime and nighttime behavior maybe, since all is controlled via SB2 only and not also via the inverter as in my case.
Anyway, its not for HA beginners, but they might give you some automation ideas for SB2 as well....

And many thanks for your support!

closed 2024-08-09 by thomluther