thomluther/anker-solix-api #123: Question: Is it possible to use this API by prometheus and scrape data to Grafana. My device is Anker Solix C800X

KristianLGeorgiev, 2024-08-05:


thomluther, 2024-08-05:
Hi, not sure if you can use the Api for that.
But you should be aware, that the Anker Api cloud does not seem to track any usage data for any of the portable power station models so far.
So I don't think you get any data out of the cloud Api, You would need to write an interface for the anker mqtt servers, but I have no clue how that works.
See #71

thomluther, 2024-11-20:
Hi @kantmn 
I'm pretty sure that this is possible. But I'm no developer and I don't know any of the tools you mention, nor do I know how to provide a web service, which I think requires to run a instance as well.
This is just a python library, with the main purpose to be used by the HA integration.
If anybody is willing to fork and expand it with those features, feel free to do so. But that is nothing that I could test or maintain.

kantmn, 2024-11-20:
hi @thomluther 
i had no patience and had to try it out. and i think it works. 
the attached script contains your api calls and a weather api i use, if not needed just delete those parts

the script now also offers a background webservice running on port ip:5000/metrics to serve a prometheus.

if anyone needs help in setting up let me know.
[Uploading api_2_prometheus_webservice.txt…]()

thomluther, 2024-11-20:
Hi, thanks for the script.
But it is nothing to include out of the box. To include a method for the flatting part of the api classes dict caches, it must be in a common format to work with all type of sites and devices.
In the recent 2.3.0 version I will add also an account dictionary to the hierarchy, not sure if that should be included in a common flattening method as well.
But as mentioned previous, actually I have no time or ressources to dig into this.
You can maybe prepare to separate the flattening method to add that to the Api Base class in apibase.py, since that is used by any api class that exists now and may still be implemented for various site types.
The rest could be going into an executable tool like the solarbank monitor and others.
I suggest you make those changes as pull request to the repo, so they can be merged once you tested them against various systems. For testing against other systems, you can use the fromFile parameter when collectiing Api data. Before you have to set test folder of your api instance to any example json extract that you want to test against. Some examples are in the lib. Other export examples are attached in various issues.

InDieTasten, 2025-03-26:
@kantmn Just so you are aware, it seems like you've left your anker credentials as well as your gps coordinates for weather service in the attached python script!

I recommend rotating the password asap, if you haven't done so already.

kantmn, 2025-03-26:
> [@kantmn](https://github.com/kantmn) Just so you are aware, it seems like you've left your anker credentials as well as your gps coordinates for weather service in the attached python script!
> 
> I recommend rotating the password asap, if you haven't done so already.

thanks for letting me know, the password was change a long while ago, but still better to cleanup things
dunno why i forget that

i have updated the whole thing anyway, it is now a docker container which can be found here
https://github.com/kantmn/anker-solix-docker-api
i had not so much time to document it