thomluther/anker-solix-api #124: Add support for smart plugs

thomluther, 2024-08-07:
The new smart plugs are announced.
New system exports with installed smart plugs are required to understand which data they deliver through the cloud and what capabilities exist for the plugs.
[Anker Smart Plugs](https://www.anker.com/eu-de/products/a17x8311)

thomluther, 2024-08-22:
Beside required full system exports with installed smart plugs showing power consumptions, exports for following conditions are required:
- Power Usage mode is set to Smart Plugs
- Smart Plugs have defined schedule and or delayed toggle (Not clear if this is included in actual system_export data)

thomluther, 2024-08-22:
Exploration of various endpoints by smart plug owners is required to verify if following is possible via the cloud Api:
- Query the switch state of a smart plug (most important)
- Change the switch state of a smart plug (equally important)
- Query (and change) the schedule of a smart plug
- Change the tag / label of a smart plug

thomluther, 2024-09-12:
Note: The smart plugs can be supported probably only when added to a power system such as created for solarbanks.
Stand alone Plugs probably do not send their data to the cloud api. They may however provide their state to the cloud. This has to be explored by owners which endpoints can be used to query and change their state.

rwe87, 2024-10-17:
Hi, I have the Solarbank 2 E1600 Pro and smart plugs running. How can I provide the info you need?

thomluther, 2024-10-17:
@rwe87, thanks for you export in #115 
The smart plug stuff should be discussed here for the smart plug support, since more exports and Api testing from you may be required.

Some questions to the exported data:

1. "home_usage_total": "1.72", from Energy export of "time": "2024-10-16",. Since you have no Smart meter, is that the energy consumed by the smart plug only? Your system has no other measure yet to declare home usage energy, only smart meter or smart plug energy can used for home usage calculations...
2. Export of home usage energy also shows following which is weird, if the 1.72 kWh are from the smart plug:
```
    "home_usage_total": "1.72",
    "solar_total": "",
    "trend_unit": "kwh",
    "battery_to_home_total": "0.85",
    "smart_plug_info": {
      "smartplug_list": [
        {
          "device_pn": "",
          "device_sn": "6IE9JIDE9ZWSAHO8",
          "device_name": "Waschmaschine",
          "device_img": "",
          "create_time": 0,
          "current_power": "",
          "tag": "",
          "tag_img": "",
          "total_power": "0.00",
          "bind_site_status": "",
          "status": "",
          "wireless_type": "",
          "err_code": 0
        }
      ],
      "total_power": "0.00"
    },
```
The smart plug details in energy export is pretty much the same than the device details that are supposed to show consumption values, however those are always show 0 in the exports. Do you see anything shown in HA? Does the App show something different for the plugs? I wonder whether the smart plug consumption is properly reported to the cloud, since I haven't seen W values in the scene_info yet...

If you are looking for support of the plug switch or schedule, you will have to use the Api library and customize a pyton test program according the example, so you can test various queries with various parameters against your plugs. When you test you will have to use your main account with the Api library. Eventually #107 will help for querying and changing the plug device attributes

rwe87, 2024-10-18:
Hi @thomluther,

> 1. "home_usage_total": "1.72", from Energy export of "time": "2024-10-16",. Since you have no Smart meter, is that the energy consumed by the smart plug only? Your system has no other measure yet to [...]

This can hardly be the energy consumption of the smart plug because I installed the plug just before I did the export. The plug is connected with a washing machine that has not been started once before the export. Therefore the measured energy consumption of the plug should be zero. According to the app, the energy consumption of the plug is currently 0.53 kWh after a first run this morning.

Looking at the numbers in the app, I guess home_usage_total is the amount of energy that was delivered to the home grid by the four solar panels (directly) and the battery.

> The smart plug details in energy export is pretty much the same than the device details that are supposed to show consumption values, however those are always show 0 in the exports. Do you see [...]

Please see explanation above.

> If you are looking for support of the plug switch or schedule, you will have to use the Api library and customize a pyton test program according the example, so you can test various queries with [...]

I'll have a look at that. Thank you.

thomluther, 2024-10-21:
Hi @rwe87 , I released a new export tool in the main branch.
Are you able to download the repo code and use the export_tool with a python runtime? You can use the default export parms, which are also used by HA integration.

I cannot add new export capabilities to the HA integration for each change and enhancement in the export, while the query is not properly validated to be useful and work under all conditions...
The export tool has now a new endpoint query to get OTA update version for devices. I need more response examples for this query, especially while there is an OTA update available for any device, to ensure the query is reliable and shows same update info as in the App.
For Smart Plug information, it will be good to know what the App can configure for the plugs, except of switching them. I understood a schedule per plug can be defined? There is also a timer per plug that can be used?
Before you run the next export, it will be good if you can setup such a schedule and timer prior export using your owner account, just to see whether the export already reveals those settings. Another setting before export is to set a default home load via SB schedule (e.g. 10W or whatever your typical base load is), and use the Smart Plug mode for adding load that is measured by the plug. The export should be done when there is really load on the plug, to see whether those power values are reported also to the cloud.
If the additional plug timer and schedules are also managed via the cloud, there is hopefully a method to query them and in a second step to modify them via the Api endpoints.
The get_device_parm endpoint may reveal device schedules. Currently the export uses only the known parameter "4" (SB1 schedules) and "6" (SB2 schedules). Other numbers may reveal smart plug schedules or timers which you would have to explore manually (0-10), but output may be visible only when they really have something configured.

The Api method get_device_parm can be used with your siteId as parameter, the paramType can be modified for other numbers, e.g.:
`_out(await myapi.get_device_parm(siteId=siteId, paramType="5"))`

where _out is defined as following in your test module (see example test module in [README.md](https://github.com/thomluther/anker-solix-api?tab=readme-ov-file#usage):
```
import logging
import common

CONSOLE: logging.Logger = common.CONSOLE

def _out(jsondata):
    CONSOLE.info(json.dumps(jsondata, indent=2))
```

For the next HA integration, I can at least enable the smart plug mode setting for the user mode entity. That was already prepared. But anything else (timers, schedules, switch) for smart plugs can only be added once their reporting and management via the Api is fully understood.

thomluther, 2024-10-21:
Actually the smart plug user mode was already in the integration, but has a bug for proper key name, therefore it was not showing up. That is now fixed and it will work in the next release.

steghoja, 2024-10-21:
I've been trying out `get_device_parm()` with different values. Nothing interesting in 1-7, but then an interesting error on 8:
```
Api Request Error: 400, message='Bad Request', url=URL('https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_device_param')
Response Text: error: value "8" is not defined in options "[1 2 3 4 5 6 7]"
```
Looks like this is not the way to get data about the smart plugs...

thomluther, 2024-10-21:
Hmm. ok. Then maybe the get_device_attr endpoint, which was introduced with SB2, but I never found proper parameters to get meaningful response with my SB1 system.

thomluther, 2024-10-28:
v2.2.0 added support for:
- Smart plug usage mode
- Support for blend_plan in the SB2 schedule to define other load to be used when smart plug usage mode is active
- Total Smart plug power and energy in system, as well as other load power as contributed by the blend_plan
- Smart plug device power and energy

closed 2026-04-14 by thomluther