thomluther/anker-solix-api #161: Add support for Anker Prime Charger

laurencijs, 2025-01-20:
It would be great if we could monitor/control Anker Prime Charger. [This one](https://www.anker.com/products/a2345-anker-prime-charger-250w-6-ports-ganprime?variant=43931956478102)

In the Android app it has power monitoring on ports (and sum) an ability to switch ports individually on/off. 
In addition there is option to change charging mode and some more settings but at least for me it is not so important I'd like to control ports on/off and monitor energy on ports.

closed 2025-01-20 by thomluther

thomluther, 2025-03-10:
Hi @laurencijs 
I found some new api endpoints, not sure however what they are used for. Eventually they are for the mini power devices, but you would have to test them with the api and document them, since those queries will require owner access.

Endpoints
```text
    mini_power/v1/app/charging/get_charging_mode_list
    mini_power/v1/app/charging/update_charging_mode
    mini_power/v1/app/charging/add_charging_mode
    mini_power/v1/app/charging/delete_charging_mode
    mini_power/v1/app/setting/set_charging_mode_status
    mini_power/v1/app/egg/get_easter_egg_trigger_list
    mini_power/v1/app/egg/add_easter_egg_trigger_record
    mini_power/v1/app/egg/report_easter_egg_trigger_status
    mini_power/v1/app/setting/get_device_setting
    mini_power/v1/app/power/get_day_power_data
```

Example how you can try them with the Api library request method:
```python
_out(await myapi.apisession.request("post", "mini_power/v1/app/charging/get_charging_mode_list",json={"device_sn": deviceSn}))
```

thomluther, 2026-01-29:
Hi Timtam

> the usb port readouts just showed up, which is nice. However they don't update quite frequently, presumably only when pressing the mqtt status check buttons. I however notice that they automatically start to update quite frequently (once every 1-2 seconds) as soon as I open the app to read them out. Is that because the app initiates an MQTT status check every few seconds?

This is all excessively described in [MQTT connection and integration](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#mqtt-connection-and-integration).

> So I guess that kinda is the same question as above, where do I start?

Yeah, that is described in the Contribution part for each release note as well as in the [README](https://github.com/thomluther/ha-anker-solix/blob/main/README.md#contributions-are-welcome). So I don't want to repeat that in each feature request.

> And are you planning to support those things, should I be able to dig up the necessary api calls?

I can implement useful features if you find and document the required Api calls with examples.
But I'm afraid for your device this may all be controlled through MQTT, and you have to use the linked MQTT decoding guidelines and mqtt_monitor tool to describe all message fields as well as providing examples and field descriptions for all command messages sent by the App.