thomluther/anker-solix-api #162: Extending the solution to support Anker Solix X1 systems

ponten-weber, 2025-02-02:
Hello Thomas,

I just saw a discussion from three weeks ago where someone was writing about extending your solution to support the X1 system of Anker.
You gave some start points to begin and were asking for some export and would need a member account later and you requested to raise an issue for that.

After this there were no more response.

I have an X1 system running, too. And I would like to support you with the needed access and export.

I attached the freshly extracted export to this post.
For a member access please get in touch with me.

Kind regards
Aleksander

[ankxx_2025-02-02_1122.zip](https://github.com/user-attachments/files/18631932/ankxx_2025-02-02_1122.zip)

thomluther, 2025-02-03:
Thanks for the output.
X1 support will be a huge enhancements, since the Api and the structures are still totally unknown.
Monitoring is just the first phase and a shared X1 access would help for that.

But lots of Api testing and documentation would be required for doing changes with the Api and that is nothing I will do on other systems, since you could potentially break the system...
So any X1 owner who wants support for changes via the cloud Api has to make familar with the Api hes_svc endpoints, explore them and their parameters to understand how they can be used to modify various settings of the system and whether they work reliably when processed via Api.
Just once this is validated and documented by the community, I can add required code to the library and HA integration.

I need to postpone any work on X1 for the next couple of weeks, since I have limited time and still working on SB2 AC support...

thomluther, 2025-02-03:
After having a first look to the exported data, there are also lots of codes used again where I have no glue what they mean.
Such things needs to be monitored by the owner as well, to compare various codes in Api responses on what they actually reflect in the App in order to 'decode' them and make them human friendly.
This must also be documented, to pick this up when developing Api support for it.

One example for the codes in hes_data in bind_devices output:
```
        "hes_data": {
          "station_id": "392ffd24-6cf4-4870-94aa-14b4a0ce6f19",
          "type": "pcs",
          "online_status": 1,
          "master_slave_status": 1,
          "network_status": 2,
          "grid_status": 0,
          "rssi": ""
        },

```

Another open question is the rssi field showing typically Wifi Strength....But Anker has different use of rssi field values...
For X1 it is always 100, but I know the rssi field have being misused as negative total dbm value for Solarbank systems.
So what does 100 mean? A Percentage of 100%, an index of 100 between 0-255 that must be converted into a percentage?

ponten-weber, 2025-02-03:
I'm not so much into coding and I'm not familar with Python. That makes it a bit difficult for me. Sorry for that.

But maybe I can help a bit with the output you posted.

Type = PCS - Power Conditioning System. This should be the power unit of the stack which has three battery modules. I think those might be listed seperatly.

online_status = 1 - yes, the system is online and connected to the Anker cloud.

master_slave_status = 1 - With a later firmware you will be able to add multiple PCS to one system (up to 6 or so) each with up to six battery packs. I only have one, this system is the master. So 1 = master.

network_status = 2 - not sure about it, maybe the way the system is connected, in my configuration it is connect by wire, not wifi.

grid_status = 0 - no explanation that I could be sure enough about to write it here.

rssi = "" - I think the value is empty because I am not using wifi or the additional available 5G module.

I know, this only helps a little bit, but every piece counts I think.

I will try to get more familar with the mentioned coding help.

thomluther, 2025-02-04:
Thanks for feedback. 

> I'm not so much into coding and I'm not familar with Python. That makes it a bit difficult for me. Sorry for that.

Yeah. It might be sufficient to use curl for running Api calls, but using the Api library with the usage example code is much easier.
In principal you only need to modify the endpoints and parameters in on line to vary and explore the queries and print the response.
Typically after making a modification, you also have to requery to see the modified values or effect.
Some changes however may have a delayed result, depending on how often the system provides updated data to the cloud.
For SB2 systems, the visible results in Power values can be delayed by up to 5 minutes after you processed a change.
For X1 I saw somewhere a data interval of 5, so probably it also reports to the cloud only every 5 minutes.
This is also the lowest change rate for monitoring via the Api that you can get.

Regading the states:
- online_status: This is the cloud connection, probably only 0 (Offline) and 1 (Online) like for other devices
- master_slave_status: If 1 is a Master, then 2 could be a Slave (0 nothing if it may be present?)
- network_status: If 2 stands for wired, it would have to be compared with a Wifi connected system to get the difference
- grid_status: Could be the same as for smart meter devices. Never saw something different than 0 = OK. But I would expect that on power outage there should be a different status code. So any export under different system conditions may reveal different codes and interpretations

I noticed that the export also contains sensitive data. I will remove those queries and verify that all serials will be randomized...

thomluther, 2025-02-11:
So far there does not seem to be a query in the export that shows the actual power values.
That would be benefitial if such a query can be identified. The scene_info as used for Solarbank devices is pretty much empty for HES devices such as X1.

One limited work around would be to extract the last data points from the intraday statisics as done for the Power Panel devices.
However, this will trigger lots more queries and Api traffic since the intraday responses are quite huge (>100kb each)
Therefore I'm reluctant to implement another device monitoring based on delayed statistic values from the cloud...

GrahamDumpleton, 2025-02-27:
When you say:

> So far there does not seem to be a query in the export that shows the actual power values.

you mean true instance spot values for specific device vs the average values returned in `api_sites.json` under `average_power`.

The average values are fine for something like `Power Flow Card Plus Card` if looking at whole of house.

Just curious as starting to look at what still needs to be done for X1 support.

GrahamDumpleton, 2025-02-27:
When you say find an API query you mean a different new one beyond what are known and documented in `apitypes`, or just trying all the known API endpoints and see if any already provide the information.

How was the list of API endpoints in `apitypes` derived in the first place, deconstructing the app code/package?

GrahamDumpleton, 2025-02-27:
Okay, I will start poking around and trying in turn all the `hes` endpoints and see what I can learn about them.

thomluther, 2025-02-27:
You can see in the `export.py` how they are used with parameters for the known usage...
Most of them need the site ID and or device ID in various naming conventions of the parameters.
Anker is extremely inconsistent in parameter naming conventions, there doesn't seem to be an overall designer for the Api probably...

If you use the https://github.com/thomluther/anker-solix-api you can already create a new export for me.
I added the requests for the energy intraday stats, and the Api also has a new throtteling implemented to avoid exceeding the Api request limit. That may however dramatically elongate the export from couple of seconds to 1 or more minutes due to throttling...

GrahamDumpleton, 2025-03-04:
Been a bit busy so haven't been able to poke around too much.

Ideally, how often are you wanting to be able to poll for actual power values?

Right now `average_power` appears to provide a copy of the values from the current open time bucket, which is a rollup average from the MQTT values if they are being reported more often to the backend.

```
    # api_sites.json

    "average_power": {
      "last_check": "2025-02-28 18:15:40",
      "valid_time": "2025-02-28 18:19:58",
      "offset_seconds": 257,
      "power_unit": "kw",
      "charge_power_avg": "6.45",
      "state_of_charge": "58",
      "solar_power_avg": "1.41",
      "home_usage_avg": "7.86",
      "grid_import_avg": "0.00"
    },

    # hes_energy_hes_today...json
      {
        "time": "18:15",
        "powerInfos": [
          {
            "type": "hes",
            "value": "6.45",
            "valuePtr": "6.45",
            "isFix": false
          }
        ]
      },
```

FWIW, I have a contact inside of Anker who I have been talking to a bit since they were here for my installation (one of the first in AU) and they have been helping sort out some issues I reported which partly turned out to be the firmware in battery systems not being quite right for AU market (fix should be in battery system firmware update later in the month). If we can get together a clear description of what would be nice to have, eg., API call which provides actual power values at specific time, vs average, along with why we need it, how it helps us and why it may be better for Anker systems (less API queries/load), then I may well be able to get the request passed on to internal folks in Anker who control the API.

thomluther, 2025-03-04:
Well, what would be nice to have is:
- A local Api, like MQTT, to allow external monitoring and automation of the Anker Solix systems via custom Smart Home solutions

I'm pretty sure Anker will not open that widely, since they already denied that there will be Interfaces or support for Solix devices and Smart Home solutions.
Therefore I think it is wasting time to request Cloud Api features for something that is not officially supported or utilized by the mobile App. In contrast, it may get worse when they start to look down non App communication to the cloud Api by enforcing encryption which is unknown today.

thomluther, 2025-03-04:
What is existing for the balcony power systems is a single scen_info endpoint which returns all actual power values to the cloud in a single query.
Those values on the cloud are refreshed by the power system in either 1 or 5 minute intervals. So it also depends on the system how often they refresh their power values in the cloud.

thomluther, 2025-03-04:
> Right now `average_power` appears to provide a copy of the values from the current open time bucket, which is a rollup average from the MQTT values if they are being reported more often to the backend.

Those average power values is what I extract out of the energy stats data. How those energy stats are generated in the cloud is unknown. I doubt that these are averaged MQTT values, since MQTT data transmission is not used permanently, but only when triggered by the device owner via the mobile App. And only the watched devices will transmit their real time data.
The energy data in the power cloud is probably accumulated from power values that are just send in regular intervals, e.g. 60 sec or 300 seconds. I don't think they are quite accurate.

thomluther, 2025-03-04:
@GrahamDumpleton it would be helpful if you can send me an actual export zip of your system as owner account.
Your screenshot shows that the generic extraction works, but I still miss the X1 discharge energy. Not sure where that is given, maybe the pps energy stats?
With a full set of data, and knowledge where to find the discharge energy, I can work an an initial monitoring release in next couple of weeks/months.
Right now I must focus of Api improvements, since Anker further locked down the endpoint query limit per minute to 10-12 per endpoint per minute.
Therefore I have to implement and endpoint throttling mechanism, which will severilly elongate data refreshes involving multiple queries to the energy endpoint. And only the average power refresh drives already 4-5 queries without that daily energy stats even have been gatherered...

thomluther, 2025-03-11:
Hi @GrahamDumpleton, @ponten-weber 
Can one of you share the system with my account lollygames2@gmail.com? Then I could start exploring what monitoring capabilities exist with the X1 system via Api and the App to see what values can be bridged and explained through the App.

ponten-weber, 2025-03-11:
> Hi [@GrahamDumpleton](https://github.com/GrahamDumpleton), [@ponten-weber](https://github.com/ponten-weber) Can one of you share the system with my account [lollygames2@gmail.com](mailto:lollygames2@gmail.com)? Then I could start exploring what monitoring capabilities exist with the X1 system via Api and the App to see what values can be bridged and explained through the App.

I sent you an invite.

Rgds
Aleksander

thomluther, 2025-03-11:
Thanks, I joined. Some first observations compared to other systems:
- The App does not show a total battery charge value (I have not found that either in the energy stats). There are only breakdowns available (charge from solar and charge from grid), but this makes it more difficult to see the storage efficiency in the app, since you have no total charge value in the battery stats that you can easily compare with the discharge total...That explains why I did not find a total charge value either in the HES energy stat responses
- I don't see a total monetary savings number like for other systems. While there is a drill in arrow in the totals, I cannot use it as member. I believe the total savings however are available via the Api at least
- There are different operation modes for PV and Battery. The other systems only have a combined mode. I will consider the PV mode as new mode for the X1 systems, since the existing charging mode may be mapped to the battery accordingly
- Not sure how the power values are merged into the homepage. They update in about 5 sek interval, not sure where the shared account gets them so quickly. Typically the shared account does not trigger real time updates through the MQTT cloud due to permission problems. It could be that X1 will provide frequent updates into the power cloud since it may have less energy efficiency requirements. but still the Api endpoint to get those HES power data have to be found if they are really in the power cloud...

thomluther, 2025-04-06:
Hi @GrahamDumpleton, I have some basic values now implemented.
It would be helpful if you can send me an actual export zip of your system as owner account.
You can do that during the day, when there is also some PV and charging.
I haven't found anything yet for the battery modules, no SOC etc.
Only the overall SOC extracted from the intraday chart data is available, but that might be sufficient for a starting point

thomluther, 2025-04-06:
@ponten-weber, You can also provide an actual export zip of your system as owner account. The more various conditions are available, the easier might be the interpretation of different states.

ponten-weber, 2025-04-07:
@thomluther I will try to switch the account tomorrow. Normally I have a member account used in this context, so I need to switch to the owner account for the export. I will than generate the new export and upload it. I will do this during the day with some pv and battery values if possible.

hblaschka, 2025-04-24:
@thomluther thanks for your work, have been lurking around here while waiting for my X1 HES (X1-H12K-T with 2 battery modules) to arrive , which finally was installed after the first release of HES support. Great work, works like a dharmn (for the limited data points identified).

Had discussions with key employees in Germany that I am in contact with due to topics I am not allowed to talk about ATM,  but I was told we (all Smart Home enthusiasts) are clearly heard. 

The topic of a local API for Smart Home Systems like Home Assistant seems to be one of the most brought up topics in the suggestion area of the App for the DACH markets (which they give a timely feedback to, each time you enter a new suggestions, and I entered a lot - strongly encourage everyone to start communicating, not only regarding the local API, but generally for any ideas and suggestions - look for the question menu point in the app settings, make sure to seledt the right product and use the feedback/suggestion seledtion,X1 is actively read/monitored by PMs/Devs, they put a lot of effort into those professional products).

Brought up the argument during phone discussions, that a local API would save them a lot of (cloud) cost (and headaches), as the load on the cloud APi would decrease once a local API would be abailable. That argument seems to have stuck and I was promised this topic will be brought up in parallel via internal channels. As I will have specials channels due to those things I cannot talk about ATM, I will make sure to use those wisely, also 🤫

Once I find some time I will make sure to support via commented data dumps.

Tom, thanks again - really appreciate your work.

thomluther, 2025-04-24:
@hblaschka Thanks for your feedback on this topic.
Good to see that Anker will start to move towards the (local) Api, although that will probably not be in the short term but hopefully in a 1-2 year range.
An official Api must be consistent and well documented, pretty much the contrary of what the current cloud Api is.

I can imagine that cloud Api traffic is increasing forever, not only by the number of new systems and App usage, but also by the number of Smart Home Systems and 24x7 data polling, which is much more than just sporadic App usage.
Especially for the HES and Power Panel systems, there are no simple one shot queries to get power values. The work around is nasty and unfortunately pulls lot of data in the response of intraday stats which cannot be limited...This causes already  ~80 MB per day per system in the HA integration.

> Once I find some time I will make sure to support via commented data dumps.

That would be great. The system export does not include much queries since their correct syntax cannot be validated without owner access to a system. So lots of testing with various endpoints will be required to get more data points and eventually find requests to modify specific settings of the system. I guess most of the App settings can be done also via the cloud Api, but this will have to be found, documented and tested under various conditions to ensure it can be safely added to the library.
I'm happy to assist with the library and integration coding once the queries and functionality are validated.

hblaschka, 2025-04-28:
@thomluther As I assumed I now have direct communication lines towards the Product Managers / Owners of both the X1 and the Solarbank product line + the General Product Manager for the Solix products (might be able to meet him in person). I already raised user stories (As a user of X I want to be able to Y) regarding API & local MTTTQ broker incl. documentation. If you have anything else or sth. to add, feel free to contact me and I will raise using these channels. As far as I understood they already seem to be aware of your work.

Everybody feel free to contact them regarding directly supporting their products on HA (Feature requests (mark as such) via support@anker.com - flood them with requests).

thomluther, 2025-04-28:
> If you have anything else or sth. to add, feel free to contact me and I will raise using these channels. As far as I understood they already seem to be aware of your work.

To bridge the gap until anything official will become available, it would be nice to 'get access' to more documentation about the cloud Api and whether that can be utilized further for the existing systems until any official interface is available.
Of course that may depend on the Product Manager's view of my current HA integration project and how much Anker is willing to tolerate or even assist with it for the user community... 
I'm pretty sure they will make many Smart Home enthusiasts very happy with more HA integration support as the [>500 Stars of the HA Integration repo](https://github.com/thomluther/ha-anker-solix) underlines the need for such monitoring and integration capabilities, to trigger any alerts or automation in user's households.

My actual points of interest for the cloud Api would be very detailed, nothing or a public use case:
- Are power consumption data available for residual systems (X1, Power Panels)?
- Are device temperatures available?
- Which device settings can be queried and modified via cloud Api?
- Can Smart Plugs be managed via cloud Api?
- How to establish cloud Api payload encryption?
- Is compression supported by the cloud servers to utilize it for traffic reduction?

thomluther, 2026-03-18:
You can see all MQTT commands to the device in the mqtt_monitor. Filter for the cmd/* topic or in the dump you can also filter for '03 00 0f' pattern. In the header you may recognize from the client id who sent it, there is the cloud, and then andoird or ios clients depending on the App. The monitor uses an android client, but that has an extended id with a random number.
Without having the app active for a while, you can launch the mqtt_monitor. It would only receive standard messages from the device. If you then open the App, you will get the commands sent by the App launch to the system.
RT trigger is typically 0057 command, status request 0040 command, but it may vary.

This way you can also dump any other MQTT control for the device from the App. I would need example message for each App control, along with information about the supported ranges/options in the app, and in which status message the command value fields are toggled after the command. Those represent the state of the control.

> I was able to intercept the traffic - there are quite a few HTTP POSTs to various endpoints to request data depending on what area in the app you drill down to. It looks like the payload is encrypted (Content-Type: application/json, body is a base64 string but decoding it doesn't yield readable text). I see you had another issue where there was discussion around request encryption on other devices; did that ever get anywhere?

If you get the Api endpoints for certain App controls, that would be more than nothing and point to the direction to be investigated.
E.g. there are probably Api endpoints to query and modify anything with the usage mode, custom schedule and backup or time of use plan. Once you know the required endpoints, you can explore them further with the Api library, to get the mandatory parameters etc. The App Api payload is ECDH encrypted with a negotiated shared secret for each session. This cannot be decoded, and I even don't know how the key exchange works to use payload encryption as well. This is discussed in detail in #70

martinmarinov, 2026-10-28:
For the open X1 trigger-format question in this thread: I captured it on an A5102.

Realtime is a **cloud-originated JSON `cmd: 9`** with `cmdId: 4`, `config: {count: 60, interval: 5}` (a five-minute stream), not `0057`. The data arrives on `dt/anker_power/A5102/<sn>/troubleshot_info/ems_second` about every five seconds.

I can share a sanitized frame privately. Details and test notes are in #361.