thomluther/anker-solix-api #178: No effect when set_sb2_home_load with preset

maxspahn, 2025-04-02:
Hi @thomluther ,

amazing work! I cannot thank you enough for this project!
I am using it for communication with OpenHab through mqtt and it works great for data transmission. However, I am not able to set the load of my Solarbank 2. 

The API-method set_sb2_home_load does change the home load, visible in both the site_details and even in the App using a shared account. However, it has no effect on the actual home_load. Only, if I change the value again in the App, it does have an effect.

Any suggestions?

Best,
Max.

maxspahn, 2025-04-02:
Ok, update on this. I reset a schedule with the app and tried again and it works.

It seems to be failing only when you attempted with a poor request using the set_device_parm method.

maxspahn, 2025-04-02:
Ok, it seems to be quite inconsistent, but once it stops working, it will never work again. Then, the app says, it was somehow interrupted in the proces of setting it.

thomluther, 2025-04-02:
Hi @maxspahn , which model do you have?
The AC model has this issue since the beginning and it was not resolved. I have no idea why, but shouldn't be related to the Api call, unless Anker does something specific for the AC which I don't know.

In general, you may have a delay of 5 or more minutes until you see the setting change reflected in the cloud values, because SB2 posts data only every 5 minutes at most (unless the App is open as admin)

thomluther, 2025-04-02:
> Ok, it seems to be quite inconsistent, but once it stops working, it will never work again. Then, the app says, it was somehow interrupted in the process of setting it.

That is what I have seen on an AC system, especially when a usage time plan was also defined. But I have no admin access to SB2, and App traffic sniffing does not work due to encryption. So eventually the Api call is incomplete or another one has to be done, or this is not fully supported through the Api alone and the App sends also something with the MQTT server???

thomluther, 2025-04-02:
If you have a AC, I bet on buggy firmware, since the AC also stops posting some values to the cloud after a while (certain power fields, battery counts, charging status etc remain 0)...

thomluther, 2025-04-02:
BTW, the SB1 also has an issue with the plan. When sending a plan with just 1 interval, the preset was not applied although in the plan. Looking in the App the output was showing 0W, which is not possible for SB1 since below limit of 100 W.
Maybe that is common issue with the preset if the sent plan has only a single interval. Try 2-3 intervals via the app, then just modify the preset of current slot via the helper method.

maxspahn, 2025-04-05:
Thank you @thomluther for the ultra fast response.

I have monitored and played around with the api calls a bit and I will just write down my conclusions. My model by the way is the Anker Solix Powerbank 2 1600 Pro.

> In general, you may have a delay of 5 or more minutes until you see the setting change reflected in the cloud values, because SB2 posts data only every 5 minutes at most (unless the App is open as admin)

That is the thing that I have started to accept now. It just takes around 5 minutes and there is nothing I can really do about it.

> So eventually the Api call is incomplete or another one has to be done, or this is not fully supported through the Api alone and the App sends also something with the MQTT server???

I don't think that there are incomplete API calls, because the delay is very consistent and my MQTT service that does the API calls and subscribes to the MQTT topic has been running for 3 days straight now.

@thomluther, let me know if you are interested in the MQTT interface and i create a PR for it. 

However, I really start to hate this latency (as I said it is not an issue of this repo, but the Anker API) so I am looking into emulating a Shelly Pro3EM that should allow me to use the smart meter interface that communicates locally, see https://github.com/sdeigm/uni-meter for a starting point.

Out of pure curiosity: How did you figure out all the API calls? What exactly do you mean by App traffic sniffing?

Cheers,
Max.

thomluther, 2025-04-05:
> However, I really start to hate this latency (as I said it is not an issue of this repo, but the Anker API) so I am looking into emulating a Shelly Pro3EM that should allow me to use the smart meter interface that communicates locally, see https://github.com/sdeigm/uni-meter for a starting point.

I'm not sure if the uni-meter works for Solarbank, that still has to be confirmed by someone. It doesn't work either yet for Zendure.
It all depends which methods the storage systems use to bind / register the Shelly. If they do the binding with the web portal, it won't work since the uni-meter cannot register the IP as virtual shelly in the cloud.

Anyway, there is no big delay for  the SB2 to react on your Api change, that is done in 3 seconds I was told.
But you don't see it. You will have to use a fast smart plug for the power feed into the house. This must be fast as well as your smart meter sensor. If those are in the range of 5 seconds update intervals, you should be able to do the regulation directly. But the actual grid value and the AC feed in from the balcony system must be fast, since you need to use them to calculate your actual house demand, which in fact is the set point for your Api call.

> [@thomluther](https://github.com/thomluther), let me know if you are interested in the MQTT interface and i create a PR for it.

If you can maintain it when there are issues, you can create a PR for adding it. I won't use it, so probably not doing anything with it or maintaining it. The main purpose of the library was to use it for the HA integration which I maintain too

thomluther, 2025-04-05:
> Out of pure curiosity: How did you figure out all the API calls? What exactly do you mean by App traffic sniffing?

I extracted them from the Apk package binary. Since they are strings, you can find them in the compiled packge. But you need to know what you are looking for...
And having the endpoints doesn't tell you how to use them and for which systems they are to be used.
I would be glad if someone finds the usage of the device_attrs endpoints. They were introduced together with SB2. I assume they could be used for device parameter or settings, or smart plugs....
They require a list of 'attributes', but I haven't found attributes that will provide any response values....

maxspahn, 2025-04-06:
> Anyway, there is no big delay for the SB2 to react on your Api change, that is done in 3 seconds I was told.
But you don't see it. You will have to use a fast smart plug for the power feed into the house. This must be fast as well as your smart meter sensor. If those are in the range of 5 seconds update intervals, you should be able to do the regulation directly. But the actual grid value and the AC feed in from the balcony system must be fast, since you need to use them to calculate your actual house demand, which in fact is the set point for your Api call.

Indeed the SB2 changes the desired home load very fast, but it only takes effect after around 5 minutes. I can see that my computed demand from smart sockets is almost instantly, so the bottle neck is the change in the actual home load. I do not understand exactly if you mean that this should be indeed faster and that there is something wrong in my setup.

> I extracted them from the Apk package binary. Since they are strings, you can find them in the compiled packge. But you need to know what you are looking for...
And having the endpoints doesn't tell you how to use them and for which systems they are to be used.
I would be glad if someone finds the usage of the device_attrs endpoints. They were introduced together with SB2. I assume they could be used for device parameter or settings, or smart plugs....
They require a list of 'attributes', but I haven't found attributes that will provide any response values....

Thanks for the details. I am fascinated how you managed all this.

Best,
Max.

closed 2025-04-10 by thomluther