thomluther/anker-solix-api #193: Add Dynamic Provider options and Price details

thomluther, 2025-06-02:
SB3 supports dynamic tariffs that can be utilized in various modes.
Basically the System price type can be toggled to dynamic, then using the prices provided by the configured dynamic price provider.
There is currently only one provider supported, but this is configurable if others will be supported in future.
Then there are also dynamic prices provided for today and tomorrow in the App which should be made available in the integration.

thomluther, 2025-06-02:
The end price displayed in the App should calculate as following:
(spot_price/1000+fee)*(1+VAT%)

DuzAwe, 2025-07-03:
Would it be possible to set these manually? The app allows you to build a time of use tariff and has behavior tied to the type of price bracket. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

thomluther, 2025-07-09:
Dynamic price options for SB3 have been added to release 3.0.0