thomluther/anker-solix-api #196: Support Time of Use settings for X1

thomluther, 2025-07-03:
> Would it be possible to set these manually? The app allows you to build a time of use tariff and has behavior tied to the type of price bracket. 
> 
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image] 

 _Originally posted by @DuzAwe in [#193](https://github.com/thomluther/anker-solix-api/issues/193#issuecomment-3031895699)_

thomluther, 2025-07-07:
@DuzAwe This are all standard queries. Without that I was ever able to sniff the endpoints, here is what I assume

1. https://ankerpower-api-eu.anker.com/v1/openapi/oauth/key/exchange
This will be the ECDH key exchange to decrypt the payload. This mechanism is not understood yet. Since all payload is encrypted, you cannot sniff the parameters and response data of the Api.

2. https://ankerpower-api-eu.anker.com/v1/openapi/devicemanage/get_user_mqtt_info
This will allow the App to get the certs for the cloud MQTT Server, which is used for the device management. Getting live data per second and manage device settings from the device page. Much of these settings may not be manageable at all via the power cloud server. The MQTT server may also be used when needed, so device may have to be triggered by the App to publish MQTT data to the cloud server. There is no 24x7 real time data transmission for sure. So nothing to be usable for HA. Also the data is ECDH encrypted from what I have read here in the repo and MQTT topics are unknown either. There is no known interface for this server.

3. https://ankerpower-api-eu.anker.com/power_service/v1/product_categories
Use the get a common Anker product list. No system specific info (see export examples)

4. https://ankerpower-api-eu.anker.com/power_service/v1/app/get_relate_and_bind_devices
Get a list of all devices that are configured as owner. This does not work for shared accounts, therefore shared accounts cannot do any device related stuff. Shared account only see the home page, with data merged from scen_info response for balcony power systems, or other endpoints, and the non balcony system power data are probably merged from real time MQTT data. So they are not available at all via the power cloud Api for Non balcony power systems.

Some schedules and price information or even changes might be possible, e.g. user mode toggling. But you will need to test this with the charging_hes_svc queries that are not used in the export module so far, because the known data from exports does not reveal much. Play around with the parameters for api calls, sometimes the error messages tell which parameter is missing or if it is in the wrong format.

DuzAwe, 2025-07-09:
@thomluther Spent some more time with this. Toggling modes in the app dont seem to be sending anything via HTTP/HTTPS so I have been unable to capture anything. Around this, Any advice on trying to capture more. Should I hook up wireshark and grab everything?

thomluther, 2025-07-09:
> [@thomluther](https://github.com/thomluther) Spent some more time with this. Toggling modes in the app dont seem to be sending anything via HTTP/HTTPS so I have been unable to capture anything. Around this, Any advice on trying to capture more. Should I hook up wireshark and grab everything?

That is weird. I thought it would at least send something via the MQTT cloud....
You may try to connect the device with the App to another network to avoid any local https or BT connection is possible. Then any configuration must go through https and cloud servers, how should it work otherwise from remote?

The question is whether you can sniff and decrypt https with wireshark for anything that is used by the device?
Knowing which cloud endpoints are being used would already help (parameters can maybe be tested by trial and error via the Api)

thomluther, 2025-07-09:
You can also try to play around with the Api using a test client as documented in the readme.
The *hes_svc endpoints are all in the apitypes.py module. For the beginning, you could start with hes queries that are also used by the export module to verify that your test client works in principal.
Then modify endpoint and/or parameters and data to explore the response of various endpoints. This is how I discovered most of the stuff, but your client should use the owner account, otherwise some endpoints always fail due to permission errors.

thomluther, 2025-07-17:
Just found new HES endpoints for dynamic prices in latest apk. They work only as system owner:

-     "charging_hes_dynamic_price_svc/get_area_by_code", # needs owner
-     "charging_hes_dynamic_price_svc/get_price_company", # needs owner
-     "charging_hes_dynamic_price_svc/get_price", # needs owner
-     "charging_hes_dynamic_price_svc/save_time_of_use", # needs owner
-     "charging_hes_dynamic_price_svc/save_dynamic_price", # needs owner

thomluther, 2025-08-19:
Hi @DuzAwe Did you make any progress with testing Api endpoints?
Is there anything useful to add to the export routine with additional datapoints or eventually even setting modifications?
I think at the leas the custom rate plan (schedule) should be somewhere in the cloud. And then most of those fields can probably be adjusted as well via the Api, like it can be done for Solarbanks.

thomluther, 2025-10-29:
Hi, you can use the mqtt_monitor in the Api library to decode mqtt messages from X1 and eventually the battery modules.
Please follow the general [mqtt decoding guidelines]( https://github.com/thomluther/anker-solix-api/discussions/222) to decode and document your device for MQTT data.
Also direct commands can be monitored, but usage modes and plans probably require Api requests.