thomluther/anker-solix-api #202: `pipenv lock` python version range specifier error

KUHLwasStolen, 2025-07-27:
When installing the project's dependencies as described in the README the following error occurs when running `pipenv lock`
```
$ pipenv lock
Error: Python version range specifier '>=3.12' is not supported. Please use an absolute version number or specify the path to the Python executable on Pipfile.
```

To fix this I see three options
1. Use absolute version number and only support one python version, i.e. `python_version = "3.13"`
2. A looser requirement like `python_version = "3"` would allow all python3 versions, but this might come with possible incompatibilities
3. This is often suggested on the Internet: migrate to `poetry` (https://github.com/python-poetry/poetry). This would support version ranges like `>=3.12`. This would of course mean the most effort, but I think would be the most future-proof solution while still supporting multiple python versions.

In case you like option 3 the most I would be willing look deeper into it and handle the migration.

pipenv version: `2025.0.4`

thomluther, 2025-07-27:
Actually it was not me who implemented pipenv and I'm not using it, since I alwayshad issueswith it and the cache. So yes, if you like you can look into replacing it with poetry, which I don't know either.

KUHLwasStolen, 2025-07-28:
Alright, I read some documentation yesterday evening and the switch shouldn't be that bad.
Can expect my pull request later today or tomorrow.

PR #203 (2025-07-28): migrate from pipenv to poetry for better python versioning
Replaced `pipenv` with `poetry` as discussed in #202 
Also adjusted README.md accordingly

**Note:** The file `poetry.lock` is more or less optional. The poetry documentation recommends committing it to the repository such that dependency versions stay consistent for everyone.
We can also delete it and add it to `.gitignore` to tell poetry to always use the latest version of the dependencies.

closes #202
closed 2025-07-28 by thomluther