thomluther/anker-solix-api #211: Add helper method to enable/disable manual backup power

cweitkamp, 2025-08-14:
Is it possible to add helper methods to enable/disable the Manual Backup Power mode (see screenshot) for Solarbank 2 devices via API?

[image]

Possible use case: If I have dynamic prices I would like to manually charge my Solarbank while the prices are low and use the charged energy at the time the prices are high.

thomluther, 2025-08-14:
Hi, its there already, see [schedule.py](https://github.com/thomluther/anker-solix-api/blob/4828bec0464e8fbb222dd94f20fe71610b2194b0/api/schedule.py#L1880C1-L1912C1)

```python
async def set_sb2_ac_charge(
    self,
    siteId: str,
    deviceSn: str,
    backup_start: datetime | None = None,
    backup_end: datetime | None = None,
    backup_duration: timedelta | None = None,
    backup_switch: bool | None = None,
    test_schedule: dict
    | None = None,  # used for testing or to apply changes only to api cache
    toFile: bool = False,  # used for testing with files
) -> bool | dict:
    """Set or change the AC charge parameters for a given site id and solarbank 2 AC device.

    The backup definition is part of the SB2 device schedule object. If no end_time or duration is specified, a default duration of 1 hour will be used
    manual_backup: Used for the manual backup period definition for max AC charge.
    Example data for provided site_id with param_type 6 for SB2:
    "schedule": {
        "mode_type": 3,
        "custom_rate_plan": null,
        "blend_plan": null,
        "use_time": null,
        "manual_backup": {
            "ranges": [
            {
                "start_time": 1734693780,
                "end_time": 1734700980
            }
            ],
            "switch": true
        "default_home_load": 200,"max_load": 800,"min_load": 0,"step": 10}
    """
```

End time is optional, but you need to set the switch true to enable the configured interval. Defaults will be applied if you don't specify start, end or duration.

cweitkamp, 2025-08-15:
Alright, thanks. I will test it.