thomluther/anker-solix-api #216: Support option to allow/deny export to the grid for Solarbank 2 E1600 Pro

jmozmoz, 2025-08-24:
Since a few weeks the Solarbank 2 E1600 Pro has an option to allow/deny the export of energy to the grid. If a smart meter is installed, the Solarbank is only supplying so much ernergy to balance to load of the house so no energy is exported to the grid.

In the German app is the option is called "Die Geräte speisen das Stromnetz ein". It can be set to "Aus" or "An". 

I looked at the output of the `export_system.py` but I couldn't find anything related to this. 

Is there a way to read and write this option with this library?

thomluther, 2025-08-24:
Hi, not in this library and not possible via cloud Api to my knowledge.
See https://github.com/thomluther/ha-anker-solix/discussions/336

thomluther, 2025-09-20:
In version 3.3.0, this setting is supported now for Solarbank 3 since that can be managed with the station settings.
Not sure whether this will become also changeable for SB2 systems in future.

jmozmoz, 2025-10-07:
And more values:

Power output set to user defined plan with output power 0:
```
ac   05  03    00:00:00:00
â””->   5 var               0;0               0           0.000         0;0;0;0
ad   02  01    13
â””->   2 ui                 19              19 --> battery_soc_total
```
and
```
â””->   2 ui                 10              10 --> input_cutoff_data
b7   05  03    00:00:00:00
â””->   5 var               0;0               0           0.000         0;0;0;0
```

Power output set to own consumption (Eigenverbrauch) with power output of about 460 W from battery to home:
```
ac   05  03    0c:12:00:00
â””->   5 var            4620;0            4620           0.000       12;18;0;0
ad   02  01    12
â””->   2 ui                 18              18 --> battery_soc_total
```
and
```
b7   05  03    78:b4:00:00
â””->   5 var          -19336;0           46200           0.000     120;180;0;0
```

Now the battery is empty ;-)

How should I provide more settings? Should I post the full dumps?

thomluther, 2025-10-08:
Hi @jmozmoz 
The grid export setting is already described as you can see in the decode output description
```
{'00': [{'name': 'grid_export_disabled', 'mask': 1}]}
```
For your other comments, I'm not sure which bytes you are referring to.
It will be better if you add comments to the output line (field decoding line with the value it should reflect. Otherwise it is not clear whether a factor must be defined.
you can also describe the mapping in your code and validate the value extraction is done correctly on the value display.

ohadlevy, 2025-11-09:
> Starting with commit 241cdd7
> I cannot connect to the mqtt server anymore. With commit e3da2de
> it worked:
>
> python mqtt_monitor.py
> Anker Solix Device MQTT Monitor:
> Trying Api authentication for user XXX...
> Anker Cloud authentication: CACHED
> Getting sites and device list...
>
> Select which device to be monitored:
> (1) XXX, A17C1, Solarbank 2 E1600 Pro, Alias: Solarbank 2 E1600 Pro, System: XXX
> (2) XXX, A17X7, Smart Meter, Alias: Smart Meter, System: XXX
> Enter device number (1-2) or nothing to quit: 1
> Do you want to dump MQTT message decoding also to file? (Y/N): y
> Filename prefix for export (A17C1_mqtt_dump):
>
> MQTT message dumping to file: A17C1_mqtt_dump_2025-11-09__13_47_14.txt
>
> Starting MQTT server connection for device XXX (model A17C1)...
> Connected successfully to MQTT server aiot-mqtt-eu.anker.com:8883
> ----------------------------------------------------------------------------------------------------
> MQTT Monitor key menu:
> ----------------------------------------------------------------------------------------------------
> [M]enu to show this key list
> [U]nsubscribe all topics. This will stop receiving MQTT messages
> [S]ubscribe root topic. This will subscribe root only
> [T]oggle subscribed topic. If only one topic identified from root topic, toggling is not possible
> [R]eal time data trigger toggle OFF (Default) or ON
> [V]iew value extraction refresh screen or MQTT message decoding
> [Q]uit, [ESC] or [CTRL-C] to stop MQTT monitor
> Hit [Enter] to continue...
>
> Starting MQTT message listener, real time data trigger is: OFF
> Api joa*** MQTT session failed connection to Anker Solix MQTT server aiot-mqtt-eu.anker.com: 8883
> Listening....Enabling real time data trigger, message frequency should increase shortly...
>
> Also the latest main commit a8dbeed
> does not work.

does it work if you use the monitor.py ?

thomluther, 2025-11-11:
I merged the above info and added a common delay into the async connection routine, that now also includes the loop_start.
This simplifies external usage and I updated the methods accordingly.
The mqtt_monitor should no longer see the above error.

thomluther, 2025-12-05:
Another major topic that is missing completely are Anker Smart plugs.
If anyone is using them and can start monitoring and describing MQTT data and commands for the plugs, that would be great.
There is already issue #124 where smart plugs are tracked, or a new discussion can be opened for MQTT with the plugs.
I'm pretty sure with MQTT you could get full control about the plugs like you can with the App.

jmozmoz, 2025-12-06:
I think, I found a new value:
```diff
diff --git a/api/mqttmap.py b/api/mqttmap.py
index bc897cd..9926d93 100644
--- a/api/mqttmap.py
+++ b/api/mqttmap.py
@@ -623,6 +623,7 @@ A17C1_0405 = {
     "cd": {"name": "pv_4_power", "factor": 0.1},
     "d3": {"name": "output_power", "factor": 0.1},
     "e0": {"name": "grid_status"},  # Grid OK (1), No grid (6), Grid connecting (3)
+    "e8": {"name": "battery_heating"}, # Not heating (1), heating (3)
     "fb": {
         "bytes": {
             "00": [{"name": "grid_export_disabled", "mask": 0x01}],
```

In Home Assistant the heating power is always 0 W. Also, I was not able to find something in the MQTT message.

thomluther, 2025-12-06:
> In Home Assistant the heating power is always 0 W. Also, I was not able to find something in the MQTT message.

I think the heating power is only supported for SB3, the SB2 only has a flag.

So the question is, which flag you found? Is it for the main battery only, or is it a general flag for any of the battery or expansion heating?

thomluther, 2025-12-06:
> Update: I think, `"c9"` is some kind of energy counter. AC_input does not make really sense for Solarbank 2 Pro, because it cannot be charge from the grid.

Don't think its energy, I got an example that shows 112, to low for energy which is often in wh.
Maybe heating power?

thomluther, 2025-12-06:
> > Hello, did someone found more fields? I think there are still energy values to be described. Furthermore someone should test all device commands and document a decoded message example here for each command. The command must also describe the used,allowed values from the App and a field name in the status messages that contains the state of the command.
> 
> How can I help? Should I start `mqtt_monitor.py`? Then modify one setting in the app, wait for the next status message and the stop `mqtt_monitor.py`?
> 
> Just to make sure: There are no private information in the dump. I think, I saw some serial numbers. Is there an option, to remove all those information?

The problem is in the monitor you cannot remove or randomize the sensitive information, because they are mostly unkonwn, except the SN of the device itself, because that is used for the subscriptions.
Furthermore you want to see the real stuff, to compare what you see.

In the export_system tool there is randomization, but only for the device_SN. The rest in the binary data is again unknown, so you cannot randomize it.

What I need for each device command type, is an example decoding of the full message with description which setting triggered the command and which value you used, to understand the 'parameter' in the command bytes.
Then you need to describe which other values are supported by the App for the same command, like options, values etc. If there is only an interger code for a setting, you need to provide a mapping what means which setting.

The other relevant part required for each command is which bytes in which status messages reflect the setting change.
This is pretty easy to find, if you request a status message after you changed a setting (or leave the realtime trigger active, but that may log many commands, and for command msg filtering, it would be easier to toggle only to the *req topic.
Anyway, copy the used value byte pattern from the command field and search for it in the following status messages. 2-4 byte values should be easy to identify. If it is a toggle, you need the same status message from before the command. Then look for the bytes that show the new value in the new message, and compare if the same byte in old message had the previous setting.

Update: If you include MQTT messages in the system export, there are only published messages included, since the export tool does not subscribe to the req topic to receive also commands (not necessary and not helpful if you don't know what action or setting triggered the command...). 
It will randomize the own device SN, but not other device SN or exp. pack SNs if in the messages since they are unknown through the Api.

jmozmoz, 2025-12-06:
> Hm, could be. *But if it is slowly counting, it could be also the full cycles with factor 0.1? Here is a larger gap of about 30 days, and increase was 26. So the question is whether you have a full cycle almost each day and whether you could have 376 cycles altogether?
> 
> ```
> 2025-10-11 19:40:32
> c9   05  03    af:0d:00:00
> └->   5 var            3503;0            3503           0.000      175;13;0;0
> 2025-10-11 19:40:26 -> 2025-11-09 14:27:42
> c9   05  03    b3:0e:00:00
> └->   5 var            3763;0            3763           0.000      179;14;0;0
> ```

Is there a way to execute `mqtt_monitor.py` in a cron job without entering something interactively? Then I could log this regularily.

thomluther, 2025-12-06:
> Is there a way to execute `mqtt_monitor.py` in a cron job without entering something interactively? Then I could log this regularily.

Currently there are no command line options implemented for the mqtt_monitor, this is more meant to be an interactive tool.

There is no step by step guide either, because finding commands and their states varies depending on what you look for.

Once you launched the monitor, issue an 'i' for a status request, this will force the first command to get *req topic into the list of received topics.
You can then toggle through the topics to see only commands if you want to focus on commands.
If you look for states, you should not filter on topics, but dump everything, however focus on a single setting per dump.

Later you can review the dumps, look for /req topic or '03 00 0f' pattern for sent messages, these are your commands. You may find the realtime trigger and status request commands eventually, than the one you used in the app.

And with the value byte pattern that was sent in the command message, you can filter the received messages before and after the command to see which field toggled to the sent value.

That's it. No automation, since automation or AI never knows what you did in the App that triggered the command messages or its values.

jmozmoz, 2025-12-07:
> I uploaded new main branch, this has now command line options for the mqtt_monitor. Use it with --help to see them, the will be documented in the next release. For automatic start, you have to use credentials in env variables or an .env file

Thank you. But now, it seems that realtime trigger is always turned on. It does not matter, if the app is open. With the app closed if I start `mqtt_monitor.py` messages are immediately showing up. I am not sure that this is the correct behaviour. 

Correction: Somehow it stopped. So is the realtime trigger only turned on for a certain amount of time?

jmozmoz, 2025-12-07:
I added an option to just print a status message:

```patch
diff --git a/mqtt_monitor.py b/mqtt_monitor.py
index 5a7ea60..aac7ad7 100644
--- a/mqtt_monitor.py
+++ b/mqtt_monitor.py
@@ -52,6 +52,12 @@ def parse_arguments() -> argparse.Namespace:
         action="store_true",
         help="Enable real-time MQTT trigger on startup",
     )
+    parser.add_argument(
+        "--status-request",
+        "--sr",
+        action="store_true",
+        help="Enable immediate status request for device",
+    )
     parser.add_argument(
         "--value-display",
         "--vd",
@@ -93,6 +99,7 @@ class AnkerSolixMqttMonitor:
         self.loop: asyncio.AbstractEventLoop
         self.filedump: bool = args.filedump
         self.realtime_trigger: bool = args.realtime
+        self.status_request_trigger: bool = args.status_request
         self.value_display: bool = args.value_display
         self.fileprefix: str | None = args.dump_prefix
 
@@ -300,6 +307,15 @@ class AnkerSolixMqttMonitor:
                     CONSOLE.info(
                         f"Starting MQTT message listener, real time data trigger is: {Color.GREEN + 'ON' if realtime else Color.RED + 'OFF'}{Color.OFF}"
                     )
+                    # individual status request
+                    if self.status_request_trigger and self.device_sn:
+                        if mqtt_session.status_request(
+                            deviceDict=self.device_selected,
+                            wait_for_publish=2,
+                        ).is_published():
+                            CONSOLE.info(
+                                f"{Color.CYAN}\nPublished status request, message 0405 should appear shortly...{Color.OFF}"
+                            )
                     # Start the background poller with subscriptions and update trigger
                     poller_task = self.loop.create_task(
                         mqtt_session.message_poller(
@@ -433,6 +449,10 @@ class AnkerSolixMqttMonitor:
                                 )
                                 break
                             await asyncio.sleep(0.5)
+                        else:
+                            if self.status_request_trigger:
+                                await asyncio.sleep(0.5)
+                                break
                 finally:
                     # Cancel the started tasks
                     poller_task.cancel()
@@ -660,6 +680,9 @@ if __name__ == "__main__":
             CONSOLE.info(
                 f"  Real-time trigger: {(Color.GREEN + 'Enabled') if arg.realtime else (Color.RED + 'Disabled')}{Color.OFF}"
             )
+            CONSOLE.info(
+                f"  Status request trigger: {(Color.GREEN + 'Enabled') if arg.status_request else (Color.RED + 'Disabled')}{Color.OFF}"
+            )
             CONSOLE.info(
                 f"  Filedump: {(Color.GREEN + 'Enabled') if arg.filedump else (Color.RED + 'Disabled')}{Color.OFF}"
             )
```

jmozmoz, 2025-12-07:
> If the App triggers, it has a timeout of 300 sek typically. So you have to wait some time after using it to see no triggered messages during start. The request parameter can be added as well, but it should trigger after the messages poller is started, otherwise the message is not printed.

Here is an update of the patch: The status requst is now issued later:
```patch
diff --git a/mqtt_monitor.py b/mqtt_monitor.py
index 5a7ea60..19bbcbb 100644
--- a/mqtt_monitor.py
+++ b/mqtt_monitor.py
@@ -52,6 +52,12 @@ def parse_arguments() -> argparse.Namespace:
         action="store_true",
         help="Enable real-time MQTT trigger on startup",
     )
+    parser.add_argument(
+        "--status-request",
+        "--sr",
+        action="store_true",
+        help="Enable immediate status request for device",
+    )
     parser.add_argument(
         "--value-display",
         "--vd",
@@ -93,6 +99,7 @@ class AnkerSolixMqttMonitor:
         self.loop: asyncio.AbstractEventLoop
         self.filedump: bool = args.filedump
         self.realtime_trigger: bool = args.realtime
+        self.status_request_trigger: bool = args.status_request
         self.value_display: bool = args.value_display
         self.fileprefix: str | None = args.dump_prefix

@@ -433,6 +440,19 @@ class AnkerSolixMqttMonitor:
                                 )
                                 break
                             await asyncio.sleep(0.5)
+                        else:
+                            if self.status_request_trigger:
+                                # individual status request
+                                if self.status_request_trigger and self.device_sn:
+                                    if mqtt_session.status_request(
+                                        deviceDict=self.device_selected,
+                                        wait_for_publish=2,
+                                    ).is_published():
+                                        CONSOLE.info(
+                                            f"{Color.CYAN}\nPublished status request, message 0405 should appear shortly...{Color.OFF}"
+                                        )
+                                await asyncio.sleep(0.5)
+                                break
                 finally:
                     # Cancel the started tasks
                     poller_task.cancel()
@@ -660,6 +680,9 @@ if __name__ == "__main__":
             CONSOLE.info(
                 f"  Real-time trigger: {(Color.GREEN + 'Enabled') if arg.realtime else (Color.RED + 'Disabled')}{Color.OFF}"
             )
+            CONSOLE.info(
+                f"  Status request trigger: {(Color.GREEN + 'Enabled') if arg.status_request else (Color.RED + 'Disabled')}{Color.OFF}"
+            )
             CONSOLE.info(
                 f"  Filedump: {(Color.GREEN + 'Enabled') if arg.filedump else (Color.RED + 'Disabled')}{Color.OFF}"
             )
```

closed 2025-12-29 by thomluther

jmozmoz, 2026-01-04:
> Hi [jmozmoz](https://github.com/jmozmoz) Have you been able to verify the commands through the monitor already?

I just have a SB2 Pro without station/extension battery. 

I tested the commands. All seem to work with the execption of the `sb_disable_grid_export_switch`.

Here is the log, if I use the app: (disable, then enable again):
```

		22:21:50 INFO: --------------------- Solarbank_2 / A17C1 / 0080 / Header ----------------------
		ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
		1e 00   : 2 Byte total message length (30) in Bytes (Little Endian format)
		03 00 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
		00 80   : 2 Byte message type pattern (varies per device model and message type)
		: 1 Byte optional message increment (  0)
		-> 18 <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
		-- Fields --|- Value (Hex/Decode Options)---------------------------------------
		Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
		a1   01  --    22
		└->   1 unk              b'"' --> pattern_22
		a5   03  02    00:00
		└->   3 sile                0               0                             0;0 --> set_disable_grid_export_a5?
		a6   03  02    01:00
		└->   3 sile                1               1                             1;0 --> set_disable_grid_export_switch
		--------------------------------------------------------------------------------
		22:21:51 INFO:

22:21:48 INFO: --------------------- Solarbank_2 / A17C1 / 0405 / Header ----------------------                        | 22:21:54 INFO: --------------------- Solarbank_2 / A17C1 / 0405 / Header ----------------------                       
a1   01  --    64                                                                                                      | a1   01  --    34                                                                                                     
└->   1 unk              b'd'                                                                                          | └->   1 unk              b'4'                                                                                         
fb   05  03    00:00:00:00                                                                                             | fb   05  03    01:00:00:00                                                                                            
└->   5 var               0;0               0           0.000         0;0;0;0 --> {'00': [{'name': 'grid_export_disabl | └->   5 var               1;0               1           0.000         1;0;0;0 --> {'00': [{'name': 'grid_export_disabl

		22:22:13 INFO: --------------------- Solarbank_2 / A17C1 / 0080 / Header ----------------------
		ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
		1e 00   : 2 Byte total message length (30) in Bytes (Little Endian format)
		03 00 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
		00 80   : 2 Byte message type pattern (varies per device model and message type)
		: 1 Byte optional message increment (  0)
		-> f2 <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
		-- Fields --|- Value (Hex/Decode Options)---------------------------------------
		Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
		a1   01  --    22
		└->   1 unk              b'"' --> pattern_22
		a5   03  02    00:00
		└->   3 sile                0               0                             0;0 --> set_disable_grid_export_a5?
		a6   03  02    00:00
		└->   3 sile                0               0                             0;0 --> set_disable_grid_export_switch
		--------------------------------------------------------------------------------
		22:22:16 INFO:

22:22:13 INFO: --------------------- Solarbank_2 / A17C1 / 0405 / Header ----------------------                        | 22:22:16 INFO: --------------------- Solarbank_2 / A17C1 / 0405 / Header ----------------------                       
bc   05  03    ee:07:00:00                                                                                             | bc   05  03    f8:07:00:00                                                                                            
└->   5 var            2030;0            2030           0.000       238;7;0;0 --> grid_to_home_power (factor 0.1)      | └->   5 var            2040;0            2040           0.000       248;7;0;0 --> grid_to_home_power (factor 0.1)     
c4   05  03    ee:07:00:00                                                                                             | c4   05  03    f8:07:00:00                                                                                            
└->   5 var            2030;0            2030           0.000       238;7;0;0 --> home_demand (factor 0.1)             | └->   5 var            2040;0            2040           0.000       248;7;0;0 --> home_demand (factor 0.1)            
fb   05  03    01:00:00:00                                                                                             | fb   05  03    00:00:00:00                                                                                            
└->   5 var               1;0               1           0.000         1;0;0;0 --> {'00': [{'name': 'grid_export_disabl | └->   5 var               0;0               0           0.000         0;0;0;0 --> {'00': [{'name': 'grid_export_disabl
```

[... cut ...]
└->   5 var               1;0               1           0.000         1;0;0;0 --> {'00': [{'name': 'grid_export_disabl | └->   5 var               0;0               0           0.000         0;0;0;0 --> {'00': [{'name': 'grid_export_disabl
```

So in `mqtt_monitor.py` `..._a5` and `..._a9` is set, while in the app only `_a5`. I removed it from the command map, but this did not help.

I am not sure, what really is used by the SB2. But the app did not show any changes, if I modify this settings with `mqtt_monitor.py`.

jmozmoz, 2026-01-05:
> With a station I mean the Station (Kraftwerk) panel, that each SB2/3 has meanwhile. It is a virtual construct to control individual device settings of one or multiple devices in the system.
So basically you need to distinguish the settings on the Station panel and on the Device panel. Device settings managed through the station are greyed out on the device panel, but you still see them.

I have app version 3.15.0.

On the start screen showing the house image with panels, network, battery I can open the settings (top right) called "Einstellen der Station". I has 6 menu entries:
- "Stromverbrauch im Alltag" where I can change between "Eigenverbrauch", "Benutzerdefinierter Modus" and Intelligenter Stecker-Modus". 
- "Strompreis festlegen"
- "Systeminformationen" with "WLAN-Informationen", "System umbennen" and "Energiesystem löschen"
- "Geräte verwalten" where I can add and remove devices (showing solarbank, smart meter).
- "Hilfe und Feedback"
- "Firmwareupdate" showing that everything is uptodate (SB: v1.0.6.4, Smart Meter: v0.0.0.40)

I cannot change the minimum SOC or export to grid settings there. These settings are only available in the settings from the "Solarbank 2 E1600 Pro" subpage, which opens, if I touch the corresponding image on the start page. 

>> I tested the commands. All seem to work with the execption of the sb_disable_grid_export_switch.

> That is a control that probably needs both, Api and MQTT command. If you check the greyed out setting on the device panel, you will see that it changes with the MQTT command, but not on the station panel.

I can only find this setting in the device panel and it is *not* greyed out there.

>> What is also not displayed correctly in the app is the "Reservierte Leistung" (output_cutoff_data). Is always says 10 % after I have changed it to 5 % by mqtt (and actually woke up the SB2). Also changing it back to 10 % by mqtt sends the SB2 to sleep again (according to the app).

> We need to distinguish the station panel and the device panel settings. I assume the station setting did not change, which is expected as explained previously. But I'm pretty sure the greyed out device setting was changed, otherwise the device would not have reacted on the control changes. At least your test proves that the MQTT control is fully applied by the device.
What you still need to check is the opposite. Is theSOC reserve change fully applied by the device if the change is done via Api?
You can verify that through the HA integration, which uses only the Api call. You will see that both SOC settings change (Station and Device panel), but I don't know if the device meanwhile fully reacts on it.
If it does, the SOC reserve does not need to be implemented at all through the MQTT command, since that was always supported through the Api.

Again, this settings seems to be only available on the device settings page and there it is *not* greyed out. 

I tested if a change of this settings by Home Assistent is visible in the app, and at first ist was not. Then, I force the app to be stopped in the Android settings page and opened ist again. Then, the change from 10 % to 5 % was visible in the app.

Then I set the value back to 10 % by `mqtt_monitor.py` and restarted the app again. Unfortnuately, this did not work. It still shows 5 % on the device page. And actually also in Home Assistent. Even after changing it back to 10 % in the app, HA still showed 5 %. But then suddendly, it shows 10 %. Perhaps it took so long for the next API update. Or it was triggered, because I switched from the "Overview" panel to the solarbank page from Integration->Anker Solix -> Solarbank?!?

Also, I changed the grid export settings by MQTT before the restart of the app, but it still shows export allowed.

thomluther, 2026-01-05:
Yeah, I was afraid of that, so basically also the SOC reserve has become a hybrid command for SB2 and requires both, Api request as well as MQTT command through the client.

jmozmoz, 2026-01-05:
This is the current output for cutoff (10 % set):
```
b4   02  01    0a
└->   2 ui                 10              10 --> output_cutoff_data
b5   02  01    05
└->   2 ui                  5               5 --> lowpower_input_data
b6   02  01    0a
└->   2 ui                 10              10 --> input_cutoff_data
```

thomluther, 2026-01-05:
Yeah, great. So the numbers that are close together wull be the same.
Once I find more time I can have a closer look.

jmozmoz, 2026-02-09:
> Is there actually any heating? Or does it go up without anything plugged in the socket?

No heating today. The "Solarbank ... Energy Consumed" went up when the bulb in the socket was turned on. 

But I think, this line in `mqttmap.py` should be changed from
```
    "c9": {NAME: "consumed_energy", FACTOR: 0.001},
```
to
```
    "c9": {NAME: "consumed_energy", FACTOR: 0.0001},
```

thomluther, 2026-03-04:
@jmozmoz 
Can you comment on [this issue](https://github.com/thomluther/ha-anker-solix/issues/463#issuecomment-3998994953)?

thomluther, 2026-03-27:
Another finding is, that your system does not provide station parameters in `device_parm_18_*` response.
That means the export switch probably cannot be toggled via the Api command that is used for system that provide `device_parm_18_*` response, since the switch is part of the station settings.
Maybe your system has to be changed with via the device attributes, which also show the setting. But I have seen, that attribute changes via Api only change the App display, but nothing on the device itself. 
So the MQTT must work, and the proper Api command may be necessary to adjust your App display as well.

jmozmoz, 2026-03-28:
> In the latter case, if the App or the cloud toggle back the switch you would have to see this MQTT command in the MQTT monitor as well. So first your toggle with the 'c' option, then later another toggle back through the cloud or App.
It does not work for you in HA because the MQTT command fails, and I need to understand why it fails. With the debug enabled, you should see the full MQTT message decoded, that is being published (if it is published).

I think, I described this behavior already in comment https://github.com/thomluther/anker-solix-api/issues/216#issuecomment-3791488127

I think, the description of the behavior there (something in the cloud is reverting the setting, because an API call is need additionally to the MQTT command) is still valid.

> In the latter case, if the App or the cloud toggle back the switch you would have to see this MQTT command in the MQTT monitor as well. So first your toggle with the 'c' option, then later another toggle back through the cloud or App.

I think this is the correct explanation.

> 'm not sure either, if there are 2 different panels where you can see the grid export switch setting:
>
>    The station panel (opened via settings from home screen)
>    The device details panel (opened via settings from the device)

I only see one place, where I can change the setting: In the device setting page (HA Sidebar: Settings -> Devices & Services -> Integrations -> Anker Solix -> Solarbank 2 E1600 Pro. But I think, we discussed also already (https://github.com/thomluther/anker-solix-api/issues/216#issuecomment-3710307924). In the app, there are also no "station settings" or "Kraftwerk". Only the device.

I cannot find it in the "station panel". Acctually, I am not sure what you mean. On the page for "System SolixMN" I can only change "Fixed price", "Price currency", "Price type". 

> What's about the export limit selection, does that work for your system?

Do you mean the setting for 350 W, 600 W and 800 W for "Output limit" on the "Solarbank 2 E1600 Pro" page? This setting seems to work. I could see, the the output changed and also the value in the app. 

> You can also switch with the monitor, using the c option like in the mqtt_monitor.

I did this: Grid export was turned off. I then "disabled" "disable_grid_export_switch", so grid export was turned on and appeared so in HA. It stayed this way, until I opened the app. Then the setting was toggling back to the orignal setting (grid export turned off). This is also shown in app.

> Hm weird, can you enable debug mode and see if there is more info with the error. Typically MQTT commands are just published, if there is an error it is either with composing the command, or the publish fails for any reason...

Sorry, but where can I turn on the debug output? I do not see such an option in the "Anker solix options" which says "Change usage of hub configuration" nor in the "system options"

thomluther, 2026-03-28:
> Sorry, but where can I turn on the debug output? I do not see such an option in the "Anker solix options" which says "Change usage of hub configuration" nor in the "system options"

[image]

thomluther, 2026-03-28:
Hm, this may be a false error in HA, the MQTT command seems to have worked, but there is maybe no response filtered out properly.

From your testing I conclude:
- The get_device_attr status for "switch_0w" is wrong, it always shows 1 and cannot be used by HA as current state. I removed that attribute already from the poller, only the MQTT value is now used as state
- The toggle is working in the app by using the set_device_attr
- The MQTT command is required in addition, and this should also update the final state

I will implement those changes, then hopefully it will work also for SB2.

thomluther, 2026-03-29:
These findings seem to match with the apk reverse engineering 
https://github.com/moag1000/anker-solix-api-exploration/blob/main/FLOWS.md#flow-7-solarbank-toggle-grid-export-switch_0w
But this suggests that switch_0w is queried to get actual status. Is that reliable to you?
Do you get status for enable_0w? This may be set only if smart meter installed.if reliable I could use it as flag to create the switch entity.

jmozmoz, 2026-03-29:
Now, I tested, what happens if I change the switch_0w (to allow grid export) by the API and before that by `mqtt_monitor.py`. This was the result:

```
Start switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": {
    "device_sn": "XXX",
    "attributes": {
      "enable_0w": 0,
      "switch_0w": 1
    }
  },
  "trace_id": "XXX"
}
Set switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": null,
  "trace_id": "XXX"
}
Final switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": {
    "device_sn": "XXX",
    "attributes": {
      "enable_0w": 0,
      "switch_0w": 0
    }
  },
  "trace_id": "XXX"
}
```

So now, it looks like the values returned by the cloud seems to be correct.

But I have a smart meter. And until now, the result for `enable_0w` has always been 0.

I changed one thing, I am only querrying for those two values. Could this explain the different behavior?

Turning off grid export again, also seems to work. First MQTT, then API:

```
Start switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": {
    "device_sn": "XXX",
    "attributes": {
      "enable_0w": 0,
      "switch_0w": 0
    }
  },
  "trace_id": "XXX"
}
Set switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": null,
  "trace_id": "XXX"
}
Final switch_0w:
{
  "code": 0,
  "msg": "success!",
  "data": {
    "device_sn": "XXX",
    "attributes": {
      "enable_0w": 0,
      "switch_0w": 1
    }
  },
  "trace_id": "XXX"
}
```

thomluther, 2026-03-30:
@jmozmoz 
I have uploaded a new 3.5.4 beta version in the HA main branch.
Please install it and check whether the settings now all work. With your confirmation, I can hopefully release it this week.

Under developper tools -> actions -> update.install action:
- Target: update.anker_solix
- version: main

[image]

Ignore the notification that a new update is available after the install, its because there is no version tag yet for the main branch.
Once 3.5.4 will be released, you will see it again as new update.